alsa-project/alsa-lib #154: Config parsing of pcm_rate broken since 1.1.7 (at least for a52 pcm plugin)

quequotion, 2021-06-11:
As noted [in the mailing list](https://mailman.alsa-project.org/pipermail/alsa-devel/2019-December/159426.html) back in 2019, config parsing (for alsa-plugins a52 pcm at least) has been broken since alsa-lib 1.1.7.

As far as I know this was never bisected to a single commit in either alsa-lib or alsa-plugins.

Looking at the mailing list discussion, it occurs to me that it may have given the impression that the config parser was broken when using [a patch that ports alsa-plugins to libswresample](https://git.harting.dev/anonfunc/alsa-plugins/commit/9cdbbb9874757b6f8fda7fb4ac2e3fc59da65946).

Config parsing is broken *regardless* of that patch.

@IdleGandalf worked around this by applying a patch to alsa-lib, in [src/pcm/pcm_rate.c](https://git.harting.dev/anonfunc/alsa-lib/commit/646bc80a63b0220300678a38e0c26a9a13af6fc0), but never got a response if this was an appropriate method to resolve the issue.

Should this be fixed in alsa-lib or in alsa-plugins?

Note: above links are to mirrors of the original patches, also in a repository owned by @IdleGandalf as far as I can tell. The links in the mailing list discussion are now 404.

perexg, 2021-06-11:
The fix does not seem relevant. What does not work? Could you show your configuration and describe the problem?

quequotion, 2021-06-11:
The same configuration most everyone across the internet uses to enable the a52 plugin:

```
# Digital 5.1 AC3 Output
pcm.a52 {
  @args [CARD]
  @args.CARD {
    type string
  }
  type rate
  slave {
    pcm {
      type a52
      bitrate 448
      channels 6
      card $CARD
    }
    rate 48000
  }
}
```

This configuration worked up to alsa-lib 1.1.6, and then it did not. There doesn't seem to be *any* configuration that works after alsa-lib 1.1.7. What @IdleGandalf did was to [filter out all keywords](https://mailman.alsa-project.org/pipermail/alsa-devel/2019-December/159425.html) to work around the problem.

Nothing has changed since: the configuration (using vanilla, stable releases of alsa-plugins and alsa-lib) still fails with "Unknown field card" today just as it did back then.

perexg, 2021-06-11:
The applied filter was for the `pcm.a52 { }` compound, so it's not relevant for `pcm.a52.slave { }` compound.

This one does work:

````
pcm.aloop48000 {
  type rate
  slave {
     pcm hw:Loopback
     rate 48000
  }
}
````

quequotion, 2021-06-11:
This may be limited to the a52 plugin; that is one of the things I am trying to find out with this issue report.

Here it is in an excerpt from pulseaudio's startup:

```
D: [pulseaudio] alsa-mixer.c: Looking at profile output:iec958-ac3-surround-51
D: [pulseaudio] alsa-mixer.c: Checking for playback on Digital Surround 5.1 (IEC958/AC3) (iec958-ac3-surround-51)
D: [pulseaudio] alsa-util.c: Trying a52:0 with SND_PCM_NO_AUTO_FORMAT ...
I: [pulseaudio] (alsa-lib)pcm_rate.c: Unknown field card
I: [pulseaudio] alsa-util.c: Error opening PCM device a52:0: Invalid argument
D: [pulseaudio] alsa-mixer.c: Caching failure to open output:iec958-ac3-surround-51
D: [pulseaudio] alsa-mixer.c: Skipping profile output:iec958-ac3-surround-51+input:analog-stereo - will not be able to open output:iec958-ac3-surround-51
D: [pulseaudio] alsa-mixer.c: Skipping profile output:iec958-ac3-surround-51+input:iec958-stereo - will not be able to open output:iec958-ac3-surround-51
```

perexg, 2021-06-11:
Could you check the merged config dump  (with 1.2.5) ? `alsactl dump-cfg` - show only the `a52` config compound .

perexg, 2021-06-11:
And btw, `a52` is already defined in 60-a52-encoder.conf ! So if you want to override it, you should create `pcm.!a52 { }` config.

quequotion, 2021-06-11:
This?

```
pcm.a52 {
	@args.0 CARD
	@args.1 SLAVE
	@args.2 RATE
	@args.3 BITRATE
	@args.4 CHANNELS
	@args.CARD {
		type string
		default {
			@func refer
			name 'defaults.pcm.card'
		}
	}
	@args.SLAVE {
		type string
	}
	@args.RATE {
		type integer
		default 48000
	}
	@args.BITRATE {
		type integer
		default 448
	}
	@args.CHANNELS {
		type string
		default 6
	}
	@args.5 CARD
	type rate
	card $CARD
	slavepcm $SLAVE
	rate $RATE
	bitrate $BITRATE
	channels $CHANNELS
	hint {
		show {
			@func refer
			name 'defaults.namehint.basic'
		}
		description 'Plugin to convert multichannel stream to A52 (AC3) bitstream'
	}
	slave {
		pcm {
			type a52
			bitrate 448
			channels 6
			card $CARD
		}
		rate 48000
	}
}
```

perexg, 2021-06-11:
Yes, and you see the culprit. The configs are merged!

quequotion, 2021-06-11:
> And btw, `a52` is already defined in 60-a52-encoder.conf ! So if you want to override it, you should create `pcm.!a52 { }` config.

I don't think *anyone* is aware of this.

perexg, 2021-06-11:
https://www.alsa-project.org/alsa-doc/alsa-lib/conf.html - "Operation modes for parsing nodes"

quequotion, 2021-06-11:
I mean that there is an existing configuration for a52.

The plugin is not activated unless using the configuration posted above, so no *users* of the plugin would be aware that a conflicting configuration to be merged with this configuration *exists*.

perexg, 2021-06-11:
It would be also nice, if you can check the default 60-a52-encoder.conf , so we can eventually fix the problem with it (rename your config for tests). The rate resampling should not be mandatory for pulseaudio.

quequotion, 2021-06-11:
Okay, now we are getting somewhere.

So, after another reboot, the plugin works *in pulseaudio* without any additional configuration (using 60-a52-encoder.conf as provided by alsa-plugins). Unfortunately, this leaves no pcm device to be tested with speaker-test when pulseaudio is disabled, but I can live with that.

I am curious if there is a way to access the a52 pcm *without* pulseaudio. What the carg-cult configuration did was to establish this device in alsa, which some were using instead.

perexg, 2021-06-11:
`speaker-test -Dplug:a52` ?

quequotion, 2021-06-11:
> `speaker-test -Dplug:a52` ?

Still getting stuck on "Front Left"

quequotion, 2021-06-12:
In any case, I am satisfied that parsing the configuration was not, or at least is not now, the issue: the way of configuring this plugin has changed (a manual configuration is no longer required to use the plugin, at least with pulseaudio)

closed 2021-06-12 by quequotion

manio, 2021-06-12:
@quequotion 
two questions:
1. you did all above tests with https://github.com/alsa-project/alsa-plugins/pull/23/ applied?
2. so you are telling me that pulseaudio is able to use the a52 plugin and the sound is magically working without 32bit sampling? 

ps. I am not using pulseaudio at all so I am also not so familiar with it.... maybe it is a time to change this habit :D

quequotion, 2021-06-12:
>1. you did all above tests with [alsa-project/alsa-plugins#23](https://github.com/alsa-project/alsa-plugins/pull/23) applied?

Yes

> 2. so you are telling me that pulseaudio is able to use the a52 plugin and the sound is magically working without 32bit sampling?

Not that well, the audio is as you described in your tests: pretty terrible really (choppy, chipmunk voice effect).

> ps. I am not using pulseaudio at all so I am also not so familiar with it.... maybe it is a time to change this habit :D

There are still many people who distrust pulseaudio or think of it as bloat. From my experience, unless you are running on an embedded system or otherwise have some critical need to save a few CPU cycles and megabytes of RAM, pulseaudio is a very reasonable thing to use. It enables quite a lotbof advanced features and provides a relatively easy way to configure them over ALSA on its own.

On the other hand, your current setup is a good test bed: we do need to make sure standalone ALSA setups are supported.

perexg, 2021-06-12:
If you use `plug:a52` then the automatic rate / format conversion is activated. If something does not work, it's problem is probably in the end plugin (a52). You may dump the chain using `aplay -Dplug:a52 -v <your_file>`.

quequotion, 2021-06-14:
I tried this with [this test file](https://web.archive.org/web/20140114061031/http://lynnepublishing.com/surround/www_lynnemusic_com_surround_test.ac3).

Normally, this announces each channel (Front left, Center, Front Right, Rear left, Rear right, <noise through LFE>).

```
aplay -D plug:a52 -v www_lynnemusic_com_surround_test.ac3
Playing raw data 'www_lynnemusic_com_surround_test.ac3' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
Plug PCM: Rate conversion PCM (48000, sformat=S16_LE)
Converter: libspeex (external)
Protocol version: 10002
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 4096
  period_size  : 256
  period_time  : 32000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 256
  period_event : 0
  start_threshold  : 4096
  stop_threshold   : 4096
  silence_threshold: 0
  silence_size : 0
  boundary     : 1152921504606846976
Slave: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0
    1 <- 0
    2 <- 0
    3 <- 0
    4 <- 0
    5 <- 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24576
  period_size  : 1536
  period_time  : 32000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 1536
  period_event : 0
  start_threshold  : 24576
  stop_threshold   : 24576
  silence_threshold: 0
  silence_size : 0
  boundary     : 6917529027641081856
Slave: A52 Output Plugin
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_NONINTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 6
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24576
  period_size  : 1536
  period_time  : 32000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 1536
  period_event : 0
  start_threshold  : 24576
  stop_threshold   : 24576
  silence_threshold: 0
  silence_size : 0
  boundary     : 6917529027641081856
```

The sound produced by `aplay` time was an atrocious grinding noise through all channels simultaneously.

I am not entirely sure if that is *only* because of the encoding, or if `aplay` needs additional parameters to use the plugin properly.

Playing the same file with `mpv` through `pulseaudio` produces a *listenable* playback of the channel announcements through their proper channels, but with a choppy chimpunk effect.

```
mpv www_lynnemusic_com_surround_test.ac3
[ffmpeg/demuxer] ac3: Estimating duration from bitrate, this may be inaccurate
 (+) Audio --aid=1 (ac3 6ch 48000Hz)
AO: [pulse] 48000Hz 5.1(side) 6ch float
A: 00:00:08 / 00:00:09 (96%)

Exiting... (End of file)
```

perexg, 2021-06-14:
All is ok here, let continue in https://github.com/alsa-project/alsa-plugins/pull/23 .