alsa-project/alsa-lib #180: MIDI 2.0 support in ALSA

robertwu1, 2021-10-04:
Hi, now that USB MIDI 2.0 is officially out (https://www.midi.org/specifications/midi-2-0-specifications), are there any updates for support in USB MIDI 2.0 in ALSA?

https://github.com/alsa-project/alsa-lib/issues/24 was closed in February of 2020 but it's been over a year and a half. There seems to be quite a bit of interest in the issue since it had been closed.

Thanks!

abique, 2022-01-14:
I'm also interested to know what is the status / roadmap with MIDI 2.

symdeb, 2022-06-06:
https://github.com/alsa-project/alsa-lib/issues/24

tiwai, 2023-05-23:
The MIDI 2.0 support (for core UMP and USB MIDI 2.0) is finally merged to sound.git tree as a 6.5 kernel material. Found in for-next branch (or topic/midi20 branch).

The updates for alsa-lib and alsa-utils will follow.

symdeb, 2023-06-06:
Found it.It looks great !  P.S: Is the USB gadget API also supported for UMP ?

symdeb, 2023-06-12:
Updated MIDI Specifications were release on the MIDI.ORG website today.