alsa-project/alsa-lib #187: alsatplg broken with musl-libc

Johnnynator, 2021-11-01:
alsatplg feeds empty and non Integer strings to `strtol(3)` and friends. Glibc just silently accepts such inputs and does not set errno and returns 0. Musl sets `errno` to `EINVAL` with such inputs.
This can easily be seen by most of the `alsatplg` invocation when compiling https://github.com/thesofproject/sof/

`STRTOL(3)` manpage explains that this behavior is allowed.
>  The implementation may also set errno to EINVAL in case no conversion
    was performed (no digits seen, and 0 returned).

There are like two ways I see this can be fixed

a) ignore EINVAL (and check if base is correct ourself)
b) make sure that strtol is only getting called with a valid input (I'm not sure if the current behavior is intended to just return 0 if e,g, `tplg_get_unsigned` did not even parse anything valid

perexg, 2021-11-02:
I would use `safe_strtol()` function and check for the valid input - commit 5fab157a593180525607b7d26263bee127f6be36 . I think that NULL pointer is not passed to `strtol`.

@plbossart , @ranj063 : FYI. It seems that the buggy input is parsed.

perexg, 2021-11-07:
Fix is available, reopen in case of other issues.

closed 2021-11-07 by perexg