alsa-project/alsa-lib #188: Issues again with Comet Lake PCH cAVS  and pipewire

leiserfg, 2021-11-12:
At a certain point after v1.2.5-65-gf0d540f8 pipewire stopped showing any device for the cards Comet Lake PCH cAVS.

perexg, 2021-11-12:
Any logs / errors / consultation with the pipewire developers ?

leiserfg, 2021-11-12:
I created an issue there but I'm still waiting for replies.

perexg, 2021-11-12:
Links ?

leiserfg, 2021-11-12:
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1798

perexg, 2021-11-12:
Ok, it's difficult to analyze something without logs. There are two changes on top of f0d540f851de3992a5e285dcea0be7cab293527e : https://github.com/alsa-project/alsa-lib/compare/f0d540f851de3992a5e285dcea0be7cab293527e..HEAD . They are not seem relevant.

Could you do a test on command line with `alsaucm -c hw:PCH dump text` ?

leiserfg, 2021-11-12:
Sorry, I was in a meeting, I ran `spa-acp-tool -vvvv lv 2>/tmp/the_name.log ` with versions (previous to November and current master) and this are the outputs:

[old.log](https://github.com/alsa-project/alsa-lib/files/7527890/old.log)
[new.log](https://github.com/alsa-project/alsa-lib/files/7527891/new.log)

perexg, 2021-11-12:
Difference:

```
Set ucm verb to HiFi
Trying _ucm0001.hw:sofhdadsp,5 with SND_PCM_NO_AUTO_FORMAT ...
Managed to open _ucm0001.hw:sofhdadsp,5
Maximum hw buffer size is 341 ms
Set buffer size first (to 4797 samples), period size second (to 1199 samples).
Device hw:sofhdadsp,5 doesn't support 44100 Hz, changed to 48000 Hz.
```

```
Set ucm verb to HiFi
Trying _ucm0001.hw:sofhdadsp,5 with SND_PCM_NO_AUTO_FORMAT ...
Error opening PCM device _ucm0001.hw:sofhdadsp,5: No such device
```

So if you run the f0d540f851de3992a5e285dcea0be7cab293527e version the things will work, but if you go to faee7f678d156e586be2198c9380b7ae322e3730 or ea5481296f281625ef6517208ebf6e41b5fd838c, there are no devices ?

Use `git checkout faee7f678d156e586be2198c9380b7ae322e3730` for example and return to master `git checkout master`.

leiserfg, 2021-11-12:
I bisected from master to 0e2b167e (this is the one was fine)
and seems like the issue started appearing with f547b2e35f71e89397e8b29cd0b9083bb68b00a6

perexg, 2021-11-12:
Makes sense. The fix is in https://github.com/alsa-project/alsa-lib/commit/050f18c525d44f3edbdd14bff189e40e45cc8b89 . Thank you for your report.

closed 2021-11-12 by perexg