alsa-project/alsa-lib #189: Unable to connect KEF X300A USB speakers

simbatl, 2021-11-12:
I'm trying to USE a pair of KEF X300A speakers. They work with aplay on Ubuntu 20.04 server, but fail on Ubuntu 18.04, 20.04 and 21.10 desktop.

Here is output from pulseaudio -vvv:
[pulse.log](https://github.com/alsa-project/alsa-lib/files/7527770/pulse.log)

And here is output from alsa-info.sh:
[alsa-info.txt](https://github.com/alsa-project/alsa-lib/files/7527773/alsa-info.txt)

Anyone who know how to solve this issue? :)

perexg, 2021-11-12:
It looks like a general USB issue. Check your kernel logs.

simbatl, 2021-11-12:
It works on windows 10 and ubuntu 20.04 server without any problems, so think usb connection is ok.

Here is output from dmesg:
[dmesg.log](https://github.com/alsa-project/alsa-lib/files/7528033/dmesg.log)

perexg, 2021-11-12:
There are a log of errors for `usb-1.8` bus. It may be a problem somewhere in hardware or kernel or a timing issue (which is a bit problematic to debug). You may check the kernel version and report the issue at https://bugzilla.kernel.org or better using the distribution specific reporting system (all distributions modifies kernels a bit). They'll work with you to resolve this issue.

closed 2021-11-12 by perexg

simbatl, 2021-11-12:
Ok, thanks! :)

perexg, 2021-11-12:
Also, it may be worth to try to connect PC and speakers to one phase (one socket).

simbatl, 2021-11-12:
They already are, and the speakers works on the same usb port if I pass it to windows 10 with virtualbox. :o