alsa-project/alsa-lib #193: project

yangyunfei16, 2021-11-21:
Dear alsa developers:
Hello. I am a graduate student in the Institute of information engineering, Chinese Academy of Sciences. We are doing a project to investigate the use of this library by some programmers who use alsa for programming, so we want to ask if there are any websites or forums on which more programmers who use this library communicate and discuss, We want them to do a questionnaire later. Thank you very much for your help!

geraldog, 2021-11-21:
Hi @yangyunfei16 , I believe you are searching for https://mailman.alsa-project.org/mailman/listinfo/alsa-devel

geraldog, 2021-11-21:
I know students are usually on tight schedules but personally I'd recommend they lurk around for at least 2 weeks  on alsa-devel to get a feel of how the list works. They will see, as with most kernel development lists, that [PATCH] is what garners attention, that is, the workflow of the mailing list is the maintainers and part of the community reviewing and eventually accepting (or not) the proposed code changes.

yangyunfei16, 2021-11-22:
@geraldog Thank you. But what I want to find is a forum. There are many people who use alsa programming. I want to ask them to fill in the questionnaire I designed about the use of alsa library.

geraldog, 2021-11-22:
Sorry @yangyunfei16 I don't know any forums specifically about ALSA, either official or unofficial. Can't they answer your questionnaire after reading the documentation?

perexg, 2021-11-23:
We have all know information on our web pages: https://www.alsa-project.org . Anyone can enter any related information there. If there is no forum link, it probably does not exists. As the founder of the ALSA project, I do not know about any other developer forum. All discussions are made in the alsa-devel mailing list or in this tracker now. The application developers usually join when they are trying to solve an issue. And it's no so difficult to find the open source applications using this library, right?

closed 2021-11-23 by perexg

yangyunfei16, 2021-11-23:
@geraldog Actually, the reason why we look for a forum is that we need a platform where we can find enough developers using ALSA to answer the questionnaire. They are supposed to read the documentation when they answer it.
It seems there is no forum. Do you have any suggestions on where and how we can find enough ALSA users? Thank you so much.

geraldog, 2021-11-23:
@yangyunfei16 I know @perexg closed this issue right before you could get back to me but I'll bite and answer anyway.

I know there are some communication barriers between Chinese and English but as a graduate student I think you should have described a bit more your intent with the questionnaire and about the aim of your research.

At this point I don't even know if it's a good fit for alsa-devel mailing-list like I originally suggested.

I suggest you read https://www.alsa-project.org/wiki/Mailing-lists specially the Big Warning.

If it's a plain-text questionnaire or you can send the link to it in plain-text to alsa-devel mailing list maybe you'll have a handful of answers but to be honest, expect half a dozen, not 100+ answers.

If you need a lot of answers for your research, I hate to say that, but you'll have to ask your institution to budget targeted advertising in Google AdWords, Facebook Ads etc. targeting ALSA-related keywords (https://alsa.opensrc.org/ is an unnoficial but juicy keyword source) and explaining about the questionnaire, maybe even throwing a reward in like some stickers to encourage people to take the questionnaire.

I hate to propose some route where you have to pay, worse yet, pay for advertising, but it may be the only way.