alsa-project/alsa-lib #194: Xonar U7 MKII Analog Multichannel and SPDIF

Gebbx, 2021-11-21:
Hi,
Usb sound card Xonar U7 MKII plays multichannel content without issue, but only through aplay
`aplay -D hw:2,0 test.wav #6 channels file`
in other cases only two channels work (downmixed)
I have tried different combinations of settings.
Editing asound.conf gives no results too.

The digital output also does not work in any combination of settings, but works (only two channels) in the following case:
`pcm_slave.u7spdif { pcm "hw:2,1" } pcm.u7spplug { type plug slave u7spdif hint { show on } }`
If I understand correctly, in this case it output the decoded signal and it will always be two-channel.
I found a topic: https://forum.kodi.tv/showthread.php?tid=253179
So, i have to edit USB-Audio.conf, 
```
USB-Audio.pcm.iec958_device {
# "NoiseBlaster 3000" 42
"USB Sound Blaster HD" 1
"Xonar U7" 1
"MKII" 1 # my card
}
```
but this is not recommended.
How can this be done in another way?
Example given in the USB-Audio.conf file forces system to reboot.
`cards.USB-Audio.pcm.iec958_device."MKII" 1`

Thanks!

My system: RPI4, LibreELEC 10

perexg, 2021-11-23:
If the multichannel stream works using aplay, it works in ALSA. You should contact the application (probably Kodi) developers for the further questions.

Can you show me `cat /proc/asound/cards` contents to verify, if the card is identified as `MKII` only ?

Gebbx, 2021-11-24:
```
 0 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
```
```
 1 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1
                      vc4-hdmi-1
```
```
 2 [MKII           ]: USB-Audio - Xonar U7 MKII
                      ASUS Xonar U7 MKII at usb-0000:01:00.0-1.4, high speed
```

perexg, 2021-12-02:
Thanks. We have already `Xonar U7 MKII` string in USB-Audio.conf (commit https://github.com/alsa-project/alsa-lib/commit/93752fb4de397554e92a4eb9079f77dabaec7d7f ). You need alsa-lib version 1.2.5 or later. The `MKII` is a card alias which may be modifed.

If `hw:` device works for you, then the ALSA subsystem is working. You need to consult the used application. In case of Kodi, try to set the audio channels in the settings - https://kodi.wiki/view/Settings/System/Audio .

closed 2021-12-02 by perexg