alsa-project/alsa-lib #199: alsa-lib-1.2.6 regression, no audio

gkubaryk, 2021-12-09:
after upgrading the following packages, there is no audio:
=media-plugins/alsa-plugins-1.2.6
=media-sound/alsa-utils-1.2.6
=media-libs/alsa-ucm-conf-1.2.6
=media-libs/alsa-lib-1.2.6

I checked each of the four individually and the last one (alsa-lib, 1.2.5.1) is the one that restored audio.

alsa-info.sh output, as run from 1.2.5.1 (but it downloaded the newer version of the script to /tmp) is at http://alsa-project.org/db/?f=f94a60ff547f9cce9e872acae67fdf35473fa754

let me know if I can be of further assistance; I need the audio working for the rest of the day but would be able to provide output from 1.2.6 tomorrow perhaps, if desired.

gkubaryk, 2021-12-09:
eh decided to go ahead and diagnose more ... using tags 1.2.5.1 good and 1.2.6 bad.
bisected to f547b2e35f71e89397e8b29cd0b9083bb68b00a6

perexg, 2021-12-09:
Could you show me output from `alsaucm -c hw:1 set _verb HiFi` and `alsaucm -c hw:1 dump text` outputs (put http links here to gist or pastebin only). Thank you.

tiwai, 2021-12-09:
> eh decided to go ahead and diagnose more ... using tags 1.2.5.1 good and 1.2.6 bad. bisected to [f547b2e](https://github.com/alsa-project/alsa-lib/commit/f547b2e35f71e89397e8b29cd0b9083bb68b00a6)

It's likely the change in uc_mgr_alibcfg_by_device(), where the name string is passed as-is now while it was name+4 to drop the prefix.

perexg, 2021-12-09:
> It's likely the change in uc_mgr_alibcfg_by_device(), where the name string is passed as-is now while it was name+4 to drop the prefix.

It's fixed in 050f18c525d44f3edbdd14bff189e40e45cc8b89 and 7ffe3d41626279e43fddfec8a02f520c3b79190e.

gkubaryk, 2021-12-09:
> It's fixed in [050f18c](https://github.com/alsa-project/alsa-lib/commit/050f18c525d44f3edbdd14bff189e40e45cc8b89) and [7ffe3d4](https://github.com/alsa-project/alsa-lib/commit/7ffe3d41626279e43fddfec8a02f520c3b79190e). It looks like an UCM configuration issue.

those two commits are in 1.2.6 though, which is the release that is not working

gkubaryk, 2021-12-09:
> Could you show me output from `alsaucm -c hw:1 set _verb HiFi` and `alsaucm -c hw:1 dump text` outputs (put http links here to gist or pastebin only). Thank you.

https://gist.github.com/gkubaryk/5985599230687f52352cd8c440c0c499

perexg, 2021-12-09:
The dump seems like for the 1.2.5 version.

gkubaryk, 2021-12-09:
it is, let me switch everything back to 1.2.6, will have new output shortly.

gkubaryk, 2021-12-09:
https://gist.github.com/gkubaryk/4b279a0ab2971752060cd9df622011a5

perexg, 2021-12-09:
It's correct. It looks like another bug. Could you show me the log from pulseaudio ? https://fedoraproject.org/wiki/How_to_debug_PulseAudio_problems

gkubaryk, 2021-12-09:
like so? https://gist.github.com/gkubaryk/c0e9821388f6b8fd35b2e96493ad640d

perexg, 2021-12-09:
No, I meant the debug log (-vvv). But it seems that your problem is similar to https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1246 . I can reproduce here - working on it.

gkubaryk, 2021-12-09:
this more useful? https://gist.github.com/gkubaryk/19d1fc975176cfcf65bcfa7925b5f9f3

perexg, 2021-12-09:
Yes, it's the same issue. Thank you. Give me few minutes.

```
Dec  9 03:46:25 localhost pulseaudio[4400]: [pulseaudio] alsa-util.c: Trying _ucm0001.hw:Generic,7 with SND_PCM_NO_AUTO_FORMAT ...
Dec  9 03:46:25 localhost pulseaudio[4400]: [pulseaudio] pcm_hw.c: Invalid type for subdevice
Dec  9 03:46:25 localhost pulseaudio[4400]: [pulseaudio] alsa-util.c: Error opening PCM device _ucm0001.hw:Generic,7: Invalid argument
```

closed 2021-12-09 by perexg

perexg, 2021-12-09:
Thank you for your help and co-operation. The fix is in https://github.com/alsa-project/alsa-lib/commit/4b22871ee5a3bbc8eef7039bfc8db5e65ff3895f .

gkubaryk, 2021-12-09:
unfortunately, with 4b22871 applied atop alsa-lib-1.2.6, the issue persists

https://gist.github.com/gkubaryk/e3b9a184bcd3068fdbe51279863276f0

perexg, 2021-12-09:
It's not relevat to alsa-lib. Could yout try fix in  https://github.com/alsa-project/alsa-ucm-conf/commit/60ea26507390bad3742af2272ba08b21f009d0ef ?

gkubaryk, 2021-12-09:
I confirm that between 4b22871 and alsa-project/alsa-ucm-conf@60ea265 - all is well!

Thank you for your work on this. Do you think there will be a 1.2.6.1 soon with these improvements? In the meantime I am using /etc/portage/patches locally so this specific version for these two packages is patched for me.

perexg, 2021-12-09:
I'll try to release 1.2.6.1 alsa-lib and alsa-ucm-conf bugfix packages in next hours. Thank you for your report again.