alsa-project/alsa-lib #200: microphone disappeared on alsa-ucm-conf 1.2.6.1

galibozek, 2021-12-09:
after upgrade testing/alsa-ucm-conf 1.2.5.1-1 to 1.2.6.1-1 on arch linux my microphone disappeared on lenovo t14 gen2i

nirs, 2021-12-10:
Same problem in Fedora 35 after upgrading from alsa-lib-1.2.5.1-3.fc35.x86_64 to alsa-lib-1.2.6-1.fc35.x86_64.
Downgrading back to 1.2.5.1-3 seems to restore the microphone.

Hardware:
Lenovo P1 Gen 3

perexg, 2021-12-10:
Does the simple change in https://github.com/alsa-project/alsa-ucm-conf/pull/127 work for you?

galibozek, 2021-12-10:
@perexg unfortunately, it does not solve my problem

perexg, 2021-12-10:
Could you give me a link to the `alsa-info.sh` output (with the recent packages)?

Also, show me error/output from `alsaucm -c hw:0 set _verb HiFi` and `alsaucm -c hw:0 dump text` (upload the output to gist or pastebin and show only URL here). Thank you.

perexg, 2021-12-10:
> Same problem in Fedora 35 after upgrading from alsa-lib-1.2.5.1-3.fc35.x86_64 to alsa-lib-1.2.6-1.fc35.x86_64. Downgrading back to 1.2.5.1-3 seems to restore the microphone.
> 
> Hardware: Lenovo P1 Gen 3

It's already fixed. Upgrade to alsa-lib-1.2.6-3 : https://bodhi.fedoraproject.org/updates/FEDORA-2021-e6f6ed66d8

galibozek, 2021-12-10:
@perexg where find patch? i'm using arch linux :)

perexg, 2021-12-10:
alsa-lib-1.2.6.1 is already fixed. -3 is the package release for Fedora only (which has this fix)

Please, follow my instructions in my previous comment - upgrade to 1.2.6.1 and fix the include path in the /usr/share/alsa/ucm2/ tree according https://github.com/alsa-project/alsa-ucm-conf/pull/127 .

galibozek, 2021-12-10:
@perexg I did so, and I gave the result above

perexg, 2021-12-10:
> @perexg I did so, and I gave the result above

??? I don't see any outputs from alsa-info.sh nor alsaucm.

galibozek, 2021-12-10:
alsa-info.sh: https://pastebin.com/JWW7bGBC

$ sudo alsaucm -c hw:0 set _verb HiFi
```
ALSA lib main.c:1412:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
alsaucm: error failed to open sound card hw:0: No such file or directory
```
$ sudo alsaucm -c hw:0 dump text
```
ALSA lib main.c:1412:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
alsaucm: error failed to open sound card hw:0: No such file or directory
```

galibozek, 2021-12-10:
in kde sound applet is shown as unavailable
[image]

perexg, 2021-12-10:
Could you run alsaucm commands as root? The output should be different, if the card is present.

galibozek, 2021-12-10:
I runned command as root via sudo

perexg, 2021-12-10:
I have a feedback from the other sources (https://bodhi.fedoraproject.org/updates/FEDORA-2021-e6f6ed66d8) that alsa-lib-1.2.6.1 and alsa-ucm-conf-1.2.6.2 should resolve the issues with the Intel SOF drriver (digital microphones). Please, wait until your distribution updates the alsa-ucm-conf package and retest.

The alsaucm output is not correct, I'll try to dig into this more when the package upgrade does not help. Thank you for your co-operation to resolve this issue.

galibozek, 2021-12-10:
no change after apply this patch
https://github.com/alsa-project/alsa-ucm-conf/commit/b0ffdb057133f6e3fe1484ff72b5115dc471f303

perexg, 2021-12-10:
The bytcht-es8316 patch is not relevant. I would try to determine, why alsaucm commands don't show anything. They should work also with the 1.2.5 version. I usually use `strace` to look which files are touched.

galibozek, 2021-12-10:
sudo strace alsaucm -c hw:0 set _verb HiFi
https://pastebin.com/HvMh8L4d

perexg, 2021-12-10:
What's your /usr/share/alsa/ucm2/ucm.conf file ?

galibozek, 2021-12-10:
alsa-ucm-conf: 1.2.6.1-1
/usr/share/alsa/ucm2/ucm.conf: https://pastebin.com/n1by5xps

galibozek, 2021-12-10:
after downgrade alsa-topology-conf to: 1.2.5.1-1

sudo strace alsaucm -c hw:0 set _verb HiFi
https://pastebin.com/jADgBwpR

perexg, 2021-12-10:
The problem with your 1.2.6 UCM files is:

```
access("/usr/share/alsa/ucm2/conf.d/sof-hda-dsp/sof-hda-dsp.conf", R_OK) = -1 ENOENT (Nie ma takiego pliku ani katalogu)
```

It should be link to:

```
alsa-ucm-conf-1.2.6.2/ucm2/conf.d/sof-hda-dsp/sof-hda-dsp.conf -> ../../Intel/sof-hda-dsp/sof-hda-dsp.conf`
```

It looks like bad packaging or so...

dvzrv, 2021-12-10:
Hi! I'm packaging alsa-ucm-conf for Arch Linux. There seems to be a bug in the package script for this package, which did not install any of the symlinks.

> It looks like a bad packaging or so...

Given that there is neither a build system nor any instructions in the project explaining where things have to go, your statement rather points out that there are open questions in regards to how to install these files and that this needs improvement.

I recall other problems around these files and I guess even a simple Makefile would already help in getting this right for everyone without anyone having to guess.
Can I assume, that all files are just copied verbatim to the file system below `/usr/share/alsa/` or are also other locations possible? I can provide a PR for this with a little guidance on what the possible locations are.

galibozek, 2021-12-10:
alsa-ucm-conf 1.2.6.2-2 (arch linux) fixes the problem

perexg, 2021-12-10:
@dvzrv : Thank you for the suggestion. Is text in https://github.com/alsa-project/alsa-ucm-conf/commit/3fe9216bc91794d77241cc7c8f21f961f83db489 enough ? There is no installation script, because it's just a simple copy. I use tar for Fedora to unpack the source directly to the proper location like:

```
tar xvjf %{SOURCE1} -C %{buildroot}/%{_datadir}/alsa --strip-components=1 "*/ucm" "*/ucm2"
```

dvzrv, 2021-12-10:
> Is text in alsa-project/alsa-ucm-conf@3fe9216 enough ?

Sure, thanks!
At first I thought the libdir is also exposed in alsa-lib's pkgconfig integration (and could e.g. be easily integrated into some install steps), but that seems to be not the case.

injcristianrojas, 2021-12-11:
Subscribing to this discussion. On Fedora 35 not only my microphone disappeared, but also a HDMI/DisplayPort profile. Sticking with alsa-lib/alsa-ucm 1.2.5.1-3.fc35 using `dnf versionlock` is a workaround.

perexg, 2021-12-11:
> Subscribing to this discussion. On Fedora 35 not only my microphone disappeared, but also a HDMI/DisplayPort profile. Sticking with alsa-lib/alsa-ucm 1.2.5.1-3.fc35 using `dnf versionlock` is a workaround.

What exactly does not work with https://bodhi.fedoraproject.org/updates/FEDORA-2021-e6f6ed66d8 ? Did you update to latest packages?

injcristianrojas, 2021-12-11:
@perexg sorry, nevermind. I did an upgrade using bodhi yesterday and it didn't work. Now I did a full `dnf upgrade --refresh` and the HDMI profile is back again. Thank you.

perexg, 2021-12-13:
Fixed in alsa-lib-1.2.6.1 and alsa-ucm-conf-1.2.6.2.

closed 2021-12-13 by perexg