alsa-project/alsa-lib #203: alsa crashing each time my screen gets locked

Fallast, 2021-12-13:
each time i get my screen lock after being inactive for 5 minutes i log back in and alsa stops working and my volume icon disappears
i have to run volumeicon to get it working back again
i'm on mx linux, based on debian bullseye

perexg, 2021-12-13:
Could you show output (attach or post a gist/pastebin URL) to the `alsa-info.sh --no-upload` output before and after suspend (don't run volumeicon).

Fallast, 2021-12-13:
https://pastebin.com/UXKX6cz7 (before)
https://pastebin.com/zefXjNeQ (after)

perexg, 2021-12-13:
Thank you. I don't see any problem here. It may be related to your GUI environment or pulseaudio or so...

Does speaker-test work for you? `pasuspender -- speaker-test -D plughw:0` before and after (from command line)?

Fallast, 2021-12-13:
thanks for your quick reply, and yes pasuspender -- speaker-test -D plughw:0 does work before and after! so i guess it must be something else

closed 2021-12-13 by Fallast