alsa-project/alsa-lib #205: Cant't open mixer and no sound , helem (MIXER,'Front Mic Boots Valume',0,1,0) appears twice or more

WolsYang, 2021-12-15:
**I had  try the way in this issue, but not work for me : (**

**helem (MIXER,'Master Playback Volume',0,1,0) appears twice or more cannot load mixer controls: Invalid argument**#47

**In default I only can play sound with Back port**

**Here is my computer audio port look like.**
|codec       | ALC888-VD | ALC888-VD |
|:-------------|:----------------:|:----------------:|
|position    |      Back       |     Front        |
|Function   | 0:Line Out, 1:MIC | 0:Line Out, 1:MIC|
|Port Num  |          2         |          2             |
|HW:dev|0:0|0:2|
|/sys/class/cound|hwC0D0|hwC0D1|

**OS:Ubuntu 20.04.2LTS**
**$ uname -r**
```
5.13.0-051300-generic
```

**$ amixer** 
```
ALSA lib simple_none.c:1544:(simple_add1) helem (MIXER,'Front Mic Boots Valume',0,1,0) appears twice or more
amixer: Mixer default load error: Invalid aggument
``` 
 
**$ aplay -l**
```
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC888-VD Analog [ALC888-VD Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 2: ALC888-VD Analog [ALC888-VD Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**$ pactl list modules**
```
Card #0
	Name: alsa_card.pci-0000_00_1b.0
	Driver: module-alsa-card.c
	Owner Module: 7
	Properties:
		alsa.card = "0"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0x81410000 irq 132"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1b.0"
		sysfs.path = "/devices/pci0000:00/0000:00:1b.0/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "2284"
		device.product.name = "Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series High Definition Audio Controller"
		device.form_factor = "internal"
		device.string = "0"
		device.description = "Built-in Audio"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	Profiles:
		input:analog-stereo: Analog Stereo Input (sinks: 0, sources: 1, priority: 65, available: yes)
		output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 6500, available: yes)
		output:analog-stereo+input:analog-stereo: Analog Stereo Duplex (sinks: 1, sources: 1, priority: 6565, available: yes)
		output:analog-surround-40: Analog Surround 4.0 Output (sinks: 1, sources: 0, priority: 1200, available: yes)
		output:analog-surround-40+input:analog-stereo: Analog Surround 4.0 Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 1265, available: yes)
		output:hdmi-stereo: Digital Stereo (HDMI) Output (sinks: 1, sources: 0, priority: 5900, available: yes)
		output:hdmi-stereo+input:analog-stereo: Digital Stereo (HDMI) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 5965, available: yes)
		output:hdmi-surround: Digital Surround 5.1 (HDMI) Output (sinks: 1, sources: 0, priority: 800, available: yes)
		output:hdmi-surround+input:analog-stereo: Digital Surround 5.1 (HDMI) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 865, available: yes)
		output:hdmi-surround71: Digital Surround 7.1 (HDMI) Output (sinks: 1, sources: 0, priority: 800, available: yes)
		output:hdmi-surround71+input:analog-stereo: Digital Surround 7.1 (HDMI) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 865, available: yes)
		output:hdmi-stereo-extra1: Digital Stereo (HDMI 2) Output (sinks: 1, sources: 0, priority: 5700, available: yes)
		output:hdmi-stereo-extra1+input:analog-stereo: Digital Stereo (HDMI 2) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 5765, available: yes)
		output:hdmi-surround-extra1: Digital Surround 5.1 (HDMI 2) Output (sinks: 1, sources: 0, priority: 600, available: yes)
		output:hdmi-surround-extra1+input:analog-stereo: Digital Surround 5.1 (HDMI 2) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 665, available: yes)
		output:hdmi-surround71-extra1: Digital Surround 7.1 (HDMI 2) Output (sinks: 1, sources: 0, priority: 600, available: yes)
		output:hdmi-surround71-extra1+input:analog-stereo: Digital Surround 7.1 (HDMI 2) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 665, available: yes)
		output:hdmi-stereo-extra2: Digital Stereo (HDMI 3) Output (sinks: 1, sources: 0, priority: 5700, available: yes)
		output:hdmi-stereo-extra2+input:analog-stereo: Digital Stereo (HDMI 3) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 5765, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: output:analog-stereo
```

perexg, 2023-01-30:
I think it's a kernel issue. The RFC patch:

https://lore.kernel.org/alsa-devel/20230130082118.3267980-1-perex@perex.cz/

perexg, 2023-02-01:
Final proposed patch: https://lore.kernel.org/alsa-devel/20230131094215.3545993-1-perex@perex.cz/

It would be nice, if you can test this one.

WolsYang, 2023-02-02:
> Final proposed patch: https://lore.kernel.org/alsa-devel/20230131094215.3545993-1-perex@perex.cz/
> 
> It would be nice, if you can test this one.

Thank you perexg !!

Unfortunately, I don't have that machine now.

If I have any new updates, I'll that you know immediately.

perexg, 2023-02-03:
@FlanOfHarmony ?

FlanOfHarmony, 2023-02-03:
Thank you for such a quick patch!

To test it I need to build a version of the kernel with your patch right? I never compiled and installed a custom kernel before, so I’ll look into it and take my chance this week-end!

FlanOfHarmony, 2023-03-25:
So I kinda forgot to perform the test and I apologize, but I just tested on my `6.1.18-200` kernel (I guess your commit is merged in this one?) and I can now lauch alsamixer without the error. Some bars were quite low and now that I’ve put them to 100%, I can now enjoy a way more decent sound. The issue should be considered fixed, but I was wondering how do I switch from the ALC1220 sound card at the back to the front one?

perexg, 2023-05-02:
This hadware may require a custom UCM configuration - https://github.com/alsa-project/alsa-ucm-conf/tree/master/ucm2/HDA/DualCodecs . It would be better to discuss things there. Don't forget to attach output from `alsainfo.sh --no-upload` .

closed 2023-05-02 by perexg

WolsYang, 2024-02-29:
> Unfortunately, I don't have that machine now.
> 
> If I have any new updates, I'll that you know immediately.

I have the same machine now.

But no luck, still got same error.

```
wols@wols:~$ uname -a
Linux wols 6.5.0-21-generic #21~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb  9 13:32:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

```
wols@wols:~$ sudo dmesg | grep snd
[sudo] password for wols: 
[   14.716855] snd_hda_intel 0000:00:1b.0: Applying patch firmware 'fixDual'
[   15.227749] snd_hda_intel 0000:00:1b.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   15.436575] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC888-VD: line_outs=2 (0x1b/0x14/0x0/0x0/0x0) type:line
[   15.436598] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   15.436607] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   15.436613] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   15.436620] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   15.436626] snd_hda_codec_realtek hdaudioC0D0:      Front Mic=0x19
[   15.436633] snd_hda_codec_realtek hdaudioC0D0:      Rear Mic=0x18
[   15.509226] snd_hda_codec_realtek hdaudioC0D1: autoconfig for ALC888-VD: line_outs=2 (0x1b/0x14/0x0/0x0/0x0) type:line
[   15.509250] snd_hda_codec_realtek hdaudioC0D1:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   15.509257] snd_hda_codec_realtek hdaudioC0D1:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   15.509264] snd_hda_codec_realtek hdaudioC0D1:    mono: mono_out=0x0
[   15.509268] snd_hda_codec_realtek hdaudioC0D1:    inputs:
[   15.509274] snd_hda_codec_realtek hdaudioC0D1:      Front Mic=0x19
[   15.509280] snd_hda_codec_realtek hdaudioC0D1:      Rear Mic=0x18
[   15.540398] snd_hda_intel 0000:00:1b.0: control 2:0:0:Front Playback Volume:0 is already present
[   15.540504] snd_hda_intel 0000:00:1b.0: control 2:0:0:Front Playback Switch:0 is already present
[   15.540572] snd_hda_intel 0000:00:1b.0: control 2:0:0:Surround Playback Volume:0 is already present
[   15.540638] snd_hda_intel 0000:00:1b.0: control 2:0:0:Surround Playback Switch:0 is already present
[   15.540702] snd_hda_intel 0000:00:1b.0: control 2:0:0:Loopback Mixing:0 is already present
[   15.540763] snd_hda_intel 0000:00:1b.0: control 2:0:0:Front Mic Playback Volume:0 is already present
[   15.540828] snd_hda_intel 0000:00:1b.0: control 2:0:0:Front Mic Playback Switch:0 is already present
[   15.540912] snd_hda_intel 0000:00:1b.0: control 2:0:0:Rear Mic Playback Volume:0 is already present
[   15.540977] snd_hda_intel 0000:00:1b.0: control 2:0:0:Rear Mic Playback Switch:0 is already present
[   15.541043] snd_hda_intel 0000:00:1b.0: control 2:0:0:Capture Volume:0 is already present
[   15.541105] snd_hda_intel 0000:00:1b.0: control 2:0:0:Capture Switch:0 is already present
[   15.541165] snd_hda_intel 0000:00:1b.0: control 2:0:0:Front Mic Boost Volume:0 is already present
[   15.541228] snd_hda_intel 0000:00:1b.0: control 2:0:0:Rear Mic Boost Volume:0 is already present
```

```
cat /boot/config-$(uname -r) |grep SND_HDA_CTL_DEV_ID
# CONFIG_SND_HDA_CTL_DEV_ID is not set

cat /etc/modprobe.d/alsa-base.conf | tail -1
options snd-hda-intel ctl_dev_id=1
```

```
wols@wols:~$ pactl list cards 
Card #0
	Name: alsa_card.pci-0000_00_1b.0
	Driver: module-alsa-card.c
	Owner Module: 7
	Properties:
		alsa.card = "0"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0x81610000 irq 134"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1b.0"
		sysfs.path = "/devices/pci0000:00/0000:00:1b.0/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
[... 6 lines cut ...]
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	Profiles:
		input:analog-stereo: Analog Stereo Input (sinks: 0, sources: 1, priority: 65, available: yes)
		output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 6500, available: yes)
		output:analog-stereo+input:analog-stereo: Analog Stereo Duplex (sinks: 1, sources: 1, priority: 6565, available: yes)
		output:analog-surround-40: Analog Surround 4.0 Output (sinks: 1, sources: 0, priority: 1200, available: yes)
		output:analog-surround-40+input:analog-stereo: Analog Surround 4.0 Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 1265, available: yes)
		output:hdmi-stereo: Digital Stereo (HDMI) Output (sinks: 1, sources: 0, priority: 5900, available: yes)
		output:hdmi-stereo+input:analog-stereo: Digital Stereo (HDMI) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 5965, available: yes)
		output:hdmi-stereo-extra1: Digital Stereo (HDMI 2) Output (sinks: 1, sources: 0, priority: 5700, available: yes)
		output:hdmi-stereo-extra1+input:analog-stereo: Digital Stereo (HDMI 2) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 5765, available: yes)
		output:hdmi-surround-extra1: Digital Surround 5.1 (HDMI 2) Output (sinks: 1, sources: 0, priority: 600, available: yes)
		output:hdmi-surround-extra1+input:analog-stereo: Digital Surround 5.1 (HDMI 2) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 665, available: yes)
		output:hdmi-surround71-extra1: Digital Surround 7.1 (HDMI 2) Output (sinks: 1, sources: 0, priority: 600, available: yes)
		output:hdmi-surround71-extra1+input:analog-stereo: Digital Surround 7.1 (HDMI 2) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 665, available: yes)
[... cut ...]
		output:hdmi-surround-extra2+input:analog-stereo: Digital Surround 5.1 (HDMI 3) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 665, available: yes)
		output:hdmi-surround71-extra2: Digital Surround 7.1 (HDMI 3) Output (sinks: 1, sources: 0, priority: 600, available: yes)
		output:hdmi-surround71-extra2+input:analog-stereo: Digital Surround 7.1 (HDMI 3) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 665, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: output:analog-stereo+input:analog-stereo
```

FlanOfHarmony, 2024-02-29:
Well the ticket got closed as completed so I didn't bother to post here, but with the latest updates (no idea which ones, I just realized by ramdomly testing after a very long time passed by where I didn't retry) I can now output the sound easily to the front jack by selecting the right sound profile and the right audio output after that, without creating any custom file or anything. Which distro are you using?

WolsYang, 2024-03-01:
> Well the ticket got closed as completed so I didn't bother to post here, but with the latest updates (no idea which ones, I just realized by ramdomly testing after a very long time passed by where I [...]

I am using Ubuntu 22.04 which kernel is 6.5.0-21-generic

FlanOfHarmony, 2024-03-01:
I’m using fedora with kernel 6.7.6. In my audio settings if I select the "Pro audio" audio profile, I can then select "Built-in Audio 4" which outputs to the front jack.

If I do `pactl list cards`, it finds successfully two cards (`#40` and `#41`), and for each of these cards I have a bigger list of profiles that you, in particular I have some "pro audio" profiles.

I don’t have the technical skill to help you further. Maybe some of the technical people maintaining the repo can find the diff between our two configs.