alsa-project/alsa-lib #209: HDMI devices lumped together and confusingly named

felagund, 2022-01-09:
I first reported this to Pipewire: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1978
but was told this is more ALSA's issue.

The problem is that ALSA treats all devices connected through HDMI/DP/USB-C as one device with different profiles. So in my setup, I can have (they dynamically come and go as I move my laptop around):
1. Monitor A with loudspeakers, attached through USB-C
2. Monitor B with a 3.5mm jack, attached through USB-C
3.  AVR connected through a HDMI port.

However, up from ALSA to PulseAudio or Pipewire, they are treated as distinct "profiles" of one device/card. For example:

```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: CX8070 Analog [CX8070 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 11: HDMI 5 [HDMI 5]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
lists them all on card 1. This maps very poorly onto the real world, where the monitors/AVR are separate devices (with names, see below). This makes switching between them not easy.

Another issue is that the devices are named by their ports (“ NVidia [HDA NVidia], device 9: HDMI 3 [HDMI 3]“) - but as a user, I do not really know which of my USB-C ports is HDMI and which is HDMI 2. But I know I want the sound to play through monitor A, not monitor B. I am not sure if it is ALSA's job to fish out the actualy names of the devices from EDID data, but it would be great. See the bug report linked above for more information and screenshots of the issues. If it is agreed this is something that should be addressed, I can copy it over here.

felagund, 2022-05-04:
Yes, naming them sensibly would be a great improvement!

perexg, 2022-05-05:
I've tried to implement this feature in 859448f01033b40feebf9a0aab467ba57a655b5b . Thanks for the idea.

closed 2022-05-05 by perexg

felagund, 2022-08-21:
Hm, would I need anything but alsa-utils and alsa-lib at 1.2.7 to make this work? I am stuck on Ubuntu with 1.2.6 for next two years so I got the tarballs, installed them with `./configure, make, make install` and `aplay -v` reports 1.2.7, but when I do `aplay -l`, I still get

```
*** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: CX8070 Analog [CX8070 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Do I need newer kernel as well, or would everything need to be rebuild?

AlynxZhou, 2023-05-09:
I am sorry to bother everyone here, but I have some questions related to HDMI devices:

Currently I have two DELL S2721DGF monitors, and I connect my speaker to one of them, however I cannot find which device is the one with speaker. Sometimes HDMI 1 is the one with speaker, but maybe after some system upgrading HDMI 2 is the one with speaker (I don't re-plug any cable and don't tweak screen order, they just changed).

After searching I find that pipewire just use device name from alsa, so I want to ask is there a fixed sequence of HDMI/DP port on GPU (for example the most left is always the 1st)? Or by default ALSA just name them randomly? Is there a way to attach fixed number to ports?

I also checks ELD feature here, but it does not help me, because both of my monitors show the same product name, and I think ELD contains no info about product serial.