alsa-project/alsa-lib #215: Sound automatically switch from USB Speaker to HeadSet

modulusmath, 2022-03-13:
Ubuntu 21.10 / Codename: impish

Running LXDE if it matters

I watch youtube listening on my external usb speakers and sounds plays just fine.

All of a sudden they go silent and sound is played on the plugged in headset (Plantronics, Inc. DA40 Adapter) without my intervention...Even if I unplug the headset it will switch over to the 'default usb audio analog stereo'. 

I do see some logs via journalctl near the time of the switch:

Mar 01 10:04:57 wilddog pulseaudio[1678]: ALSA woke us up to write new data to the device, but there was actually nothing to write.
Mar 01 10:04:57 wilddog pulseaudio[1678]: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
Mar 01 10:04:57 wilddog pulseaudio[1678]: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.

This is a pretty fresh install.

How can I keep it stable on the external speakers?

Here's my also info: http://alsa-project.org/db/?f=3589356420b912e19608eaee1960f19b3f0a4e22

perexg, 2022-03-14:
It looks like a driver problem. Do you see anything USB driver related in the dmesg output ?

modulusmath2c, 2022-03-14:
Thanks for the reply. 

Yes, quite a few. I didn't install anything per se. It's all default ubuntu FWIW:

A bunch of these:

[ 1211.136810] usb 3-4.4: Device not responding to setup address.
[ 1211.344676] usb 3-4.4: device not accepting address 15, error -71

Here' s the full output:

$ sudo dmesg | grep -i usb

[    0.523334] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.523504] ehci-pci 0000:00:1a.0: new USB bus registered, assigned bus number 1
[    0.542699] ehci-pci 0000:00:1a.0: USB 2.0 started, EHCI 1.00
[    0.542750] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.13
[    0.542753] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.542754] usb usb1: Product: EHCI Host Controller
[    0.542755] usb usb1: Manufacturer: Linux 5.13.0-35-generic ehci_hcd
[    0.542756] usb usb1: SerialNumber: 0000:00:1a.0
[    0.542891] hub 1-0:1.0: USB hub found
[    0.543164] ehci-pci 0000:00:1d.0: new USB bus registered, assigned bus number 2
[    0.562697] ehci-pci 0000:00:1d.0: USB 2.0 started, EHCI 1.00
[    0.562743] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.13
[    0.562745] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.562746] usb usb2: Product: EHCI Host Controller
[    0.562747] usb usb2: Manufacturer: Linux 5.13.0-35-generic ehci_hcd
[    0.562748] usb usb2: SerialNumber: 0000:00:1d.0
[    0.562883] hub 2-0:1.0: USB hub found
[    0.563054] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.563075] uhci_hcd: USB Universal Host Controller Interface driver
[    0.785881] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 3
[    0.787104] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.13
[    0.787107] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.787109] usb usb3: Product: xHCI Host Controller
[    0.787110] usb usb3: Manufacturer: Linux 5.13.0-35-generic xhci-hcd
[    0.787112] usb usb3: SerialNumber: 0000:00:14.0
[    0.787261] hub 3-0:1.0: USB hub found
[    0.789131] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 4
[    0.789134] xhci_hcd 0000:00:14.0: Host supports USB 3.0 SuperSpeed
[    0.789165] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.13
[    0.789166] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.789167] usb usb4: Product: xHCI Host Controller
[    0.789168] usb usb4: Manufacturer: Linux 5.13.0-35-generic xhci-hcd
[    0.789169] usb usb4: SerialNumber: 0000:00:14.0
[    0.789252] hub 4-0:1.0: USB hub found
[    0.882708] usb 1-1: new high-speed USB device number 2 using ehci-pci
[    0.898708] usb 2-1: new high-speed USB device number 2 using ehci-pci
[    1.039017] usb 1-1: New USB device found, idVendor=8087, idProduct=8009, bcdDevice= 0.00
[    1.039020] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.039155] hub 1-1:1.0: USB hub found
[    1.055155] usb 2-1: New USB device found, idVendor=8087, idProduct=8001, bcdDevice= 0.00
[    1.055169] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.055452] hub 2-1:1.0: USB hub found
[    1.122709] usb 3-1: new full-speed USB device number 2 using xhci_hcd
[    1.273460] usb 3-1: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=12.11
[    1.273473] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.273475] usb 3-1: Product: USB Receiver
[    1.273476] usb 3-1: Manufacturer: Logitech
[    1.406709] usb 3-3: new full-speed USB device number 3 using xhci_hcd
[    1.556090] usb 3-3: New USB device found, idVendor=0a67, idProduct=11c0, bcdDevice= 0.01
[    1.556094] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.556096] usb 3-3: Product: USB Audio Device
[    1.556098] usb 3-3: Manufacturer: JinAudio
[    1.556099] usb 3-3: SerialNumber: 20190801_M0_V001
[    1.682710] usb 3-4: new high-speed USB device number 4 using xhci_hcd
[    1.845300] usb 3-4: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 1.33
[    1.845315] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.845316] usb 3-4: Product: 4-Port USB 2.0 Hub
[... 293 lines cut ...]
[    5.553332] ppdev: user-space parallel port driver
[    5.872287] usbcore: registered new interface driver usblp
[    6.008245] usbcore: registered new interface driver uvcvideo
[    7.695051] Generic FE-GE Realtek PHY r8169-0-300:00: attached PHY driver (mii_bus:phy_addr=r8169-0-300:00, irq=MAC)
[    8.955790] usbcore: registered new interface driver snd-usb-audio
[ 1209.476604] Generic FE-GE Realtek PHY r8169-0-300:00: attached PHY driver (mii_bus:phy_addr=r8169-0-300:00, irq=MAC)
[18878.800178] usbcore: registered new interface driver usb-storage
[18878.801558] usbcore: registered new interface driver uas
[22319.414573] Generic FE-GE Realtek PHY r8169-0-300:00: attached PHY driver (mii_bus:phy_addr=r8169-0-300:00, irq=MAC)
[22327.412437] ntfs: driver 2.1.32 [Flags: R/O MODULE].
[173179.349826] Generic FE-GE Realtek PHY r8169-0-300:00: attached PHY driver (mii_bus:phy_addr=r8169-0-300:00, irq=MAC)
[201212.901009] sd 6:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK
[230538.032147] Generic FE-GE Realtek PHY r8169-0-300:00: attached PHY driver (mii_bus:phy_addr=r8169-0-300:00, irq=MAC)
[... cut ...]
[    6.080705] input: HDA Intel PCH Front Mic as /devices/pci0000:00/0000:00:1b.0/sound/card2/input23
[    6.080749] input: HDA Intel PCH Rear Mic as /devices/pci0000:00/0000:00:1b.0/sound/card2/input24
[    6.080806] input: HDA Intel PCH Line as /devices/pci0000:00/0000:00:1b.0/sound/card2/input25
[    6.080864] input: HDA Intel PCH Line Out as /devices/pci0000:00/0000:00:1b.0/sound/card2/input26
[    6.080923] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1b.0/sound/card2/input27

modulusmath2c, 2022-03-14:
OK - interesting - I do have quite a few usb device plugged , maybe that's a problem. I did have some issue booting w/usb, maybe this is related... Thanks for your thoughts on the matter!

modulusmath2c, 2022-03-15:
Just to update.  When I unplug and plug the speakers back in they no longer register in dmesg. And no lo longer play sound...I guess if I waited a day or 2 I would have at least had that answer. Bought a new usb speaker, will see how it goes...

modulusmath2c, 2022-03-19:
I have not seen the problem since I replaced the hardware. I really thank you for the help @perexg - From my perspective the reported issue  appears to have been due to a failing USB speaker.

closed 2022-05-03 by perexg