alsa-project/alsa-lib #216: USB audio not detected with kernels 5.15.02 and 5.16.0; snd_usb_audio and snd_usbmidi_lib not loaded.

stevesr0, 2022-03-14:
System running antiX linux Sid version (based on antiX-19).

Using Pipewire on top of ALSA, both internal card and usb headset detected and play with kernel 5.10.88 (antiX nosystemd kernel).

With Debian 5.15.5 and 5.16.0 kernels, the usb card is not noted in aplay -l or with /proc/asound/cards.

Comparison of loaded modules for working kernel and non working kernel: 
      --Working kernel has both snd_usb_audio and snd_usbmidi_lib loaded.
      --Non working kernel has NEITHER of these.  Instead usbcore and usb_common are loaded.

I am seeking advice about fixing the nonworking kernels.  Not sure if this is a bug in the kernels, but both nonworking kernels have identical outputs in /proc/asound/cards and pactl list cards.

Thanks.

Happy to provide files as suggested; not sure what data would be useful.

Steve Rinsler

perexg, 2022-03-14:
Could you post output from dmesg and lsusb (URL only) from the latest kernel? It looks like a driver issue.

stevesr0, 2022-03-14:
Hi perexg,
Attached is a file with both.  I am unclear what you mean by URL only.  The whole dmesg is included.  It was easy to search for instances of usb.  I didn't see anything that gives me a clue.
I should note that today I realized that this might have to do with the udev setup in antiX, which uses eudev.

It may be that the problem is with "regular" kernels that are not customized for antiX.  I wrote to the lead developer today about this, but I haven't heard from him.  If that is the case, there is not necessarily any issue.
Steve Rinsler

[dmesg and lsusb 5161.txt](https://github.com/alsa-project/alsa-lib/files/8248805/dmesg.and.lsusb.5161.txt)

closed 2022-05-03 by perexg

stevesr0, 2022-10-11:
Hi again,
After my reply to your question, I tried loading the modules missing on the non “working” kernels.  They load fine, but that did not enable alas to detect the usb attached headset.
I suspect I have to find a way to either load them at boot or force reloading of alsa in order to see if this enables the detection of the usb headset.  I am not very knowledgeable about this.  The command sudo alsa force-reload was not recognized and I don’t recall if I tried “sudo service alsa stop” and then “sudo alsa service restart” after loading the snd_usb_audio module.
If everything is detected and plays after I accomplish that, the question will be why with these kernels, the necessary modules aren’t automatically loaded.
Steve Rinsler