alsa-project/alsa-lib #219: Line-in is abnormally loud for snd-intel-hda

mpeter50, 2022-03-22:
When I'm on Windows, I sometimes use the line-in Jack to connect my phone to it, so I can either listen to it's sounds along with my PCs sounds or use it as a microphone.

I've tried to set this functionality up on my Linux installation, but it seems something is not right.
The most basic example is with `arecord`: after setting line-in as the default input device with `alsamixer` I've ran `arecord -vv -fdat foo.wav` so I can check what the ALSA drivers actually "hear" without possible interference of any audio servers. While the command was running, I changed the volume of line-in using he appropriate slider of pavucontrol.
When playing the recording, it is clear that while line-in is at 100%, the sound is so loud that it is distorted, and the normal volume is around 12-15% of the actual volume.

I don't experience this on Windows, so this is probably not a hardware issue.
If I plug a headphone directly into my phone and listen to it, the volume is normal, so this is probably not an issue on the signal producer side either.
Checking with `alsamixer`, all capture controls are at 0 volume. But, the controls there might not tell the full truth for multiple reasons:
- after reducing the Capture control's volume I can still hear the sound coming from line-in
- the Capture control's volume is not in sync with what I see in pavucontrol and other places like KDE's volume menu
- the controls Master and PCM in the Playback category does not seem to have any effect. As such, none of the controls control the actual output to my headphone
- the "default" card has a single slider, it is named Master and this actually controls the master volume, at least for my headphone

Screenshot relevant to the 1st and 2nd points:

[image]

Additional information:
- output of the alsa-info script: [alsa-info.txt](https://github.com/alsa-project/alsa-lib/files/8327703/alsa-info.txt)
- Content of `/proc/asound/card*/codec#*` files:
  - [/proc/asound/card0/codec#0](https://github.com/alsa-project/alsa-lib/files/8327730/card0_codec0.txt)
  - [/proc/asound/card1/codec#0](https://github.com/alsa-project/alsa-lib/files/8327735/card1_codec0.txt) 
  - note: card0 is the HDMI section of my graphics card. I never use it for audio, only jack and bluetooth, so it can probably be ignored
- output of `lspci -vvnn`: [lspci.txt](https://github.com/alsa-project/alsa-lib/files/8327777/lspci.txt)

perexg, 2022-05-03:
The driver issues are tracked at https://bugzilla.kernel.org . Also note that the aplay/arecord is redirected to pulseaudio by default, so the tests should use only ALSA subsystem (plughw: or hw: PCM devices for aplay/arecord and alsamixer or amixer tools to modify the mixer settings).

closed 2022-05-03 by perexg

perexg, 2022-05-04:
Example: `arecord -D plughw:0 -f dat a.wav` (list your cards and devices using `arecord -l`)

The mixer should be set using `alsamixer` or `amixer`.

Note if pulseaudio / pipewire blocks your device, try to suspend the device usage - see `man pasuspender`.