alsa-project/alsa-lib #220: Cannot run amixer due to: helem (MIXER,'Mic Boost Volume',0,1,0) appears twice or more

0jrp0, 2022-04-02:
After a recent update to Debian, I am receiving the following error:

```shell
$ amixer
ALSA lib simple_none.c:1555:(simple_add1) helem (MIXER,'Mic Boost Volume',0,1,0) appears twice or more
amixer: Mixer default load error: Invalid argument
```

I attempted to remove `/var/lib/alsa/asound.state`, but it didn't help. If I run `alsactl restore`, it does nothing. If I run `alsactl store`, it just dumps the duplicate Mic Boost Volume entry again.

I have an HDMI device that I don't use where this is possibly coming from. I attempted to blacklist the kernel module for `snd_hda_codec_hdmi`, but it didn't help (nor could I figure out how to persist the black list across reboots even after modifying `/etc/modprobe.d/`).

My current setup is using `pipewire` using the `pipewire-pulse` binary which connects up to `alsa`.

Some info:

```
$ pactl list short sinks
89      alsa_output.pci-0000_00_1f.3.pro-output-0       PipeWire        s32le 2ch 48000Hz       SUSPENDED
90      alsa_output.pci-0000_00_1f.3.pro-output-4       PipeWire        s32le 2ch 48000Hz       SUSPENDED
111     bluez_output.04_52_C7_B0_B3_08.a2dp-sink        PipeWire        s16le 2ch 48000Hz       SUSPENDED
```

```
$ sudo dmesg |grep snd
[   24.786073] snd_hda_intel 0000:00:1f.3: enabling device (0140 -> 0142)
[   24.786637] snd_hda_intel 0000:65:00.1: enabling device (0140 -> 0142)
[   24.786746] snd_hda_intel 0000:65:00.1: Disabling MSI
[   24.786790] snd_hda_intel 0000:65:00.1: Handle vga_switcheroo audio client
[   24.814739] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:line
[   24.814745] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   24.814748] snd_hda_codec_generic hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   24.814751] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
[   24.814752] snd_hda_codec_generic hdaudioC0D0:    inputs:
[   24.814754] snd_hda_codec_generic hdaudioC0D0:      Mic=0x18
[   24.814756] snd_hda_codec_generic hdaudioC0D0:      Line=0x1a
[   24.832905] snd_hda_codec_generic hdaudioC0D1: autoconfig for Generic: line_outs=1 (0x1b/0x0/0x0/0x0/0x0) type:speaker
[   24.832912] snd_hda_codec_generic hdaudioC0D1:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   24.832914] snd_hda_codec_generic hdaudioC0D1:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   24.832916] snd_hda_codec_generic hdaudioC0D1:    mono: mono_out=0x0
[   24.832917] snd_hda_codec_generic hdaudioC0D1:    inputs:
[   24.832918] snd_hda_codec_generic hdaudioC0D1:      Mic=0x19
[   24.837834] snd_hda_intel 0000:00:1f.3: control 2:0:0:Capture Volume:0 is already present
[   24.837840] snd_hda_intel 0000:00:1f.3: control 2:0:0:Capture Switch:0 is already present
[   24.837843] snd_hda_intel 0000:00:1f.3: control 2:0:0:Mic Boost Volume:0 is already present
[   24.966257] usbcore: registered new interface driver snd-usb-audio
```

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: Generic Analog [Generic Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 4: Generic Analog [Generic Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: NVidia [HDA NVidia], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
$ lspci -v
...
00:1f.3 Audio device: Intel Corporation 200 Series PCH HD Audio
        Subsystem: Lenovo 200 Series PCH HD Audio
        Flags: bus master, fast devsel, latency 32, IRQ 65, NUMA node 0, IOMMU group 21
        Memory at 43ffff20000 (64-bit, non-prefetchable) [size=16K]
        Memory at 43ffff00000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: <access denied>
        Kernel driver in use: snd_hda_intel
        Kernel modules: snd_hda_intel
...
```

```
$ inxi -SAxc 0
System:
  Host: {redacted} Kernel: 5.15.15-{redacted}-amd64 x86_64
    bits: 64 compiler: gcc v: 11.2.0 Desktop: awesome 4.3
    Distro: Debian GNU/Linux 12 
Audio:
  Device-1: Intel 200 Series PCH HD Audio vendor: Lenovo
    driver: snd_hda_intel v: kernel bus-ID: 00:1f.3
  Device-2: NVIDIA GP107GL High Definition Audio driver: snd_hda_intel
    v: kernel bus-ID: 65:00.1
  Device-3: Logitech Webcam C930e type: USB driver: snd-usb-audio,uvcvideo
    bus-ID: 1-14:9
  Sound Server-1: ALSA v: k5.15.15-1rodete2-amd64 running: yes
  Sound Server-2: PulseAudio v: 15.0 running: no
  Sound Server-3: PipeWire v: 0.3.47 running: yes
```

```
$ head -n 1 /proc/asound/card0/codec*
==> /proc/asound/card0/codec#0 <==
Codec: Realtek Generic

==> /proc/asound/card0/codec#1 <==
Codec: Realtek Generic
```

```
$ /bin/cat /etc/modprobe.d/alsa-base.conf 
options snd-hda-intel model=generic
```

```
$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Direct hardware device without any conversions
hw:CARD=PCH,DEV=4
    HDA Intel PCH, Generic Analog
    Direct hardware device without any conversions
plughw:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Hardware device with all software conversions
plughw:CARD=PCH,DEV=4
    HDA Intel PCH, Generic Analog
    Hardware device with all software conversions
default:CARD=PCH
    HDA Intel PCH, Generic Analog
    Default Audio Device
sysdefault:CARD=PCH
    HDA Intel PCH, Generic Analog
    Default Audio Device
front:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Front output / input
surround21:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
dmix:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Direct sample mixing device
dmix:CARD=PCH,DEV=4
    HDA Intel PCH, Generic Analog
    Direct sample mixing device
usbstream:CARD=PCH
    HDA Intel PCH
    USB Stream Output
usbstream:CARD=C930e
    Logitech Webcam C930e
    USB Stream Output
hw:CARD=NVidia,DEV=3
    HDA NVidia, HDMI 0
    Direct hardware device without any conversions
hw:CARD=NVidia,DEV=7
    HDA NVidia, HDMI 1
    Direct hardware device without any conversions
hw:CARD=NVidia,DEV=8
    HDA NVidia, HDMI 2
    Direct hardware device without any conversions
hw:CARD=NVidia,DEV=9
    HDA NVidia, HDMI 3
    Direct hardware device without any conversions
plughw:CARD=NVidia,DEV=3
    HDA NVidia, HDMI 0
    Hardware device with all software conversions
plughw:CARD=NVidia,DEV=7
    HDA NVidia, HDMI 1
    Hardware device with all software conversions
plughw:CARD=NVidia,DEV=8
    HDA NVidia, HDMI 2
    Hardware device with all software conversions
plughw:CARD=NVidia,DEV=9
    HDA NVidia, HDMI 3
    Hardware device with all software conversions
hdmi:CARD=NVidia,DEV=0
    HDA NVidia, HDMI 0
    HDMI Audio Output
hdmi:CARD=NVidia,DEV=1
    HDA NVidia, HDMI 1
    HDMI Audio Output
hdmi:CARD=NVidia,DEV=2
    HDA NVidia, HDMI 2
    HDMI Audio Output
hdmi:CARD=NVidia,DEV=3
    HDA NVidia, HDMI 3
    HDMI Audio Output
dmix:CARD=NVidia,DEV=3
    HDA NVidia, HDMI 0
    Direct sample mixing device
dmix:CARD=NVidia,DEV=7
    HDA NVidia, HDMI 1
    Direct sample mixing device
dmix:CARD=NVidia,DEV=8
    HDA NVidia, HDMI 2
    Direct sample mixing device
dmix:CARD=NVidia,DEV=9
    HDA NVidia, HDMI 3
    Direct sample mixing device
usbstream:CARD=NVidia
    HDA NVidia
    USB Stream Output
```

geraldog, 2022-04-03:
> After a recent update to Debian, I am receiving the following error:
[...]
> options snd-hda-intel model=generic

Hi @0jrp0 could you try setting the model=auto and reboot?

0jrp0, 2022-04-05:
Magical. That fixed it.

Furthermore, I also installed wireplumber using pipewire and things are back to normal for me:
https://wiki.debian.org/PipeWire#Debian_Testing.2FUnstable

Other changes I made:

```
$ touch /etc/pipewire/media-session.d/with-pulseaudio
$ touch /etc/pipewire/media-session.d/with-alsa
```

Then reboot.

closed 2022-04-05 by 0jrp0

geraldog, 2022-04-06:
> Magical. That fixed it.

Great! Glad it worked for you.

It may not fix every use case though, apparently it is related to https://bugzilla.kernel.org/show_bug.cgi?id=195305 and may need even more magic to work with, if you actually have two snd-hda-intel identical boards. I think...

> Furthermore, I also installed wireplumber using pipewire and things are back to normal for me: https://wiki.debian.org/PipeWire#Debian_Testing.2FUnstable
>
> Other changes I made:
>
> ```
> $ touch [...]