alsa-project/alsa-lib #221: Play jitter coming from dshare

davidboulangerlatour, 2022-04-28:
I have an application which plays a wav file (using dshare) at with a specific period (ie. every 720ms). This always worked well up to recently after upgrading alsa-lib from 1.1.7 to 1.2.4. With 1.2.4, the cycles look like this [750ms, 750ms, 750ms, 625ms], normally it would be [720ms, 720ms, ...].

When doing a bisect to find the issue, I found this commit to be the cause:
```
7265e603bf880a9ec2cd01c3cf2afbd7709e5af4 is the first bad commit
commit 7265e603bf880a9ec2cd01c3cf2afbd7709e5af4
Author: Vanitha Channaiah <vanitha.channaiah@in.bosch.com>
Date:   Wed May 15 11:56:33 2019 +0530

    pcm: dshare: Added "hw_ptr_alignment" option in configuration for alignment of slave pointers
    
    This change adapt the fix commit 6b058fda9dce
    ("pcm: dmix: Add option to allow alignment of slave pointers")
    for dshare plugin
    
    Issue is that snd_pcm_wait() goes back to waiting because the hw_ptr
    is not period aligned. Therefore snd_pcm_wait() will block for a longer
    time as required.
    
    With these rcar driver changes the exact position of the dma is returned.
    During snd_pcm_start they read hw_ptr as reference, and this hw_ptr
    is now not period aligned, and is a little ahead over the period while it
    is read. Therefore when the avail is calculated during snd_pcm_wait(),
    it is missing the avail_min by a few frames.
    
    An additional option hw_ptr_alignment is provided to dshare configuration,
    to allow the user to configure the slave application and hw pointer
    alignment at startup
    
    Signed-off-by: Vanitha Channaiah <vanitha.channaiah@in.bosch.com>
    Signed-off-by: Takashi Iwai <tiwai@suse.de>

 src/pcm/pcm_dshare.c | 40 +++++++++++++++++++++++++++++++++++-----
 1 file changed, 35 insertions(+), 5 deletions(-)
```

I tried the more recent 1.2.6 with the same issue. I also tried all the "hw_ptr_alignment" options with no success.

perexg, 2022-05-03:
If you set `hw_ptr_alignment` to `no` (or `off` with 3f108fc5b85880fc696ce70fbcc55437bd32dab6), then the new code is turned off (previous behaviour). Could you show your config ?

Could you dump the pcm state using `snd_pcm_dump()` function for the further analysis to get the `auto` code working? Example: https://github.com/alsa-project/alsa-utils/blob/904bc1dfd4a97002e1bb05058cfb15a7f486d493/speaker-test/speaker-test.c#L1200

perexg, 2023-03-31:
Stale.

closed 2023-03-31 by perexg