alsa-project/alsa-lib #239: Regression: Upgrade to 1.2.7 Causes No HDMI Audio Device on AMD Trinity APU

lovenemesis, 2022-06-17:
After upgrading to alsa-lib 1.2.7, HDMI Audio from AMD A10-5800K "Trinity" series APU stopped working. HDMI Audio is no longer presented as a Output Device in GNOME Setting - Audio. aplay -l shows error for HDMI device: 

`aplay: device_list:304: control digital audio info (0): Input/output error`

**Downgrading to alsa-lib to 1.2.6** restore the HDMI audio device on A10-5800K system. 

Below are some system info I collected when running alsa-lib-1.2.6 and alsa-lib-1.2.7

[alsa-lib-1.2.6.txt](https://github.com/alsa-project/alsa-lib/files/8923675/alsa-lib-1.2.6.txt)

[alsa-lib-1.2.7.txt](https://github.com/alsa-project/alsa-lib/files/8923676/alsa-lib-1.2.7.txt)

It appears to me that kernel, pipewire, wireplumber versions have no impact on the occurrence of the issue.

It is also reported in [RedHat Bugzilla](https://bugzilla.redhat.com/show_bug.cgi?id=2097574)

perexg, 2022-06-17:
Dup of #235 .

perexg, 2022-06-17:
Version 1.2.7.1 was released with this fix.

closed 2022-06-17 by perexg

lovenemesis, 2022-06-21:
Upgraded to 1.2.7.1 and confirmed the fix.