alsa-project/alsa-lib #240: alsa-lib parser error: "sequence command 'devdisall' is ignored" when attempting alsactl restore, nrestore, init, etc.

salperinlea, 2022-06-17:
This may play into a much larger issue I have been having with an Intel tiger lake soundcard, which cannot be activated after my machine has been suspended, but which always loads on boot.

In an attempt to solve this error, I have tried to use alsactl restore, at which point I get the error:
<code><pre>
sudo alsactl restore
alsa-lib parser.c:1194:(parse_sequence) error: sequence command 'devdisall' is ignored
alsa-lib parser.c:1194:(parse_sequence) error: sequence command 'devdisall' is ignored
</code></pre>
I believe the reason that this repeats is that I have two soundcards; one is an Nvidia card, hard-wired to a dGPU and the HDMI port.

Info:
Arch Linux 251.2, Kernel 5.18.3, KDE 5.25 DE.

<code><pre>
lspci | grep 'Audio'
0000:00:1f.3 Multimedia audio controller: Intel Corporation Tiger Lake-H HD Audio Controller (rev 11)
0000:01:00.1 Audio device: NVIDIA Corporation Device 228e (rev a1)
</code></pre>

I am not sure what other info may be of use, but I will provide it upon request.

dominiksr, 2022-06-18:
I have the same errors. Usually everything works. Sometimes after restarting the sound card doesn't work and only turning the laptop off again helps.

OS: Fedora Linux 36 (Workstation Edition) x86_64 
Kernel: 5.18.5-200.fc36.x86_64
CPU: 11th Gen Intel i5-11300H

perexg, 2022-06-19:
Already fixed in https://github.com/alsa-project/alsa-ucm-conf/commit/c802ca0b7d8186413b728382f8310359afa6a144 .

closed 2022-06-19 by perexg

allhailwesttexas, 2022-06-20:
That's great, is it possible to push out a release with this fix?