alsa-project/alsa-lib #243: Alsa-to-Jack bridge broken in alsa* 1.2.7.x

hlekin, 2022-06-19:
I am using a setup according to https://alsa.opensrc.org/Jack_and_Loopback_device_as_Alsa-to-Jack_bridge

Since the upgrade of alsa-lib, alsa-ucm-conf, and alsa-utils to version 1.2.7, my alsa-to-jack-bridge setup is broken:
$ jack_connect system:capture_1 ploop:playback_1
ERROR ploop:playback_1 not a valid port
A downgrade to versions 1.2.6 resolves the issue. Versions 1.2.7.1 result in the same error.

The relevant output of qjackctl 0.9.7-2 > Messages is:
Cannot write socket fd = 25 err = Broken pipe
CheckRes error
Could not write notification
ClientNotify fails name = cloop notification = 0 val1 = 0 val2 = 0
NotifyAddClient old_client fails name = ploop
Cannot write socket fd = 28 err = Broken pipe
CheckRes error
Could not write notification
ClientNotify fails name = ploop notification = 1 val1 = 0 val2 = 0
ERROR ploop:playback_1 not a valid port
ERROR ploop:playback_2 not a valid port
ERROR cloop:capture_1 not a valid port
ERROR cloop:capture_2 not a valid port

Please find attached alsa-info.sh output for alsa-versions 1.2.7.1 and 1.2.6
[alsa-info.sh-output-220619.zip](https://github.com/alsa-project/alsa-lib/files/8935020/alsa-info.sh-output-220619.zip)
.
I am aware, that this is a very poor bug report. Please ask for any information you need!

By the way, many thanks for all your programs and the effort you put in to them.

perexg, 2022-06-19:
The port names are different in alsa-plugins-1.2.7.1 . List the jack port names and use the new one.

hlekin, 2022-06-19:
$ jack_lsp
system:capture_1
system:capture_2
system:playback_1
system:playback_2
ploop:playback_1
ploop:playback_2
cloop:capture_1
cloop:capture_2
Calf Studio Gear:Automation MIDI In
Calf Studio Gear:Equalizer 8 Band In #1
Calf Studio Gear:Equalizer 8 Band In #2
Calf Studio Gear:Equalizer 8 Band Out #1
Calf Studio Gear:Equalizer 8 Band Out #2

# upgrade alsa packages
Package (6)                  Old Version  New Version  Net Change
extra/alsa-lib               1.2.6.1-1    1.2.7.1-1      0,03 MiB
extra/alsa-plugins           1:1.2.6-3    1:1.2.7.1-1    0,00 MiB
extra/alsa-ucm-conf          1.2.6.3-1    1.2.7.1-1      0,04 MiB
extra/alsa-utils             1.2.6-1      1.2.7-1        0,05 MiB
multilib/lib32-alsa-lib      1.2.6.1-1    1.2.7.1-1      0,01 MiB
multilib/lib32-alsa-plugins  1.2.6-1      1.2.7.1-1      0,00 MiB

# qjackctl > Start
$ jack_lsp
system:capture_1
system:capture_2
system:playback_1
system:playback_2
Calf Studio Gear:Automation MIDI In
Calf Studio Gear:Equalizer 8 Band In #1
Calf Studio Gear:Equalizer 8 Band In #2
Calf Studio Gear:Equalizer 8 Band Out #1
Calf Studio Gear:Equalizer 8 Band Out #2

There are no new jack port names...

perexg, 2022-06-19:
The loop ports are missing after upgrade. Are `alsa_out` and `alsa_in` tools running?

hlekin, 2022-06-19:
How can I verify that alsa_out and alsa_in tools  are running?

hlekin, 2022-06-19:
This is from the first line in my loop2jack script:
$ /usr/bin/alsa_out -j ploop -dploop
Unable to get buffer size back: Invalid argument
Setting of hwparams failed: Invalid argument

perexg, 2022-06-19:
Could you try to bisect the commit which makes this utility fail? You may use LD_PRELOAD to load the test library without installing it. Example (assuming that the alsa-lib sources are in `~/git/alsa-lib` tree):

```
LD_PRELOAD="~/git/alsa-lib/src/.libs/libasound.so.2" /usr/bin/alsa_out -j ploop -dploop
```

Thanks.

hlekin, 2022-06-19:
Yes, I can try.
Which test library exactly? If I check out the master brach and compile it, I will get what I have already got, v1.2.7.1?

perexg, 2022-06-19:
There is v1.2.7.1 tag (which actually matches to master branch - f91c89da999cf4f9445ada0206ef4b7c25255cdf).

hlekin, 2022-06-20:
The package from my distribution, which I used here, is already alsa-lib 1.2.7.1-1 (s. a.).