alsa-project/alsa-lib #245: HDA Nvidia no profile

decepticlown, 2022-06-28:
I have a TV connected through `HDMI` to laptop which has `NVidia 1650ti` card. In `pavucontrol` I can not see any `HDA NVidia` profiles. It only shows `off`. I can solve it by `downgrading to alsa-lib 1.2.4`. After that HDMI profiles are shown properly and sound works. Issue persists with both proprietary and open source NVidia drivers. 

pipewire version: 0.3.52-2
pipewire-alsa version: 0.3.52-2
pipewire-pulse version: 0.3.52-2
wireplumber version: 0.4.10-3
alsa-lib version: 1.2.7-1
Archlinux kernel version: 5.18.7

I haven't done any configuration for above packages. Its basically fresh install.
<details>
<summary> $ pactl list cards</summary>

    Card #44
	Name: alsa_card.pci-0000_01_00.1
	Driver: alsa
	Owner Module: n/a
	Properties:
		api.acp.auto-port = "false"
		api.acp.auto-profile = "false"
		api.alsa.card = "0"
		api.alsa.card.longname = "HDA NVidia at 0xfc080000 irq 99"
		api.alsa.card.name = "HDA NVidia"
		api.alsa.path = "hw:0"
		api.alsa.use-acp = "true"
		api.dbus.ReserveDevice1 = "Audio0"
		device.api = "alsa"
		device.bus = "pci"
		device.bus_path = "pci-0000:01:00.1"
		device.description = "HDA NVidia"
		device.enum.api = "udev"
		device.icon_name = "audio-card-analog-pci"
		device.name = "alsa_card.pci-0000_01_00.1"
		device.nick = "HDA NVidia"
		device.plugged.usec = "3396199"
		device.product.id = "4346"
		device.subsystem = "sound"
		device.sysfs.path = "/sys/devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0"
		device.vendor.id = "4318"
		device.vendor.name = "NVIDIA Corporation"
		media.class = "Audio/Device"
		factory.id = "14"
		client.id = "34"
		object.id = "42"
		object.serial = "44"
		object.path = "alsa:pcm:0"
		alsa.card = "0"
		alsa.card_name = "HDA NVidia"
		alsa.long_card_name = "HDA NVidia at 0xfc080000 irq 99"
		alsa.driver_name = "snd_hda_intel"
		device.string = "0"
	Profiles:
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: off

    Card #45
	Name: alsa_card.pci-0000_06_00.6
	Driver: alsa
	Owner Module: n/a
	Properties:
		api.acp.auto-port = "false"
		api.acp.auto-profile = "false"
		api.alsa.card = "1"
		api.alsa.card.longname = "HD-Audio Generic at 0xfc5c0000 irq 100"
		api.alsa.card.name = "HD-Audio Generic"
		api.alsa.path = "hw:1"
		api.alsa.use-acp = "true"
		api.dbus.ReserveDevice1 = "Audio1"
		device.api = "alsa"
		device.bus = "pci"
[... 272 lines cut ...]
	Jun 28 15:44:26 acpomen systemd[752]: Stopped PipeWire Multimedia Service.
	Jun 28 15:44:26 acpomen systemd[752]: pipewire.service: Consumed 49.188s CPU time.
	-- Boot 9c1e5f7301a64e9180bd6a9bf6cb3ec6 --
	Jun 28 15:45:13 acpomen systemd[696]: Started PipeWire Multimedia Service.
	Jun 28 16:05:17 acpomen systemd[696]: Stopping PipeWire Multimedia Service...
	Jun 28 16:05:17 acpomen systemd[696]: Stopped PipeWire Multimedia Service.
	-- Boot 619883556f584c45b36b0072399b3621 --
	Jun 28 16:05:38 acpomen systemd[678]: Started PipeWire Multimedia Service.
	Jun 28 16:07:51 acpomen systemd[678]: Stopping PipeWire Multimedia Service...
	Jun 28 16:07:51 acpomen systemd[678]: Stopped PipeWire Multimedia Service.
	Jun 28 16:07:51 acpomen systemd[678]: Started PipeWire Multimedia Service.
	Jun 28 16:13:04 acpomen systemd[678]: Stopping PipeWire Multimedia Service...
	Jun 28 16:13:04 acpomen systemd[678]: Stopped PipeWire Multimedia Service.
	-- Boot 017f208f301c4e3787ee98c5827a39eb --
	Jun 28 16:13:22 acpomen systemd[682]: Started PipeWire Multimedia Service.
	Jun 28 16:16:30 acpomen systemd[682]: Stopping PipeWire Multimedia Service...
	Jun 28 16:16:30 acpomen systemd[682]: Stopped PipeWire Multimedia Service.
	-- Boot b9c9a99d501148028e7db65ec457fd0a --
	Jun 28 16:16:50 acpomen systemd[665]: Started PipeWire Multimedia Service.
	Jun 28 16:34:10 acpomen systemd[665]: Stopping PipeWire Multimedia Service...
	Jun 28 16:34:10 acpomen systemd[665]: Stopped PipeWire Multimedia Service.
	Jun 28 16:43:22 acpomen systemd[665]: Started PipeWire Multimedia Service.
</details>

Please, let me know if you need any relevent logs. I'm not sure which ones to post.

perexg, 2022-06-28:
Could you check which `hdmi:X,Y` and `hw:X,Y` devices are probed in PW with alsa-lib 1.2.4 and the recent alsa-lib ?

The 'Device or resource busy' errors are probably the culprit, why things do not work.

decepticlown, 2022-06-28:
Could you please tell me how I should check that?

perexg, 2022-06-28:
```
systemctl --user stop pipewire.socket
spa-acp-tool -vvv -c 2 list-ports
```

Replace 2 with the alsa card number (you can obtain it using `aplay -l` for example).

Start pipewire again:

```
systemctl --user start pipewire.socket
```

decepticlown, 2022-06-28:
#### alsa-lib 1.2.7-1
<details>
<summary> Card0 (NVidia) probes </summary>

	UCM available for card hw:0
	UCM _alibpref=_ucm0001.
	Set UCM verb to HiFi
	Got PlaybackPCM for device HDMI6: _ucm0001.hw:NVidia,11
	Got PlaybackCTL for device HDMI6: _ucm0001.hw:NVidia
	Got PlaybackPriority for device HDMI6: 1600
	Got CaptureCTL for device HDMI6: _ucm0001.hw:NVidia
	Got TQ for device HDMI6: HiFi
	Got JackControl for device HDMI6: HDMI/DP,pcm=11 Jack
	UCM file does not specify 'PlaybackChannels' for device HDMI6, assuming stereo.
	No _conflictingdevs for device HDMI6
	No _supporteddevs for device HDMI6
	Got PlaybackPCM for device HDMI5: _ucm0001.hw:NVidia,10
	Got PlaybackCTL for device HDMI5: _ucm0001.hw:NVidia
	Got PlaybackPriority for device HDMI5: 1500
	Got CaptureCTL for device HDMI5: _ucm0001.hw:NVidia
	Got TQ for device HDMI5: HiFi
	Got JackControl for device HDMI5: HDMI/DP,pcm=10 Jack
	UCM file does not specify 'PlaybackChannels' for device HDMI5, assuming stereo.
	No _conflictingdevs for device HDMI5
	No _supporteddevs for device HDMI5
	Got PlaybackPCM for device HDMI4: _ucm0001.hw:NVidia,9
	Got PlaybackCTL for device HDMI4: _ucm0001.hw:NVidia
	Got PlaybackPriority for device HDMI4: 1400
	Got CaptureCTL for device HDMI4: _ucm0001.hw:NVidia
	Got TQ for device HDMI4: HiFi
	Got JackControl for device HDMI4: HDMI/DP,pcm=9 Jack
	UCM file does not specify 'PlaybackChannels' for device HDMI4, assuming stereo.
	No _conflictingdevs for device HDMI4
	No _supporteddevs for device HDMI4
	Got PlaybackPCM for device HDMI3: _ucm0001.hw:NVidia,8
	Got PlaybackCTL for device HDMI3: _ucm0001.hw:NVidia
	Got PlaybackPriority for device HDMI3: 1300
	Got CaptureCTL for device HDMI3: _ucm0001.hw:NVidia
	Got TQ for device HDMI3: HiFi
	Got JackControl for device HDMI3: HDMI/DP,pcm=8 Jack
	UCM file does not specify 'PlaybackChannels' for device HDMI3, assuming stereo.
	No _conflictingdevs for device HDMI3
	No _supporteddevs for device HDMI3
	Got PlaybackPCM for device HDMI2: _ucm0001.hw:NVidia,7
	Got PlaybackCTL for device HDMI2: _ucm0001.hw:NVidia
	Got PlaybackPriority for device HDMI2: 1200
	Got CaptureCTL for device HDMI2: _ucm0001.hw:NVidia
	Got TQ for device HDMI2: HiFi
	Got JackControl for device HDMI2: HDMI/DP,pcm=7 Jack
	UCM file does not specify 'PlaybackChannels' for device HDMI2, assuming stereo.
	No _conflictingdevs for device HDMI2
	No _supporteddevs for device HDMI2
	Got PlaybackPCM for device HDMI1: _ucm0001.hw:NVidia,3
	Got PlaybackCTL for device HDMI1: _ucm0001.hw:NVidia
	Got PlaybackPriority for device HDMI1: 1100
	Got CaptureCTL for device HDMI1: _ucm0001.hw:NVidia
	Got TQ for device HDMI1: HiFi
	Got JackControl for device HDMI1: HDMI/DP,pcm=3 Jack
	UCM file does not specify 'PlaybackChannels' for device HDMI1, assuming stereo.
	No _conflictingdevs for device HDMI1
	No _supporteddevs for device HDMI1
	Found UCM profiles
	UCM mapping: HiFi: hw:NVidia,11: sink dev HDMI6
	UCM mapping: HiFi: hw:NVidia,10: sink dev HDMI5
	UCM mapping: HiFi: hw:NVidia,9: sink dev HDMI4
	UCM mapping: HiFi: hw:NVidia,8: sink dev HDMI3
	UCM mapping: HiFi: hw:NVidia,7: sink dev HDMI2
	UCM mapping: HiFi: hw:NVidia,3: sink dev HDMI1
	Profile HiFi (Play HiFi quality Music), input=null, output=null priority=8000, supported=yes n_input_mappings=0, n_output_mappings=6
	Output HiFi: hw:NVidia,11: sink
[... 2324 lines cut ...]
	Skipping profile output:hdmi-surround71+input:analog-surround-41 - will not be able to open output:hdmi-surround71
	Skipping profile output:hdmi-surround71+input:analog-surround-50 - will not be able to open output:hdmi-surround71
	Skipping profile output:hdmi-surround71+input:analog-surround-51 - will not be able to open output:hdmi-surround71
	Skipping profile output:hdmi-surround71+input:analog-surround-71 - will not be able to open output:hdmi-surround71
	Skipping profile output:hdmi-surround71+input:iec958-stereo - will not be able to open output:hdmi-surround71
	Looking at profile output:hdmi-dts-surround
	Checking for playback on Digital Surround 5.1 (HDMI/DTS) (hdmi-dts-surround)
	Trying dcahdmi:1 with SND_PCM_NO_AUTO_FORMAT ...
	Unknown PCM dcahdmi:1
	Error opening PCM device dcahdmi:1: No such file or directory
	Caching failure to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:analog-stereo - will not be able to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:analog-surround-21 - will not be able to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:analog-surround-40 - will not be able to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:analog-surround-41 - will not be able to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:analog-surround-50 - will not be able to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:analog-surround-51 - will not be able to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:analog-surround-71 - will not be able to open output:hdmi-dts-surround
	Skipping profile output:hdmi-dts-surround+input:iec958-stereo - will not be able to open output:hdmi-dts-surround
	Looking at profile output:hdmi-stereo-extra1
	Checking for playback on Digital Stereo (HDMI 2) (hdmi-stereo-extra1)
	Trying hdmi:1,1 with SND_PCM_NO_AUTO_FORMAT ...
	Error opening PCM device hdmi:1,1: No such file or directory
	Caching failure to open output:hdmi-stereo-extra1
	Skipping profile output:hdmi-stereo-extra1+input:analog-stereo - will not be able to open output:hdmi-stereo-ex

decepticlown, 2022-06-28:
I'm sorry I don't really understand what that means. What I'd like to know is whether the fix for this issue is in the scope of pipewire or alsa-libs development? Or is it one those rare cases where issue is related to minor category of devices or instances where fixing it isn't deemed worthwhile?

Thank you for helping me so far. I'm also following the new issue you opened for pipewire.