alsa-project/alsa-lib #248: How to get mixer control index from UCM

Zames-Chang, 2022-07-07:
I use  Corsair VOID ELITE USB Gaming Headset. However this headset have two same name control
```
$ amixer -c 1 scontrols
Simple mixer control 'Auto Gain Control',0
Simple mixer control 'Headset',0
Simple mixer control 'Headset',1
```
use command to change 'Headset',0 is actually not working. But change 'Headset',1 will work. I want write a UCM to indicate that when seeing this headset, choose the name 'Headset' with index 1 not index 0. 

Does UCM and alsa api able to do so?

perexg, 2022-07-07:
```
PlaybackMixerElem "Headset,1"
```

or

```
PlaybackMixerElem "'Headset',1"
```

or

```
PlaybackMixerElem '"Headset",1'
```

Zames-Chang, 2022-07-07:
Many Thanks

closed 2022-07-07 by perexg