alsa-project/alsa-lib #252: snd_hda_intel - HDMI missing channels / wrong mapped / no passthru

Nightwing0815, 2022-07-14:
Hey Guys,

I have since a few days trouble to get sound configured...more problems than on windows...
I ran the alsa-info.sh script, here's the output: [https://alsa-project.org/db/?f=a95b27a60a9e38120a1d3b121ff4a93d89b1ac09](https://alsa-project.org/db/?f=a95b27a60a9e38120a1d3b121ff4a93d89b1ac09)

I only want to configure the usuals:

- card1 is microphone / headset / stereo on laptop @ full duplex,
- card 0 should be HDMI stereo, HDMI digital passthru AC3 / DTS bitstreaming.

I hope, you can help me with my configuration, I'm confused.

Thanks in advance,

ToM

Nightwing0815, 2023-01-10:
update, maybe helpful :-)

perexg, 2023-01-29:
It does not appear like alsa-lib issue. Please, continue in the SOF project.

closed 2023-01-29 by perexg