alsa-project/alsa-lib #254: Audio interface unable to be loaded in 1.2.7.2

KevHg, 2022-07-17:
I upgraded from 1.2.6.1 to 1.2.7.2 and found that pulseaudio no longer detects my Motu M2 interface.

Main error log seems to be:
`module-alsa-card.c: Failed to find a working profile.`
`failed to load module module-alsa-card`

I'm new to debugging Linux issues, let me know what other info I can provide to help out on this topic.

ember-rose, 2022-07-17:
I don't know if it's the same issue, but my MOTU M2 started causing segfaults in PipeWire on boot and when it's reconnected. It was last working on 1.2.7.1 for me. 

OS: Fedora 36 SilverBlue x86_64
kernel: 5.18.11-200
alsa-lib: 1.2.7.2-1
pipewire: 0.3.55-2
wireplumber: 0.4.11-2

I've attached the core dump summaries below, let me know if I can help diagnose!

<details>
<summary>pipewire</summary>

```
           PID: 1994 (pipewire)
           UID: 1000 (ember)
           GID: 1000 (ember)
        Signal: 11 (SEGV)
     Timestamp: Sat 2022-07-16 23:49:35 EDT (1h 52min ago)
  Command Line: /usr/bin/pipewire
    Executable: /usr/bin/pipewire
 Control Group: /user.slice/user-1000.slice/user@1000.service/session.slice/pipewire.service
          Unit: user@1000.service
     User Unit: pipewire.service
         Slice: user-1000.slice
     Owner UID: 1000 (ember)
       Boot ID: ae4a258ce55948a482ed6b97022a2d93
    Machine ID: ****
      Hostname: ****
       Storage: /var/lib/systemd/coredump/core.pipewire.1000.ae4a258ce55948a482ed6b97022a2d93.1994.1658029775000000.zst (present)
     Disk Size: 558.7K
       Message: Process 1994 (pipewire) of user 1000 dumped core.
                
                Module linux-vdso.so.1 with build-id a5a2fc4f28eb11c41f67d73441080a265c6df3fe
                Module libspa-audiomixer.so with build-id eef62fd635c0f3d9f4e4675bf8dafca1b3504581
                Metadata for module libspa-audiomixer.so owned by FDO found: {
                	"type" : "rpm",
                	"name" : "pipewire",
                	"version" : "0.3.55-2.fc36",
                	"architecture" : "x86_64",
                	"osCpe" : "cpe:/o:fedoraproject:fedora:36"
                }
                
                Module libudev.so.1 with build-id 002cce667c466b2f92130c99f55d21afa5669636
                Metadata for module libudev.so.1 owned by FDO found: {
                	"type" : "rpm",
                	"name" : "systemd",
                	"version" : "250.8-1.fc36",
                	"architecture" : "x86_64",
                	"osCpe" : "cpe:/o:fedoraproject:fedora:36"
                }
                
                Module libspa-alsa.so with build-id 53e4ea71491882c3bfd9cb9482e13b97f65a9f82
                Metadata for module libspa-alsa.so owned by FDO found: {
                	"type" : "rpm",
                	"name" : "pipewire",
                	"version" : "0.3.55-2.fc36",
                	"architecture" : "x86_64",
                	"osCpe" : "cpe:/o:fedoraproject:fedora:36"
                }
                
                Module libspa-audioconvert.so with build-id 3f23ac8ad9382061de133cef9d3ee068f01b7f8b
                Metadata for module libspa-audioconvert.so owned by FDO found: {
                	"type" : "rpm",
                	"name" : "pipewire",
                	"version" : "0.3.55-2.fc36",
                	"architecture" : "x86_64",
                	"osCpe" : "cpe:/o:fedoraproject:fedora:36"
                }
                
[... 185 lines cut ...]
                #2  0x00007f8e19919361 loop_iterate (libspa-support.so + 0x7361)
                #3  0x00007f8e27521447 do_loop (libpipewire-0.3.so.0 + 0x42447)
                #4  0x00007f8e27349e2d start_thread (libc.so.6 + 0x8ce2d)
                #5  0x00007f8e273ce7d4 __clone (libc.so.6 + 0x1117d4)
                
                Stack trace of thread 2025:
                #0  0x00007f8e273c2bef __poll (libc.so.6 + 0x105bef)
                #1  0x00007f8e2766923d g_main_context_iterate.constprop.0 (libglib-2.0.so.0 + 0xaa23d)
                #2  0x00007f8e27611940 g_main_context_iteration (libglib-2.0.so.0 + 0x52940)
                #3  0x00007f8e27611991 glib_worker_main (libglib-2.0.so.0 + 0x52991)
                #4  0x00007f8e2763e302 g_thread_proxy (libglib-2.0.so.0 + 0x7f302)
                #5  0x00007f8e27349e2d start_thread (libc.so.6 + 0x8ce2d)
                #6  0x00007f8e273ce7d4 __clone (libc.so.6 + 0x1117d4)
                
                Stack trace of thread 2035:
                #0  0x00007f8e273c2bef __poll (libc.so.6 + 0x105bef)
                #1  0x00007f8e2766923d g_main_context_iterate.constprop.0 (libglib-2.0.so.0 + 0xaa23d)
                #2  0x00007f8e276136cf g_main_loop_run (libglib-2.0.so.0 + 0x546cf)
                #3  0x00007f8e271edada gdbus_shared_thread_func.lto_priv.0 (libgio-2.0.so.0 + 0x114ada)
                #4  0x00007f8e2763e302 g_thread_proxy (libglib-2.0.so.0 + 0x7f302)
                #5  0x00007f8e27349e2d start_thread (libc.so.6 + 0x8ce2d)
                #6  0x00007f8e273ce7d4 __clone (libc.so.6 + 0x1117d4)
                ELF object binary architecture: AMD x86-64
```
</details>

ebluke, 2022-07-17:
Having the same issue with 1.2.7.7-1 from the arch linux repos. Downgrading alsa-lib, alsa-ucm-conf, and lib32-alsa-lib to 1.2.7.1-1, and also alsa-card-profiles to 1:0.3.54-1 was the fix for me. Also new to debugging in linux, but happy to upgrade again to post the output of any inputs you would like :)

ember-rose, 2022-07-17:
I suspected this might be a UCM issue, so I tried rolling [alsa-lib](https://packages.fedoraproject.org/pkgs/alsa-lib/alsa-lib/) forward to 1.2.7.2 while holding [alsa-ucm](https://packages.fedoraproject.org/pkgs/alsa-lib/alsa-ucm/) back to 1.2.7.1 and my M2 still worked. 

Next, I looked for a culprit UCM commit between 1.2.7.1 and 1.2.7.2 and found applying https://github.com/alsa-project/alsa-ucm-conf/commit/2cc5a68e9798bc3226e2349fad5d3faa886368c1 locally caused the PipeWire segfault to start occurring again.

sreich, 2022-07-17:
Hey! I've got the same issue with my Motu M2, after upgrade to 1.2.7.2, on arch

Downgraded `downgrade --cached-only alsa-ucm-conf=1.2.7.1-1 alsa-lib=1.2.7.1-1 lib32-alsa-lib=1.2.7.1-1` to at least get my audio back.