alsa-project/alsa-lib #257: ALSA: Setting FIFO type as the slave for dsnoop

cnkoder, 2022-08-10:
**Problem**

I am facing problems in using "dsnoop" to share/divide the audio stream into two. When dsnoop plugins' slave is set to be a FIFO it throws an error.

executed:

    sudo arecord -D default -f cd defRecording.wav -c 1 -r 32000

error:

    ALSA lib pcm_direct.c:1809:(_snd_pcm_direct_get_slave_ipc_offset) Invalid type 'fifo' for slave PCM

    arecord: main:828: audio open error: Invalid argument

This is the current asound.conf settings

asound.conf

    pcm.!default {
        type asym
        playback.pcm "playback"
        capture.pcm "capture1 "
    }

    pcm.playback {
        type plug
        slave.pcm "eci"
    }

    # Stream Output 1: Final
    pcm.capture1 {
        type plug
        slave.pcm "array"
    }

    # Stream Output 2: Final
    pcm.capture2 {
        type plug
        slave.pcm "array"
    }

    # Used to share the record device
    pcm.array {
        type dsnoop
        slave {
            pcm "eco"
            channels 1
        }
        ipc_key 666666
    }

    # Writes audio coming from any sort of player to ec.input, this is read by the echo
    # cancellation software.
    pcm.eci {
        type plug
        slave {
            format S16_LE
            rate 32000
            channels 1
            pcm {
                type file
                slave.pcm null
                file "/tmp/ec.input"
                format "raw"
            }
        }
    }

    # Read FIFO output which contains echo cancelled audio
    pcm.eco {
        type plug
        slave.pcm {
            type fifo
            infile "/tmp/ec.output"
            rate 32000
            format S16_LE
            channels 1
        }
        #ipc_key 666666
    }

Note:
eco is used to read the FIFO file which contains the echo canceled audio coming in from cancellation software. This software's input is hw:0 and records audio directly from the microphone, and then processes and passes this over to ec.output

Dsnoop works well when the slave.pcm is a hardware device but as soon as I point to something else it fails. 
Is there a workaround or any other solution to tackle this problem?

perexg, 2022-08-10:
> Dsnoop works well when the slave.pcm is a hardware device but as soon as I point to something else it fails.
Is there a workaround or any other solution to tackle this problem?

`dsnoop`, `dshare`, `dmix` plugins works only with `hw:` plugins (direct ALSA devices). Why you cannot use the `eco` pcm device two times directly ?

perexg, 2022-08-10:
Replying to myself - pipe right...

perexg, 2022-08-10:
Pehaps, the used software may write output to two pipes, if possible. Or you may use the ALSA loopback soundcard (snd-aloop kernel module) and `dsnoop` on top of it.

cnkoder, 2022-08-10:
> Pehaps, the used software may write output to two pipes, if possible. Or you may use the ALSA loopback soundcard (`snd-aloop ` kernel module) and `dsnoop` on top of it.

Thank you, will search on the compatibility of dsnoop with snd-aloop and try this out first and then maybe move the first option.
Will update as soon as I get some results

cnkoder, 2022-08-12:
Replying to inform on what was done.
Searched on snd-aloop but I could not achieve what was required, therefore edited the software to write data to two pipes and managed to solve the problem.
Thanks again

closed 2022-08-12 by cnkoder