alsa-project/alsa-lib #261: Channel mapping randomly change on usb soundcard and sucpicious dmesg log

sprudent, 2022-08-17:
Hello erveryone,

I am getting random channel swap when using speaker test to output sequencially a pink noise on every channel of the soundcard.
I tested two distinct usb soundcard getting the same issue.
I could identify a suspicious log showing at the same time the channels swapped.
Tested and reproduced on a recent archlinux image (see system information).

```
[Aug16 18:26] xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 11 comp_code 1
[  +0.000134] xhci_hcd 0000:00:14.0: Looking for event-dma 0000000014ed98e0 trb-start 0000000014ed98f0 trb-end 0000000014ed98f0 seg-start 0000000014ed9000 seg-end 0000000014ed9ff0
```

### system information
```
uname -r

5.18.7-arch1-1
```
```
aplay -l

**** List of PLAYBACK Hardware Devices ****
card 0: Audio [Intel HDMI/DP LPE Audio], device 0: HdmiLpeAudio [Intel HDMI/DP LPE Audi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Audio [Intel HDMI/DP LPE Audio], device 1: HdmiLpeAudio [Intel HDMI/DP LPE Audi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Audio [Intel HDMI/DP LPE Audio], device 2: HdmiLpeAudio [Intel HDMI/DP LPE Audi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: ICUSBAUDIO7D [ICUSBAUDIO7D], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: bytchtes8316 [bytcht-es8316], device 0: Audio (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: bytchtes8316 [bytcht-es8316], device 1: Deep-Buffer Audio (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
[alsa-info.txt](https://github.com/alsa-project/alsa-lib/files/9358483/alsa-info.txt)
### Test with dmix
#### Description
First test was made with alsa dmix plugin. Multiple processes running in parrallel and outputing to dmix pcm device.

#### Alsa configuration
```
pcm.play_8
{
	@args [ CARD ]
	@args.CARD 
	{
		type string
	}

    type dmix
    
    ipc_key 5678
    
    slave
    {
		pcm $CARD
        channels 8
        period_time 0
        period_size 1024
        buffer_size 4096
        rate 44100
	}
    
	bindings {
		0 0
		1 1
		2 2
		3 3
		4 4
		5 5
		6 6
		7 7
	} 
}
```
#### Step to reproduce
Run this script to simulate multiple processes writing to dmix device at the same time
```
for i in $(seq 20)
do
		speaker-test -r44100 -D"play_8:\"hw:1\"" -c8 -t pink &
done
```
Then on another tty, start a speaker-test command outputing a sine wave in order to identify channel mapping by correlating speaker test output with audio output.
`speaker-test -r44100 -D"play_8:\"hw:1\"" -c8 -t sine`
```
 0 - Front Left
 4 - Front Center
 1 - Front Right
 3 - Side Right
 7 - Side Right
 2 - Side Left
 6 - Side Left
 5 - LFE
Time per period = 21.998519
...
```

#### Expected result
Audio output to the same channels during all the test
#### Actual result
Random audio channel swap correlated with dmesg log mentioned above.

### Second test with direct output
#### Alsa configuration
No specific configuration required
#### Step to reproduce
Run speaker-following test command and wait for channel swap
`speaker-test -r44100 -Dsurround71:ICUSBAUDIO7D -c8`
#### Expected result
Audio output to the same channels during all the test
#### Actual result
Random audio channel swap correlated with dmesg log mentioned above. The chanel swap happen far less often than with dmix.

### Interpretation
The DMA transfer ring buffer get messed up after a while (see dmesg log) and interleaved audio frames get shifted in soundcard buffer. If I unplug/replug usb soundcard  and try again, channel mapping goes back to the correct order.

My alsa, kernel and usb driver knowledge is limited, so I come to you for a little help. Tell me if you need more information to reproduce.

Thank's,
Sylvain

perexg, 2022-08-17:
This does not look like an alsa-lib's issue. Please, create a driver bug at https://bugzilla.kernel.org (Audio group). Although this issue may be also USB core related.

closed 2022-08-17 by perexg

sahi1422, 2023-10-20:
@sprudent I am facing the same issue with kernel 5.10
Did you get the fix?

sahi1422, 2023-10-20:
@sprudent , with kernel 4.19 it used to work fine.

frno7, 2024-05-05:
Hello. I’ve got Linux 6.1.72 with this ICUSBAUDIO7D, messing up the channels randomly as described above. My trigger is the `mpv` media player, and the channels rearrange randomly when fast forwarding for example a video with 5.1 surround sound. As mentioned, the physical USB hardware has to be unplugged and replugged to correct the channels. Is there any known fix for this?