alsa-project/alsa-lib #262: snd_hda_intel: Woke up but no data

shmohawk, 2022-08-17:
```
Aug 17 01:11:23 hellokitty pulseaudio[12938]: [alsa-sink-CX20590 Analog] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write.
Aug 17 01:11:23 hellokitty pulseaudio[12938]: [alsa-sink-CX20590 Analog] alsa-sink.c: Most likely this is a bug in the ALSA driver 'snd_hda_intel'. Please report this issue to the ALSA developers.
Aug 17 01:11:23 hellokitty pulseaudio[12938]: [alsa-sink-CX20590 Analog] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
```

It is symptomous with a crackling sound for ~5 seconds after any playback starts.

perexg, 2022-08-18:
Duplicate #90 .

For the crackling sound - it's probably a driver issue. Please, report it here: https://bugzilla.kernel.org (Audio group). Attach output from `alsa-info.sh --no-upload`.

shmohawk, 2022-08-18:
I updated to pulseaudio 16 and the latest alsa-plugins, but it's still crackling. For some reason it does not happen with software using pulseaudio directly, only with alsa-over-pulseaudio, so I'm hesitant to be accepting of this being a kernel driver issue.

perexg, 2022-08-18:
How do you test the pulseaudio -> alsa plugin ? Do things work with `speaker-test` or `aplay` utility ?

shmohawk, 2022-08-18:
cmus is using pulseaudio directly (displaying "C* music player: playback" in pavucontrol), where it does not crackle. 
Google Chrome is showing up in pavucontrol as "ALSA plug-in [chrome]: ALSA playback".

I tested aplay with CantinaBand60.wav from https://www2.cs.uic.edu/~i101/SoundFiles/, displaying similar to Chrome in pavucontrol - without crackling. So this may very well be a problem in the ALSA interface of Google Chrome.

perexg, 2022-08-19:
So the issue is not directly ALSA related. Closing.

closed 2022-08-19 by perexg