alsa-project/alsa-lib #271: Alsa - has invalid interface number

smangels, 2022-09-19:
hi,
I'm running the latest ArchLinux installation using a Behringer Flow8 as my main audio interface. Today I'm running into an issue where the USB audio hardware can not be loaded successfully.

```
[ 1690.797522] usb 1-3.1.1.4: new high-speed USB device number 14 using xhci_hcd
[ 1690.891936] usb 1-3.1.1.4: config 1 has an invalid interface number: 7 but max is 5
[ 1690.891940] usb 1-3.1.1.4: config 1 has no interface number 5
[ 1690.892620] usb 1-3.1.1.4: New USB device found, idVendor=1397, idProduct=050d, bcdDevice= 2.10
[ 1690.892624] usb 1-3.1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[ 1690.892627] usb 1-3.1.1.4: Product: FLOW 8
[ 1690.892628] usb 1-3.1.1.4: Manufacturer: Behringer
[ 1690.892630] usb 1-3.1.1.4: SerialNumber: 18-04-ED-FB-BE-A0
```

smangels, 2022-09-19:
The USB audio interfaces are not shown in Gnome Settings but appear as OFF in PavuControl.

perexg, 2022-09-20:
Those messages are from the kernel. Most likely, you're hitting this kernel bug (kernel 5.19.9):

https://bugzilla.kernel.org/show_bug.cgi?id=216500

smangels, 2022-09-20:
Yes, I can confirm that downgrading to 5.19.8 solved that issue. Thanks for the hint @perexg

closed 2022-09-20 by smangels