alsa-project/alsa-lib #273: ALSA multi plugin error when combining two soundcards

anthonio9, 2022-10-05:
My goal is to mix the output of two sound cards (RAVENNA from [bondagit](https://github.com/bondagit/aes67-linux-daemon.git) and hifiberry), then pass it to a DSP pcm plugin. As it's not possible with `dmix` (dmix only supports output to hw devices) the reasonable next step was `multi` plugin, but my tries always end up in a failure. Below is a quite standard `multi` configuration inspired by alsa docs that I'm using:

```sh
pcm.quad {
    type multi
    slaves.a.pcm "hw:RAVENNA"
    slaves.a.channels 2
    slaves.b.pcm "hw:sndrpihifiberry"
    slaves.b.channels 2

    bindings.0.slave a
    bindings.0.channel 0
    bindings.1.slave a
    bindings.1.channel 1
    bindings.2.slave b
    bindings.2.channel 0
    bindings.3.slave b
    bindings.3.channel 1
    master 1
}

pcm.quad2 {
    type plug
    slave.pcm "quad"
}
```

This is run with `alsaloop` to be input into an extplug plugin that easily accepts 4 channels of audio - `alsadsp`:
```sh
$ alsaloop -C quad -P alsadsp -c 4
Rate 48000Hz not available for capture quad: Invalid argument
Unable to set parameters for capture quad stream: Invalid argument
Loopback start failure.
```
This results in a failure, moreover running `quad2` also results in the same type of error.

Next step was to try plug pcm:

```sh                                                                                                                                
pcm.quad_plug {                                                                                                                              
    type multi                                                                                                                               
    slaves.a.pcm "plughw:RAVENNA"                                                                                                                
    slaves.a.channels 2                                                                                                                      
    slaves.b.pcm "plughw:sndrpihifiberry"                                                                                                        
    slaves.b.channels 2                                                                                                                      
                                                                                                                                             
    bindings.0.slave a                                                                                                                       
    bindings.0.channel 0                                                                                                                     
    bindings.1.slave a                                                                                                                       
    bindings.1.channel 1                                                                                                                     
    bindings.2.slave a                                                                                                                       
    bindings.2.channel 0                                                                                                                     
    bindings.3.slave a                                                                                                                       
    bindings.3.channel 1                                                                                                                     
}

pcm.quad_plug2 {                                                                                                                             
    type plug                                                                                                                                
    slave.pcm "quad_plug"                                                                                                                         
}  
```
With `quad_plug` the same error occurs, with `quad_plug2` the error is as below:

```sh
$ alsaloop -C quad_plug2 -P alsadsp -c 4
Broken configuration for capture quad_plug2 PCM: no configurations available: Invalid argument
Unable to set parameters for capture quad_plug2 stream: Invalid argument
Loopback start failure.
```
Then the last thing I tried was to force the 48kHz sampling rate directly on the hardware devices with the help of the following config:

```sh
pcm.quad_48k {                    
    type multi                    
    slaves.a {                    
        pcm {                     
            type plug     
            slave.pcm "hw:RAVENNA"
            slave.rate 48000      
        }                         
        channels 2                
    }                             
                                  
    slaves.b {                    
        pcm {                     
            type plug             
            slave.pcm "hw:sndrpihifiberry"
            slave.rate 48000              
        }                                 
        channels 2                        
    }                                     
                                          
    bindings.0.slave a            
    bindings.0.channel 0          
    bindings.1.slave a            
    bindings.1.channel 1          
    bindings.2.slave b                    
    bindings.2.channel 0          
    bindings.3.slave b                    
    bindings.3.channel 1                  
}
```
and again it ends up with a failure:

```sh
$ alsaloop -C quad_48k -P alsadsp -c 4
Sample format not available for capture quad_48k: Invalid argument
Unable to set parameters for capture quad_48k stream: Invalid argument
Loopback start failure.
```

I know that both these cards are capable of running 48kHz, I'm doing that all the time. So what could be the cause of this error? `alsadsp` wants them to be 48kHz and if run separately it all works well, both cards set themselves according to `alsadsp` requirements. It works even if there's only one card in the multi plugin that occupies all 4 output ports - my conclusion is that it's not a problem of `alsadsp`.
Are there any other steps that could help solving the issue?

perexg, 2022-10-05:
Sample format issue? Try `-f S32_LE` for alsaloop or check the supported formats by `arecord -Dhw:RAVENNA --dump-hw-params`.

perexg, 2022-10-05:
... and basically it's a wrong idea to combine two clocks (two different cards) with the multi plugin ... The clocks difference will increase in time and the stream won't be usable.

anthonio9, 2022-10-05:
Thanks, it seems to work quite well with the `-f S32_LE` option, but I've only tested it for a minute or so.
Would a sound server (like pipewire) make this combination work better? I know that pipewire does have some tricks, but have no idea what would that turn out to be in the long run.
Is there a way to inspect that clock differece? That would actually help me a lot.