alsa-project/alsa-lib #279: Missing quirk for JBL Quantum 810 (?)

Xenorio, 2022-11-04:
Hi there!

Recently, I received a JBL Quantum 810 headset as a replacement for my Quantum 800 (which worked totally fine until a hardware issue). When I connected my new headset via the provided 2.4Ghz dongle, all I got was microphone input, but no sound output. Taking a look at pavucontrol (I use PipeWire on Fedora), it only has 2 configuration options (other than "off"), which are "Pro Audio" and "Mono Input". With the old one I had a lot more, including "Analog Stereo Duplex", which is what I always used. With "Pro Audio" selected, it registers one input device and 2 output devices (from what I could gather, there are 2 because of the hardware slider used for mixing game and voice chat audio). No matter which of the output devices I use, no sound is coming out, and they don't even have a volume slider.

After some research, my suspicion is that all that needs to be done is assigning an already existing quirk for JBL headsets like what has been done [here](https://lore.kernel.org/alsa-devel/s5h4kb4m3ht.wl-tiwai@suse.de/T/). Although I'm in no way an expert in kernel or driver related matters, so take this with a grain of salt. I have tried tinkering with aliases in ``/etc/modprobe.d/alsa-base.conf`` but I seem to be too inexperienced to get it to do anything at all.

I am able to connect the headset via bluetooth, which works fine, but this results in noticeable quality loss and missing features (for example, the hardware volume slider can't be used on bluetooth connections). For now, I'm connecting it via bluetooth **and** the dongle, setting the bluetooth connection to output only via blueman, then assigning input streams to the mic from the dongle connection and output streams to the bluetooth connection (this way, I'm at least getting full audio quality on the mic). This is definitely not a viable long-term solution though.

And now, all the data:

**Relevant output from lsusb**
```
Bus 001 Device 007: ID 0ecb:2069 Harman International Inc JBL Quantum810 Wireless
```

**uname -r**
```
6.0.5-200.fc36.x86_64
```

**alsa-info.sh** (headset only connected via dongle here)
http://alsa-project.org/db/?f=db549be4f37c3c8617f567b1aaf26fd77c25e030

P.S.: I have done a full system update, tons of reboots, and the firmware on the headset is on the most recent version (installed Windows on a separate machine to get into the Quantum Engine software). In my desperation, I have even installed "alsa-*" via dnf.

geraldog, 2022-11-04:
Perhaps @puleglot can help you, he's the author of the JBL Quantum quirks patch.

Xenorio, 2022-11-05:
Something I just found and thought I should add, right after plugging in the dongle, the kernel log is spamming this stuff:
```
[ 3773.204629] usb 1-5: 1:1: cannot set freq 48000 to ep 0x1
[ 3773.206147] usb 1-5: 1:1: cannot set freq 48000 to ep 0x1
[ 3773.207602] usb 1-5: 1:1: cannot set freq 48000 to ep 0x1
[ 3773.209061] usb 1-5: 1:1: cannot set freq 48000 to ep 0x1
[ 3773.210407] usb 1-5: 1:1: cannot set freq 48000 to ep 0x1
[ 3773.212920] usb 1-5: 3:1: cannot set freq 48000 to ep 0x2
[ 3773.213919] usb 1-5: 3:1: cannot set freq 48000 to ep 0x2
```
This goes on for a lot more lines, always either 0x1 or 0x2 at the end. And right after registering the dongle as a usb device.

joakim-tjernlund, 2022-11-06:
> Something I just found and thought I should add, right after plugging in the dongle, the kernel log is spamming this stuff:
>
> ```
> [ 3773.204629] usb 1-5: 1:1: cannot set freq 48000 to ep 0x1
> [ [...]

That could very well the same as lots of other USB audio devices, try adding QUIRK_FLAG_CTL_MSG_DELAY_1M quirk in kernel, sound/usb/quirks.c for your device.

Xenorio, 2022-11-06:
> That could very well the same as lots of other USB audio devices, try adding QUIRK_FLAG_CTL_MSG_DELAY_1M quirk in kernel, sound/usb/quirks.c for your device.

Sorry, again, I know nothing about kernel related stuff. All I know is how to use a package manager to update it. Are you asking me to modify that file in source and compile the kernel myself? If yes, how do I do that, how do I even add a flag to the device, and how do i get a custom kernel to load without bricking my OS? Also, is this even something that could be related to the original problem?

venzislav, 2022-11-24:
I just bought the same JBL Quantum 810 and I cannot make it work in linux - my best success was I manage to make mic working only. 
Please help !  I am so desperate :(

jradelmo, 2022-12-09:
I have same problem with JBL Quantum 610, connected with usb dongle.

perexg, 2022-12-11:
It looks like a kernel issue. Please, create a new issue in https://bugzilla.kernel.org (Audio group).

closed 2022-12-11 by perexg

Xenorio, 2022-12-11:
For everyone involved, bug has been filed here: https://bugzilla.kernel.org/show_bug.cgi?id=216798

Xenorio, 2022-12-14:
For everyone already here or stumbling upon this via search engine, a solution has been found! For more info, check the bugzilla link above!

joakim-tjernlund, 2022-12-14:
Looked at for the bugzilla briefly and wonder, is there perhaps a new FW for the device you can download/install ?

Xenorio, 2022-12-14:
> Looked at for the bugzilla briefly and wonder, is there perhaps a new FW for the device you can download/install ?

Literally the first thing I did when i got the thing

jonathanglima, 2022-12-31:
Quck update: it looks like that kernel 6.2-rc1 already has code that seems to fix this, such as QUIRK_FLAG_FIXED_RATE in usbaudio.h: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/sound/usb/usbaudio.h?h=v6.2-rc1

joakim-tjernlund, 2023-05-26:
> Hi! apparently it has been fixed but it's not working for me. tried the latest version of linux, linux lqx and linux zen and none of them are working. the only thing that is working is this patch: https://bugzilla.kernel.org/show_bug.cgi?id=216798#c10 but i have to do everything again everytime there's a new kernel update. I'm using arch.

lsusb anc check your USB device ID, you may have another ID

Xenorio, 2023-05-26:
I also use arch (distro-hopped between original issue creation and now). Can definitely confirm that it works fine on kernel 6.3.3 with the Quantum 810 headset. 

If you're using pulseaudio, this **could** be the problem. Have to set the configuration profile (pavucontrol) to "Pro Audio" for everything to work 100%, which according to some reports does not exist on pulse, only on pipewire (which I'm using). If you're already on pipewire, check if you set your profile to that.

A less likely reason could be xorg, as it does interface with your devices. Not sure how much (if at all) it messes with audio stuff tho. I'm on wayland.

Also, try updating the firmware on the headset. You'll need to run the JBL tool for that on a separate Windows machine (I suppose a VM with USB passthrough should also work).

And yes, as the comment by @joakim-tjernlund that just popped up while writing this correctly states, they sometimes change device IDs. So if your headset is pretty much new, this could also be the reason. In that case you'll have to use an alias in your alsa config (can't remember the specific details from the top of my head rn tho)

MopigamesYT, 2023-05-27:
I found what was the problem!! i tried something a long ago and it failed, something in /etc/modprobe.d/usb-audio.conf. deleting it and rebooting it fixed everything!