alsa-project/alsa-lib #280: [snd-dice] latency 13 times higher compared to FFADO (Saffire 14 PRO) with jack2

grinness, 2022-11-05:
Hi,

I hope this is the right place to report this (please let me know if it is not)
On a 5900x, 32 GB RAM, Archlinux (kernel 6.0.2), I have been investigating crazy high latency using a pipewire-jack, see:

https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2790

I ended up testing jack_iodelay with jack2, comparing snd-dice and FFADO

Running qjackctl jack2, ALSA backed (snd-dice) 48000Hz, 1024 samples, 2 periods:

```
> jack_iodelay                                                                                                                                                                                                                                                                                                                           
new capture latency: [0, 0]
new playback latency: [0, 0]
Signal below threshold...
Signal below threshold...
new capture latency: [1024, 1024]
Signal below threshold...
Signal below threshold...
new playback latency: [2048, 2048]
  5198.818 frames    108.309 ms total roundtrip latency
        extra loopback latency: 2126 frames
        use 1063 for the backend arguments -I and -O Inv
  5198.818 frames    108.309 ms total roundtrip latency
        extra loopback latency: 2126 frames
        use 1063 for the backend arguments -I and -O ?? Inv
```
Running qjackctl jack2, firewire backend (FFADO) 4800, 512 samples, 2 periods:
```
> jack_iodelay                                                                                                                                                                                                                                                                                                                                    
new capture latency: [0, 0]
new playback latency: [0, 0]
Signal below threshold...
new capture latency: [1024, 1024]
Signal below threshold...
new playback latency: [3072, 3072]
  4258.811 frames     88.725 ms total roundtrip latency
        extra loopback latency: 162 frames
        use 81 for the backend arguments -I and -O Inv
  4258.812 frames     88.725 ms total roundtrip latency
        extra loopback latency: 162 frames
        use 81 for the backend arguments -I and -O Inv
```
Note that qjackctl seems to not set the number of periods correctly with firewire backend, the above playback latency suggests 3 periods instead of 2.
It seems that Cadence fixes the issue, but the extra loopback latency stays at 162 frames with FFADO:
```
> jack_iodelay                                                                                                                                                                                                                                                                                                                                        
new capture latency: [0, 0]
new playback latency: [0, 0]
Signal below threshold...
Signal below threshold...
Signal below threshold...
Signal below threshold...
Signal below threshold...
Signal below threshold...
new capture latency: [512, 512]
  8354.799 frames    174.058 ms total roundtrip latency
        extra loopback latency: 7842 frames
        use 3921 for the backend arguments -I and -O ?? Inv
  8354.779 frames    174.058 ms total roundtrip latency
        extra loopback latency: 7842 frames
        use 3921 for the backend arguments -I and -O Inv
Signal below threshold...
Signal below threshold...
new playback latency: [1024, 1024]
Signal below threshold...
  1698.777 frames     35.391 ms total roundtrip latency
        extra loopback latency: 162 frames
        use 81 for the backend arguments -I and -O Inv
  1698.777 frames     35.391 ms total roundtrip latency
        extra loopback latency: 162 frames
        use 81 for the backend arguments -I and -O Inv
  1698.777 frames     35.391 ms total roundtrip latency
        extra loopback latency: 162 frames
        use 81 for the backend arguments -I and -O Inv
  1698.777 frames     35.391 ms total roundtrip latency
        extra loopback latency: 162 frames
        use 81 for the backend arguments -I and -O Inv
  1698.777 frames     35.391 ms total roundtrip latency
        extra loopback latency: 162 frames

```

Please note that the pipewire developer pointed out that:

"My problem is that the period size in the driver is set to 1024. For batch devices and a quantum of 256 this should be set to 128 to get decent latency.
So, either the driver does not report itself as batch, or something else started the nodes with a higher quantum. I'm betting it's the batch thing."

I also reported this in the following:
https://github.com/takaswie/snd-firewire-improve/issues/47

thanks

grinness, 2022-11-05:
Hi,

to make sure that the information above were correct, I re-checked the latency of snd-dice via jack2 (Cadence), the latency is smaller than previously reported, but still almost 1 order of magnitude higher than FFADO(same settings-- buffer size 512 and 2 periods/buffers):

```
>jack_iodelay                                                                                                                                                                                                                                                                                                                                        
new capture latency: [0, 0]
new playback latency: [0, 0]
Signal below threshold...
Signal below threshold...
Signal below threshold...
Signal below threshold...
new capture latency: [512, 512]
Signal below threshold...
Signal below threshold...
Signal below threshold...
Signal below threshold...
Signal below threshold...
new playback latency: [1024, 1024]
  2634.732 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
  2634.732 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
  2634.731 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
  2634.731 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
  2634.732 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
  2634.732 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
  2634.731 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
  2634.731 frames     54.890 ms total roundtrip latency
        extra loopback latency: 1098 frames
        use 549 for the backend arguments -I and -O Inv
```

Again, please let me know if this is not the right place to report the issue.

Thanks

takaswie, 2022-11-16:
Let me close the issue since it is not an issue of alsa-lib.