alsa-project/alsa-lib #287: alsa-project.org is down

afq984, 2022-12-09:
```
curl -i https://alsa-project.org
HTTP/2 404 
server: nginx/1.14.1
date: Fri, 09 Dec 2022 16:34:22 GMT
content-type: text/html; charset=UTF-8
x-powered-by: PHP/7.4.30

No input file specified.
```

afq984, 2022-12-09:
Thanks for the update. We were using FTP to download the library in our CI and https://alsa-project.org/files/ works for us.

perexg, 2022-12-09:
Ohh. FTP is back, too.

perexg, 2022-12-11:
Wiki is live again now. Closing.

closed 2022-12-11 by perexg

esramirez, 2022-12-11:
The git proxy is not running. Can some one please help?

perexg, 2022-12-13:
Fixed. Thanks.

esramirez, 2022-12-13:
Perfect! Thank you

forca84, 2024-06-06:
https://git.alsa-project.org
down again :(

AkraAME, 2024-09-02:
Git service is unfortunately down again.

perexg, 2024-09-02:
Service provider status:

"""
2.09.2024 02:07 pm (UTC+2) Disruption in our Nuremberg Data Center
We regret to inform you that an unexpected cooling issue at our Nuremberg Data Center is affecting several services:

- Some servers may be unreachable.
- Placing new orders is currently not possible.
- Accessing the Customer Control Panel is unavailable.
- Object Storage in the EU may be inaccessible.
- The customer support ticketing system is not functioning.

Our technicians are actively working on resolving these issues, but there is currently no estimated time for resolution. We kindly ask that you refrain from creating new tickets during this time. We apologize for the inconvenience and appreciate your patience.
"""