alsa-project/alsa-lib #288: Alsa device

andy155, 2022-12-21:
alsa device disappears completely after a random time usually after about a day

[    2.192815] ALSA device list:
[    2.192818]   #0: HDA ATI SB at 0xfe300000 irq 16
[    2.360481] Freeing unused kernel image (initmem) memory: 1424K
--
[54091.430777] Hardware name: To be filled by O.E.M. To be filled by O.E.M./M5A97 R2.0, BIOS 2603 06/26/2015
[54091.430779] RIP: 0010:snd_dma_vmalloc_get_addr+0x19/0x46
[54091.430787] Code: e9 c3 5a 58 00 48 03 77 18 48 89 f7 e9 0c 16 92 ff 53 48 8b 47 18 48 89 f3 81 e3 ff 0f 00 00 48 01 f0 48 89 c7 e8 f3 15 92 ff <48> 8b 38 48 89 c1 48 c1 ef 2d e8 76 fe ff ff 48 8b 10 48 89 c8 48
--
[54091.430812]  <TASK>
[54091.430814]  snd_dma_sg_fallback_alloc+0x148/0x16e
[54091.430819]  snd_dma_alloc_dir_pages+0x60/0x7e
[54091.430822]  do_alloc_pages+0x47/0x7e
[54091.430826]  snd_pcm_lib_malloc_pages+0xce/0x15a
[54091.430829]  snd_pcm_hw_params+0x175/0x393
[54091.430832]  snd_pcm_common_ioctl+0x355/0xe36
[54091.430835]  ? __seccomp_filter+0x81/0x36b
[54091.430841]  snd_pcm_ioctl+0x2a/0x31
[54091.430844]  vfs_ioctl+0x19/0x2a
--
[54091.430898] ---[ end trace 0000000000000000 ]---
[54091.430900] RIP: 0010:snd_dma_vmalloc_get_addr+0x19/0x46
[54091.430903] Code: e9 c3 5a 58 00 48 03 77 18 48 89 f7 e9 0c 16 92 ff 53 48 8b 47 18 48 89 f3 81 e3 ff 0f 00 00 48 01 f0 48 89 c7 e8 f3 15 92 ff <48> 8b 38 48 89 c1 48 c1 ef 2d e8 76 fe ff ff 48 8b 10 48 89 c8 48
--
[140042.480211] Hardware name: To be filled by O.E.M. To be filled by O.E.M./M5A97 R2.0, BIOS 2603 06/26/2015
[140042.480214] RIP: 0010:snd_dma_vmalloc_get_addr+0x19/0x46
[140042.480222] Code: e9 c3 5a 58 00 48 03 77 18 48 89 f7 e9 0c 16 92 ff 53 48 8b 47 18 48 89 f3 81 e3 ff 0f 00 00 48 01 f0 48 89 c7 e8 f3 15 92 ff <48> 8b 38 48 89 c1 48 c1 ef 2d e8 76 fe ff ff 48 8b 10 48 89 c8 48
--
[140042.480253]  <TASK>
[140042.480255]  snd_dma_sg_fallback_alloc+0x148/0x16e
[140042.480261]  snd_dma_alloc_dir_pages+0x60/0x7e
[140042.480265]  do_alloc_pages+0x47/0x7e
[140042.480269]  snd_pcm_lib_malloc_pages+0xce/0x15a
[140042.480274]  snd_pcm_hw_params+0x175/0x393
[140042.480278]  snd_pcm_common_ioctl+0x355/0xe36
[140042.480282]  ? __seccomp_filter+0x81/0x36b
[140042.480288]  snd_pcm_ioctl+0x2a/0x31
[140042.480291]  vfs_ioctl+0x19/0x2a
--
[140042.480365] ---[ end trace 0000000000000000 ]---
[140042.480367] RIP: 0010:snd_dma_vmalloc_get_addr+0x19/0x46
[140042.480372] Code: e9 c3 5a 58 00 48 03 77 18 48 89 f7 e9 0c 16 92 ff 53 48 8b 47 18 48 89 f3 81 e3 ff 0f 00 00 48 01 f0 48 89 c7 e8 f3 15 92 ff <48> 8b 38 48 89 c1 48 c1 ef 2d e8 76 fe ff ff 48 8b 10 48 89 c8 48

andy155, 2022-12-21:
https://pastebin.com/4yy1TJKQ

perexg, 2022-12-21:
The kernel bugs should be entered to https://bugzilla.kernel.org (Audio group).

closed 2022-12-21 by perexg