alsa-project/alsa-lib #289: Return -DPIC with `pkg-config --cflags alsa` when appropriate

afq984, 2022-12-31:
Some build systems (we are using bazel) do not define -DPIC when compiling position independent objects. 

When building a alsa module, if `PIC` is not defined, `global.h` requires the presence of the `snd_dlsym_start` symbol, which is absent from `libasound.so` if alsa itself is built with `-DPIC`.

https://github.com/alsa-project/alsa-lib/blob/9447e57d7c1602a861635487ca56c452f3472965/include/global.h#L72

If the `PIC` macro is non-standard I would suggest exposing it via pkg-config so position independent alsa modules can pick up the flag and apply it to the alsa headers appropriately.

perexg, 2023-01-05:
It's a third-party module ? This flag is set by configure (libtool) and it is not required for the standard alsa-lib apps (except the modules). I would suggest to add `#define PIC 1` line to the third-party module source or build rules.

afq984, 2023-01-06:
yeah we're building an alsa plugin for our audio server (CRAS). We currently add -DPIC to our build rules, and I'm wondering if it make sense for this to exposed from pkg-config.
If you think nothing needs to be changed feel free to close this, thabks!