alsa-project/alsa-lib #291: simple mixer element from a control event

flatmax, 2023-01-06:
I would like to get a snd_mixer_elem_t from a control event :

```
snd_ctl_event_t *event;
snd_ctl_event_alloca(&event);
int err = snd_ctl_read(ctl, event);
if (err < 0)
  return ALSADebug().evaluateError(err, "snd_ctl_read error when handling poll events\n");
```

My problem is sometimes there are mixer elements with the same names. I can get the mixer element's name like so :

`std::string elemName(snd_ctl_event_elem_get_name(event));`

But when I search for the simple mixer element based on the element name, the search will stop at the first found matching element. A few issues arise with this approach, for example when the same simple mixer element is in both the playback and capture alsamixer sections. None the less I can iterate through the name for comparson :

```
 for (snd_mixer_elem_t *elem = snd_mixer_first_elem(handle); elem; elem = snd_mixer_elem_next(elem)){
  // Do a search using elemName here breaking on the first match, which may not be the correct match.

```

It would be better if there was a way to find the exact simple mixer element which triggered the event in the first place. Is that possible ? I am aware of the snd_ctl_event_elem_get_numid function, but I can't see how to find a simple mixer element which matches that numid.

perexg, 2023-01-06:
The name is only one from multiple fields identifying the control element. Use `snd_ctl_event_elem_get_id` to get the id structure which contains all fields.

perexg, 2023-01-06:
Also, it seems that you mix two APIs. Do not use raw `snd_ctl_*` functions with the `snd_mixer_*` functions. It won't work. The control mapping between those two APIs is not straight.

perexg, 2023-01-06:
What you're trying to implement?

flatmax, 2023-01-08:
I'm trying to catch mixer element changes as events. Then I want to 
fetch the changed mixer element and propagate that element state through 
other user applications (over the network for example),

flatmax, 2023-01-09:
I added a MixerEvents class code to handle mixer events in the same way 
amixer handles it. Unfortunately I can't seem to get my threaded 
snd_mixer_wait function to release with an event.

https://github.com/flatmax/gtkiostream/commit/6f4ed685c95db2321e267d0b65025742fccf3931#diff-ccebe112c934530cd53d80c769c599ab3abd878d40278bf66909aa0c66fb3a47R35-R38

Matt

flatmax, 2023-01-09:
OK, got it to work now.

The thing I missed was that snd_mixer_wait should only be called once 
the mixer has been loaded. If it is called before, then it will wait 
forever missing events.

Here is the code :

https://github.com/flatmax/gtkiostream/blob/master/include/ALSA/MixerEvents.H

closed 2023-01-24 by flatmax