alsa-project/alsa-lib #292: `USB-Audio`: Wrong S/PDIF (`iec958`) input and output

sibzou, 2023-01-15:
I am new to the Linux sound system. As I understand, USB sound cards doesn't directly tell the kernel or alsa-lib how many S/PDIF inputs and outputs they have. But they obviously tell their "brand name" like "Plantronics USB Headset" or "HP Digital Stereo Headset" for example.

To present the correct number of S/PDIF inputs and outputs to the rest of the userspace, alsa-lib seems to maintain a kind of known devices database in `src/conf/cards/USB-Audio.conf`.

If a card brand is not known in this file, alsa-lib always present a single S/PDIF input and a single S/PDIF output. If the index of this card is `1` for example, `snd_pcm_open(&pcm, "iec958:1", SND_PCM_STREAM_PLAYBACK, 0)` and `snd_pcm_open(&pcm, "iec958:1", SND_PCM_STREAM_CAPTURE, 0)` will always return 0 even if the USB device doesn't have a real S/PDIF input or output.

My question is why this is the default behavior? We are in a "every USB cards have S/PDIF, except those who are known in `USB-Audio.conf`" logic. I think it should be "None USB card has S/PDIF except those who are known for it in `USB-Audio.conf`" because USB devices without S/PDIF are much more common.

Cf https://bugs.launchpad.net/ubuntu/+source/alsa-lib/+bug/1002952

Thank you!

insidesecurity-yhojann-aguilera, 2024-11-11:
Any update for this? have same chip but optical audio does not work.

lkraav, 2026-09-29:
https://forum.endeavouros.com/t/sb-omni-5-1-digital-out-no-go/5364/18 pointed me in this direction and I, after years of not knowing how, have a working SPDIF output with a simple Wireplumber override conf 🙌 

- **Device:** `041e:322c` SB Omni Surround 5.1, card ID `S51`, alsa-lib 1.2.16.1, kernel 6.18.
- **Symptom:** `iec958:CARD=S51` plays to PCM 0 (analog) even though `USB-Audio.pcm.iec958_device` maps "SB Omni Surround 5.1" to 1. Using `hw:S51,1` works.
- **Suspected cause:** device name is joined into config key `cards.USB-Audio.pcm.iec958_device.SB Omni Surround 5.1`, and the dot in "5.1" breaks the lookup, so it falls back to `default 0`. I haven't confirmed this. Presenting it as a suspected cause, perhaps maintainers know better.
- **Workaround:** WirePlumber `api.alsa.path = "hw:S51,1"` override.

```
$ cat .config/wireplumber/wireplumber.conf.d/51-sb-omni-spdif.conf 
# SB Omni Surround 5.1: S/PDIF is PCM device 1, but alsa-lib's USB-Audio.conf
# iec958_device lookup breaks on the "." in the card name and falls back to 0 (analog).
monitor.alsa.rules = [
  {
    matches = [
      { node.name = "alsa_output.usb-Creative_Technology_Ltd_SB_Omni_Surround_5.1_000001FH-00.iec958-stereo" }
    ]
    actions = {
      update-props = {
        api.alsa.path = "hw:S51,1"
      }
    }
  }
]
```

Seems like this is the right issue to hop onto, but lmk if this is actually "SB Omni Surround 5.1" specific, I can file a separate one.

iibaranov-IG, 2026-10-05:
The recorded offline lookup test on alsa-lib revision f84cd4ced7b36fddb8e4ee24404cf7c091d27020 returned device 0 for SB Omni Surround 5.1 before our unpublished local draft and device 1 afterward. All 49 lookup checks passed afterward, and make check reported two passing tests. This supports the lookup issue you described, but does not establish physical S/PDIF playback or capture, or WirePlumber behavior with the change. It also does not resolve the broader default-routing policy discussed in this issue.

If anyone still has this hardware, confirmation of IEC958 PCM selection and optical playback/capture would be useful.