alsa-project/alsa-lib #295: add a new mixer element to an existing simple mixer

flatmax, 2023-01-24:
For testing purposes I would like to be able to open a simple mixer and add a dummy element for testing purposes.
The mixer's event handler will catch events and allow my code to react to these mixer events.

Is there a way to add a dummy volume control with a particular name to an open simple mixer ?

flatmax, 2023-02-01:
I've got a module now which loads. I took the mixer volume setup from soc_dummy and reused it. But for some reason, when I try to use alsamixer it hangs. I can't seem to work out why - nor unload the mixer.

Here is a link to the repo : 

https://github.com/flatmax/alsa-mixer-module

Any pointers on how to fix these problems would be greatly appreciated.

perexg, 2023-02-25:
This is an issue tracker for the alsa-lib package. The kernel related issues should go to the alsa-devel mailing list or https://bugzilla.kernel.org .

I really don't understand what you're trying to do. The elements can be also added from the user space to an existing card. Also we have already snd_dummy module.

closed 2023-02-25 by perexg