alsa-project/alsa-lib #296: alsa without udev

stefan11111, 2023-01-27:
I am using gentoo and trying to only use a static /dev. I Only use alsa for sound, no pulseaudio or anything like that.
If I am using udev, alsa works perfectly fine, even if I then stop udev.
I I don't use udev, alsa won't work:
`ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.8-r1/work/alsa-lib-1.2.8/src/confmisc.c:855(parse_card) cannot find card '0'`

perexg, 2023-01-27:
Static device nodes are supported only when the kernel is compiled with `CONFIG_SND_DYNAMIC_MINORS=n`. Note that some drivers will force the dynamic minor number allocation and almost all platforms including embedded, are using udev nowadays.

closed 2023-01-27 by perexg

stefan11111, 2023-01-28:
Thanks for responding.
I have recompiled my kernel with this option and I still have this problem. It appears that udev also populates /sys. When not using udev, the folder /sys/class/sound does not exist and I cannot create it to make the proper symlinks.

perexg, 2023-01-28:
The standard alsa-lib functions does not require sysfs (/sys tree). Only UCM code may use this subsystem for additional checks. It looks like another problem (minor numbers etc.). You may use `strace` tool to check which system calls fails when the ALSA card is probed.

stefan11111, 2023-01-28:
I have used strace as instructed.
Here is the output of strace speaker-test:
https://pastebin.com/6fY2WygZ

I may need to use mknod to manually populate /dev, but I am not sure how.

perexg, 2023-01-28:
Something like:

```
mkdir /dev/snd
mknod -m 660 /dev/snd/controlC0 c 116 0
mknod -m 660 /dev/snd/pcmC0D0p c 116 16
mknod -m 660 /dev/snd/timer c 116 33
chown root:audio /dev/snd/*
```

The full list of minors is here: https://github.com/torvalds/linux/blob/master/include/sound/minors.h

stefan11111, 2023-01-28:
I have made a screenshot of all the files that udev loads into /dev and used mknod like in the example above.

```
stefan@GentooPC ~ $ ls -lah /dev/snd
total 0
drwxr-xr-x  2 root root      320 Jan 28 21:19 .
drwxr-xr-x 17 root root     4.2K Jan 28 21:18 ..
crw-rw----  1 root audio 116,  0 Jan 28 21:19 controlC0
crw-rw----  1 root audio 116, 32 Jan 28 21:19 controlC1
crw-rw----  1 root audio 116, 64 Jan 28 21:19 controlC2
crw-rw----  1 root audio 116, 36 Jan 28 21:19 hwC1D0
crw-rw----  1 root audio 116, 68 Jan 28 21:19 hwC2D0
crw-rw----  1 root audio 116, 24 Jan 28 21:19 pcmC0D0c
crw-rw----  1 root audio 116, 16 Jan 28 21:19 pcmC0D0p
crw-rw----  1 root audio 116, 56 Jan 28 21:19 pcmC1D0c
crw-rw----  1 root audio 116, 48 Jan 28 21:19 pcmC1D0p
crw-rw----  1 root audio 116, 49 Jan 28 21:19 pcmC1D1p
crw-rw----  1 root audio 116, 58 Jan 28 21:19 pcmC1D2c
crw-rw----  1 root audio 116, 83 Jan 28 21:19 pcmC2D3p
crw-rw----  1 root audio 116,  1 Jan 28 21:18 seq
crw-rw----  1 root audio 116, 33 Jan 28 21:18 timer
stefan@GentooPC ~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
stefan@GentooPC ~ $ aplay -l
aplay: device_list:274: no soundcards found...
stefan@GentooPC ~ $

```

Using strace, I still see messages like this:
`openat(AT_FDCWD, "/dev/snd/controlC0", O_RDONLY|O_CLOEXEC) = -1 ENODEV (No such device)`

even though /dev/snd/controlC0 clearly exists

perexg, 2023-01-28:
Do you have the sound driver active ? (cat /proc/asound/cards)

stefan11111, 2023-01-28:
```
stefan@GentooPC ~ $ cat /proc/asound/cards
--- no soundcards ---

```
Guess not, though I have never installed a sound driver as an out of tree module or something.

perexg, 2023-01-28:
You should select the ALSA sound driver for your hardware in the kernel config.

stefan11111, 2023-01-28:
I have build my kernel using genkernel and I have just changed the options needed for `CONFIG_SND_DYNAMIC_MINORS=n`.
I will try to build all the sound drivers directly into the kernel.
If that doesn't fix it, I'll do a manual kernel config.
Wish me luck.

perexg, 2023-01-28:
In my first reply, I noted that some drivers require the dynamic minor number allocations. You should check the kernel's CONFIG_SND_ configuration options, if they are enabled before the kernel compilation.

perexg, 2023-01-28:
It may be eventually possible to use the dynamic minors - but a custom script which will create the device nodes (mknod) from /proc/asound/devices file must be written. The first number is the minor.

stefan11111, 2023-01-28:
I do not really use a sound card. I have a razer headset that is connected via usb and I do not use openrazer.

Both /proc/asound/cards and /proc/asound/devices are empty

perexg, 2023-01-28:
It's probably an USB sound card built-in to the headset. Enable ALSA USB sound driver.

stefan11111, 2023-01-29:
I have recompiled my kernel with the alsa sound module.
This is the output of cat /proc/asound/cards:

```
stefan@GentooPC ~ $ cat /proc/asound/cards
 0 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0xf7140000 irq 132
 1 [NVidia         ]: HDA-Intel - HDA NVidia
                      HDA NVidia at 0xf7080000 irq 17
 2 [Ultimate       ]: USB-Audio - Razer Kraken Ultimate
                      Razer Razer Kraken Ultimate at usb-0000:00:14.0-12, full speed
```
This is the output of cat ~/.asoundrc:

```
stefan@GentooPC ~ $ cat ~/.asoundrc
defaults.pcm.card 2
defaults.ctl.card 2
```

This is the output of  speaker-test:

```
stefan@GentooPC ~ $ speaker-test

speaker-test 1.2.8

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.8-r1/work/alsa-lib-1.2.8/src/pcm/pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave
Playback open error: -13,Permission denied
```
This is the output of aplay -l:

```
stefan@GentooPC ~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC892 Analog [ALC892 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 1: ALC892 Digital [ALC892 Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: NVidia [HDA NVidia], device 3: Generic Digital [Generic Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Ultimate [Razer Kraken Ultimate], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I have written 2 scripts to load the necessary modules and make the necessary /dev files:

```
stefan@GentooPC ~ $ cat mknod_snd
mkdir /dev/snd
mknod -m 660 /dev/snd/controlC0 c 116 0
mknod -m 660 /dev/snd/pcmC0D0p c 116 16
mknod -m 660 /dev/snd/timer c 116 33
mknod -m 660 /dev/snd/seq c 166 1
mknod -m 660 /dev/snd/controlC2 c 116 64
mknod -m 660 /dev/snd/controlC1 c 116 32
mknod -m 660 /dev/snd/pcmC0D0c c 116 24
mknod -m 660 /dev/snd/hwC1D0 c 116 36
mknod -m 660 /dev/snd/hwC2D0 c 116 68
mknod -m 660 /dev/snd/pcmC1D0c c 116 56
mknod -m 660 /dev/snd/pcmC1D0p c 116 48
mknod -m 660 /dev/snd/pcmC1D1p c 116 49
mknod -m 660 /dev/snd/pcmC1D2c c 116 58
mknod -m 660 /dev/snd/pcmC2D3p c 116 83
chown root:audio /dev/snd/*
```
```
stefan@GentooPC ~ $ cat sound_modules
modprobe snd_rawmidi
modprobe snd_usbmidi_lib
modprobe snd_hda_core
modprobe snd_hda_codec
modprobe snd_hda_codec_generic
modprobe snd_hda_intel
modprobe snd_hda_codec_realtek
modprobe ledtrig_audio
modprobe pcspkr
modprobe snd_usb_audio
modprobe snd_hwdep
modprobe snd_intel_sdw_acpi
modprobe snd_intel_dspcfg
modprobe snd_usb_audio
modprobe mc
modprobe snd
modprobe soundcore
modprobe snd_seq_device
modprobe snd_timer
modprobe snd_pcm
```

Here is the output of strace speaker-test:

https://pastebin.com/brjkbNDR

perexg, 2023-01-29:
```
openat(AT_FDCWD, "/dev/snd/pcmC2D0p", O_RDWR|O_NONBLOCK|O_CLOEXEC) = -1 EACCES (Permission denied)
```

Check if the user is in the audio group.

stefan11111, 2023-01-29:
This is the output of groups:

```
stefan@GentooPC ~ $ groups
tty wheel audio video input users stefan
```
My user seems to be in the audio group.
I have just tested running speaker-test and youtube on librewolf with root privileges and it works properly.
I would call this progress.

perexg, 2023-01-29:
Check the other permission related setup - /dev/snd directory permissions, SELinux or so... It's out of scope of support for the audio. Ask on gentoo forums.

stefan11111, 2023-01-29:
I will do so. Thanks for helping me so far.

stefan11111, 2023-02-15:
@perexg 
I have fixed the issue.
I will post it here for anyone who will encounter this issue.
I wrote a script to make all the necessary nodes and load all the necessary modules:
```
mkdir /dev/snd
mknod -m 660 /dev/snd/controlC0 c 116 0
mknod -m 660 /dev/snd/pcmC0D0p c 116 16
mknod -m 660 /dev/snd/timer c 116 33
mknod -m 660 /dev/snd/seq c 166 1
mknod -m 660 /dev/snd/controlC2 c 116 64
mknod -m 660 /dev/snd/controlC1 c 116 32
mknod -m 660 /dev/snd/pcmC0D0c c 116 24
mknod -m 660 /dev/snd/hwC1D0 c 116 36
mknod -m 660 /dev/snd/hwC2D0 c 116 68
mknod -m 660 /dev/snd/pcmC1D0c c 116 56
mknod -m 660 /dev/snd/pcmC1D0p c 116 48
mknod -m 660 /dev/snd/pcmC1D1p c 116 49
mknod -m 660 /dev/snd/pcmC1D2c c 116 58
mknod -m 660 /dev/snd/pcmC2D3p c 116 83
chown root:audio /dev/snd/*
chown root:audio /dev/snd
modprobe snd_rawmidi
modprobe snd_usbmidi_lib
modprobe snd_hda_core
modprobe snd_hda_codec
modprobe snd_hda_codec_generic
modprobe snd_hda_intel
modprobe snd_hda_codec_realtek
modprobe ledtrig_audio
modprobe pcspkr
modprobe snd_usb_audio
modprobe snd_hwdep
modprobe snd_intel_sdw_acpi
modprobe snd_intel_dspcfg
modprobe snd_usb_audio
modprobe mc
modprobe snd
modprobe soundcore
modprobe snd_seq_device
modprobe snd_timer
modprobe snd_pcm
chmod -R a+rw /dev/snd
```
for convenience, I added this to my .xinitrc:
```
#!/bin/sh

slstatus &
xcompmgr &
/usr/bin/feh --bg-fill '/home/stefan/images/wallpapers/2.jpg' &
xset -dpms &
xset s off &
rdo ntpd &
qbittorrent-nox &
rdo curl https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts > /etc/hosts &
rdo mount -a &
#make /dev nodes
mkdir /dev/snd
rdo mknod -m 660 /dev/snd/controlC0 c 116 0
rdo mknod -m 660 /dev/snd/pcmC0D0p c 116 16
rdo mknod -m 660 /dev/snd/timer c 116 33
rdo mknod -m 660 /dev/snd/seq c 166 1
rdo mknod -m 660 /dev/snd/controlC2 c 116 64
rdo mknod -m 660 /dev/snd/controlC1 c 116 32
rdo mknod -m 660 /dev/snd/pcmC0D0c c 116 24
rdo mknod -m 660 /dev/snd/hwC1D0 c 116 36
rdo mknod -m 660 /dev/snd/hwC2D0 c 116 68
rdo mknod -m 660 /dev/snd/pcmC1D0c c 116 56
rdo mknod -m 660 /dev/snd/pcmC1D0p c 116 48
rdo mknod -m 660 /dev/snd/pcmC1D1p c 116 49
rdo mknod -m 660 /dev/snd/pcmC1D2c c 116 58
rdo mknod -m 660 /dev/snd/pcmC2D3p c 116 83
rdo chown root:audio /dev/snd/*
rdo chown root:audio /dev/snd
#load kernel modules
rdo modprobe snd_rawmidi
rdo modprobe snd_usbmidi_lib
rdo modprobe snd_hda_core
rdo modprobe snd_hda_codec
rdo modprobe snd_hda_codec_generic
rdo modprobe snd_hda_intel
rdo modprobe snd_hda_codec_realtek
rdo modprobe ledtrig_audio
rdo modprobe pcspkr
rdo modprobe snd_usb_audio
rdo modprobe snd_hwdep
rdo modprobe snd_intel_sdw_acpi
rdo modprobe snd_intel_dspcfg
rdo modprobe snd_usb_audio
rdo modprobe mc
rdo modprobe snd
rdo modprobe soundcore
rdo modprobe snd_seq_device
rdo modprobe snd_timer
rdo modprobe snd_pcm
#set permissions
rdo chmod -R a+rw /dev/snd
#exec dbus-launch --exit-with-session dwm
exec dwm
```
rdo is a less bloated alternative to doas. All it does in this context is execute those commands as root.
This line seems to have fixed it:
`rdo chmod -R a+rw /dev/snd`
I don't know if this is the best way to solve this, but it works. I would like some feedback.

Here is my .asoundrc:
```
defaults.pcm.card 2
defaults.ctl.card 2
```