alsa-project/alsa-lib #297: hdspm MIDI-in crash

aiobofh, 2023-01-29:
I've tried to find some insights in a weird problem I have with snd_hdspm. I
guess it is an error-report/bug-posting.

I'm using hdspm for my RME RayDAT card. Using _every_ input I can, in my studio :D. Upon ALSA i run jackd for audio, to get use for the awesome application support for studios. However I never used jackd for MIDI. That's pure ALSA in my set-up.

Audio-wise everything is super-duper, so don't worry. However... MIDI only works
outbound from the on-board MIDI interface.

Whenever I connect anything to the MIDI-in and start feeding some MIDI-data, it
seems ALSA crashes on me. here's a more detailed description:

* I'm running linux-rt 6.0.5.14.realtime1-3-rt #1 SMP PREEMPT_RT (Arch)

* When i give it a stream (randomly slam keys on a MIDI keyboard, or a controller) of MIDI data on the MIDI-in while playing some audio that application (e.g. mplayer, live inputs sources directly connected to playback sinks or what not) i see application error messages like:

  ALSA: poll time out, polled for 1999005 usecs, Retrying with a
  recovery, retry cnt = 1 (2, 3, 4, 5, and so on)

  And ALSA seem stuck in looping the last buffer over and over, until i reboot the machine.

* Nothing in journal, nothing in dmesg tells anything more about it.

Is this a known problem? How can I provide more information in a good way, if needed - I really can't see anything else in the logging facilities of the system. Just get this behavior.

tiwai, 2023-05-23:
Did the suggested patch help?
  https://lore.kernel.org/r/871qnapx0n.wl-tiwai@suse.de

aiobofh, 2023-09-18:
It didn't :(

tiwai, 2023-09-18:
Then try to set up the kdump and get the backtraces of the hanging tasks.
It's difficult to diagnose without any logs.