alsa-project/alsa-lib #299: No sound on HP Pavilion All-in-One Desktop 27-ca1xxx with ALC274

ernstp, 2023-02-05:
This is an AIO with build in speakers, B&O branded. There is no sound from them.
I have tried:
options snd-hda-intel model=alc285-hp-amp-init
options snd-hda-intel model=dell-headset-multi
options snd-hda-intel model=alc256-asus-aio

Suspend/resume doesn't help either.

Here's my alsa-info, I think it includes most details?
https://alsa-project.org/db/?f=8c6b4e7932d4bce0ba534a73d44e10f32e9dcd01
( bought the Pebble V3 to use in the meantime...)

I notice the internal mic is working actually.

ernstp, 2023-02-05:
Here's one where I booted with model=generic
https://alsa-project.org/db/?f=9831f3d8241c3aa4fdaf9415ff835bab92429b52

ernstp, 2023-02-05:
Normal boot with dmesg included:
https://alsa-project.org/db/?f=2967b1a901f35e1b2f509405db8592a5b201930c

perexg, 2023-03-31:
The driver problems should be reported in https://bugzilla.kernel.org (Audio group).

closed 2023-03-31 by perexg