alsa-project/alsa-lib #309: Does this compile any code?

stefan11111, 2023-03-29:
If I build the same kernel twice, I get this output when building it the second time:
```
# make
  CALL    scripts/checksyscalls.sh
  DESCEND objtool
Kernel: arch/x86/boot/bzImage is ready  (#17)
```
If I build the kernel again, only changing SND_X86, I get this output:
```
# make
  SYNC    include/config/auto.conf
  CALL    scripts/checksyscalls.sh
  DESCEND objtool
Kernel: arch/x86/boot/bzImage is ready  (#17)

```
It doesn't look like that option compiles any code.

stefan11111, 2023-03-29:
Also, can someone tell me what /dev node does CONFIG_SND_ALOOP create, and what major and minor number does it have?

chleroy, 2023-03-30:
As far as I can see, the only thing SND_X86 does is to enable selection of HDMI_LPE_AUDIO, see https://github.com/torvalds/linux/blob/master/sound/x86/Kconfig

Unless you also select HDMI_LPE_AUDIO, nothing more will happen

chleroy, 2023-03-30:
> 

CONFIG_SND_ALOOP is a ALSA driver, it doesn't create a /dev by itself.

Use alsa utils to know