alsa-project/alsa-lib #316: Bug: premature stringification in SND_DLSYM_BUILD_VERSION produces wrong results for alsa-plugins.

MIvanchev, 2023-05-07:
The macro:

```
#define SND_DLSYM_BUILD_VERSION(name, version) \
  static struct snd_dlsym_link __SND_DLSYM_VERSION(snd_dlsym_, name, version); \
  void __SND_DLSYM_VERSION(snd_dlsym_constructor_, name, version) (void) __attribute__ ((constructor)); \
  void __SND_DLSYM_VERSION(snd_dlsym_constructor_, name, version) (void) { \
    __SND_DLSYM_VERSION(snd_dlsym_, name, version).next = snd_dlsym_start; \
    __SND_DLSYM_VERSION(snd_dlsym_, name, version).dlsym_name = # name; \
    __SND_DLSYM_VERSION(snd_dlsym_, name, version).dlsym_ptr = (void *)&name; \
    snd_dlsym_start = &__SND_DLSYM_VERSION(snd_dlsym_, name, version); \
  }
```
should actually be
```
#define SND_DLSYM_NAME(name) #name

#define SND_DLSYM_BUILD_VERSION(name, version) \
  static struct snd_dlsym_link __SND_DLSYM_VERSION(snd_dlsym_, name, version); \
  void __SND_DLSYM_VERSION(snd_dlsym_constructor_, name, version) (void) __attribute__ ((constructor)); \
  void __SND_DLSYM_VERSION(snd_dlsym_constructor_, name, version) (void) { \
    __SND_DLSYM_VERSION(snd_dlsym_, name, version).next = snd_dlsym_start; \
    __SND_DLSYM_VERSION(snd_dlsym_, name, version).dlsym_name = SND_DLSYM_NAME(name) \
    __SND_DLSYM_VERSION(snd_dlsym_, name, version).dlsym_ptr = (void *)&name; \
    snd_dlsym_start = &__SND_DLSYM_VERSION(snd_dlsym_, name, version); \
  }
```
This is because for instance [this code](https://github.com/alsa-project/alsa-plugins/blob/master/pulse/ctl_pulse.c#LL823C1-L823C30) `SND_CTL_PLUGIN_SYMBOL(pulse);` expands to

```
SND_DLSYM_BUILD_VERSION(SND_CTL_PLUGIN_ENTRY(pulse), SND_CONTROL_DLSYM_VERSION);
```
and this expands to
```
...
__SND_DLSYM_VERSION(snd_dlsym_, name, version).dlsym_name = # SND_CTL_PLUGIN_ENTRY(pulse)
...
```
but the C preprocessor doesn't expand stringified tokens so the value of `dlsym_name` is actually `"SND_CTL_PLUGIN_ENTRY(pulse)"`.

perexg, 2023-05-11:
External plugins must be compiled with -DPIC, so this path is not used at all in reality. Or are you trying to build a static library with external plugins?

MIvanchev, 2023-05-11:
Even better, I'm building a static library with static plugins for my project [static-wine32](https://github.com/MIvanchev/static-wine32). It works nicely! Here are the patches I use: [alsa-lib](https://github.com/MIvanchev/static-wine32/blob/master/dependencies/patches/alsa-lib-1.2.9.patch) and [alsa-plugins](https://github.com/MIvanchev/static-wine32/blob/master/dependencies/patches/alsa-plugins-1.2.7.1.patch). I understand this is not in the scope of your project though.

closed 2023-05-16 by perexg