alsa-project/alsa-lib #325: ASUS RoG Strix Z790-F: No SPDIF Sound output

Nightwing0815, 2023-06-09:
Hey guys,

what Can I do to get my optical toslink output ( pcm signal undecoded ) working?
Here are an inxi report of my system:

```
System:
  Host: Ultron Kernel: 6.1.0-1013-oem x86_64 bits: 64 compiler: N/A
    parameters: BOOT_IMAGE=/vmlinuz-6.1.0-1013-oem
    root=UUID=aa125f57-ee1e-4fd4-93fc-a77a180ec451 ro quiet splash
  Desktop: Cinnamon 5.6.8 tk: GTK 3.24.33 wm: muffin vt: 7
    dm: LightDM 1.30.0 Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: ASUS product: N/A v: N/A serial: <superuser required>
  Mobo: ASUSTeK model: ROG STRIX Z790-F GAMING WIFI v: Rev 1.xx
    serial: <superuser required> UEFI: American Megatrends v: 0904
    date: 03/29/2023
Battery:
  Message: No system battery data found. Is one present?
Memory:
  RAM: total: 62.57 GiB used: 2.13 GiB (3.4%)
  RAM Report:
    unknown-error: Unknown dmidecode error. Unable to generate data.
PCI Slots:
  Unknown-error: Unknown dmidecode error. Unable to generate data.
CPU:
  Info: model: 13th Gen Intel Core i7-13700K bits: 64 type: MST AMCP
    arch: N/A family: 6 model-id: 0xB7 (183) stepping: 1 microcode: 0x113
  Topology: cpus: 1x cores: 16 mt: 8 tpc: 2 st: 8 threads: 24 smt: enabled
    cache: L1: 1.4 MiB desc: d-8x32 KiB, 8x48 KiB; i-8x32 KiB, 8x64 KiB
    L2: 24 MiB desc: 8x2 MiB, 2x4 MiB L3: 30 MiB desc: 1x30 MiB
  Speed (MHz): avg: 3100 high: 3400 min/max: 800/5300:5400:4200 scaling:
    driver: intel_pstate governor: powersave cores: 1: 1100 2: 3400 3: 3400
    4: 3400 5: 3400 6: 3400 7: 3400 8: 3400 9: 3400 10: 3400 11: 3400
    12: 3400 13: 3400 14: 3400 15: 1100 16: 3400 17: 800 18: 3400 19: 3400
    20: 3400 21: 3400 22: 3400 23: 3400 24: 3400 bogomips: 164044
Graphics:
  Device-1: NVIDIA vendor: CardExpert driver: nvidia v: 530.41.03
    alternate: nvidiafb,nouveau,nvidia_drm ports: active: none
    off: DP-1,HDMI-A-1 empty: DP-2,DP-3 bus-ID: 0000:01:00.0
    chip-ID: 10de:2782 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    gpu: nvidia display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x2160 s-dpi: 46 s-size: 1060x1167mm (41.7x45.9")
    s-diag: 1577mm (62.1")
  Monitor-1: DP-1 pos: primary,bottom res: 1920x1080 hz: 60 dpi: 102
    size: 477x268mm (18.8x10.6") diag: 547mm (21.5")
  Monitor-2: HDMI-0 pos: primary,top res: 1920x1080 hz: 60 dpi: 38
    size: 1280x720mm (50.4x28.3") diag: 1469mm (57.8")
  OpenGL: renderer: NVIDIA GeForce RTX 4070 Ti/PCIe/SSE2
    v: 4.6.0 NVIDIA 530.41.03 direct render: Yes
Audio:
  Device-1: Intel vendor: ASUSTeK driver: snd_hda_intel v: kernel
    alternate: snd_sof_pci_intel_tgl bus-ID: 0000:00:1f.3 chip-ID: 8086:7a50
    class-ID: 0403
  Device-2: NVIDIA vendor: CardExpert driver: snd_hda_intel v: kernel
    bus-ID: 0000:01:00.1 chip-ID: 10de:22bc class-ID: 0403
  Device-3: ASUSTek USB Audio type: USB
    driver: hid-generic,snd-usb-audio,usbhid bus-ID: 1-2:2 chip-ID: 0b05:1a52
    class-ID: 0300
  Sound Server-1: ALSA v: k6.1.0-1013-oem running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: no
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel vendor: ASUSTeK driver: igc v: kernel port: N/A
    bus-ID: 0000:05:00.0 chip-ID: 8086:125c class-ID: 0200
  IF: eno2 state: up speed: 1000 Mbps duplex: full mac: c8:7f:54:6a:7e:55
  IP v4: 192.168.178.10/24 type: noprefixroute scope: global
    broadcast: 192.168.178.255
  IP v6: 2a02:8070:8a80:77c0:aac9:dade:e3e2:8611/64 type: temporary dynamic
    scope: global
  IP v6: 2a02:8070:8a80:77c0:64ef:c8d5:1d2b:3a7f/64
    type: dynamic mngtmpaddr noprefixroute scope: global
  IP v6: fe80::6d94:62d1:90b5:608b/64 type: noprefixroute scope: link
  IF-ID-1: virbr0 state: down mac: 52:54:00:e9:bf:38
  IP v4: 192.168.100.1/24 scope: global broadcast: 192.168.100.255
  WAN IP: 46.5.221.128
Bluetooth:
  Message: No bluetooth data found.
Logical:
  Message: No logical block device data found.
RAID:
  Hardware-1: Intel Device driver: vmd v: 0.6 port: N/A bus-ID: 0000:00:0e.0
    chip-ID: 8086:a77f rev: class-ID: 0104
Unmounted:
  Message: No unmounted partitions found.
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 16 rev: 2.0
    speed: 480 Mb/s chip-ID: 1d6b:0002 class-ID: 0900
  Device-1: 1-2:2 info: ASUSTek USB Audio type: Audio,HID
    driver: hid-generic,snd-usb-audio,usbhid interfaces: 8 rev: 2.0
    speed: 480 Mb/s power: 100mA chip-ID: 0b05:1a52 class-ID: 0300
  Device-2: 1-4:3 info: ASUSTek AURA LED Controller type: HID
    driver: hid-generic,usbhid interfaces: 2 rev: 2.0 speed: 12 Mb/s
    power: 16mA chip-ID: 0b05:19af class-ID: 0300 serial: 9876543210
  Hub-2: 1-10:4 info: ASMedia ASM1074 High-Speed hub ports: 4 rev: 2.1
    speed: 480 Mb/s power: 100mA chip-ID: 174c:2074 class-ID: 0900
  Device-1: 1-10.3:6 info: Fujitsu Siemens s KB955 USB Keyboard
    type: Keyboard,HID driver: hid-generic,usbhid interfaces: 2 rev: 1.1
    speed: 1.5 Mb/s power: 100mA chip-ID: 0bf8:102e class-ID: 0300
  Device-2: 1-10.4:8 info: Logitech G5 Laser Mouse type: Mouse,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 2.0 speed: 12 Mb/s
    power: 98mA chip-ID: 046d:c049 class-ID: 0300
  Hub-3: 1-11:5 info: Genesys Logic Hub ports: 4 rev: 2.0 speed: 480 Mb/s
    power: 100mA chip-ID: 05e3:0608 class-ID: 0900
  Hub-4: 1-13:7 info: Genesys Logic Hub ports: 4 rev: 2.0 speed: 480 Mb/s
    power: 100mA chip-ID: 05e3:0608 class-ID: 0900
  Hub-5: 2-0:1 info: Super-speed hub ports: 8 rev: 3.1 speed: 20 Gb/s
    chip-ID: 1d6b:0003 class-ID: 0900
  Hub-6: 2-8:2 info: ASMedia ASM1074 SuperSpeed hub ports: 4 rev: 3.0
    speed: 5 Gb/s power: 8mA chip-ID: 174c:3074 class-ID: 0900
Sensors:
  System Temperatures: cpu: 32.5 C mobo: 30.0 C gpu: nvidia temp: 36 C
  Fan Speeds (RPM): fan-1: 0 fan-2: 808 fan-3: 0 fan-4: 0 fan-5: 447
    fan-6: 453 fan-7: 2288 gpu: nvidia fan: 0%
```

Thanks in advance,

ToM

andromedarion, 2023-07-25:
I'm having a similar issue on a Z790-A Board. Sound of for HDMI and Bluetooth but not for SPDIF

somejavadev, 2023-08-30:
I am experiencing the same with an Asus Rog Maximus Z790 Hero on Ubuntu 22.04. It seems like both our boards have the USB Audio card. @andromedarion is yours also an Asus board with USB audio, perhaps you can add the inxi data (`sudo apt install inxi && inxi --audio --machine --system`)? I have bumped the kernel to 6.4.12 and updated the USB card to the latest firmware without success. Strangely if I switch to the alsamixer and select the USB sound card it doesn't show any SPDIF outputs.

inxi:

```
System:
  Host: 2fast Kernel: 6.4.12-060412-generic x86_64 bits: 64
    Desktop: GNOME 42.9 Distro: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
Machine:
  Type: Desktop System: ASUS product: N/A v: N/A serial: <superuser required>
  Mobo: ASUSTeK model: ROG MAXIMUS Z790 HERO v: Rev 1.xx
    serial: <superuser required> UEFI: American Megatrends v: 1202
    date: 06/13/2023
Audio:
  Device-1: Intel driver: snd_hda_intel
  Device-2: AMD driver: snd_hda_intel
  Device-3: ASUSTek USB Audio type: USB
    driver: hid-generic,snd-usb-audio,usbhid
  Sound Server-1: ALSA v: k6.4.12-060412-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
```

alsamixer:

[image]

pavucontrol

[image]

[image]

@Nightwing0815 Perhaps you can also add screenshots from pavucontrol and alsamixer as additional information to narrow down the issue.

BrettRyland, 2023-08-30:
I also have this issue with a Z790-F motherboard.
```
$ sudo inxi --audio --machine --system
System:
  Host: brett-home Kernel: 6.4.6-060406-generic arch: x86_64 bits: 64
    Desktop: KDE Plasma v: 5.27.4 Distro: Ubuntu 23.04 (Lunar Lobster)
Machine:
  Type: Desktop System: Komplett product: KomplettPC v: N/A serial: 25769254
  Mobo: ASUSTeK model: ROG STRIX Z790-F GAMING WIFI v: Rev 1.xx
    serial: 230215882500616 UEFI: American Megatrends v: 1202 date: 06/13/2023
Audio:
  Device-1: Intel driver: snd_hda_intel
  Device-2: NVIDIA AD102 High Definition Audio driver: snd_hda_intel
  Device-3: Logitech HD Pro Webcam C920 type: USB
    driver: snd-usb-audio,uvcvideo
  Device-4: ASUSTek USB Audio type: USB
    driver: hid-generic,snd-usb-audio,usbhid
  Sound API: ALSA v: k6.4.6-060406-generic running: yes
  Sound Server-1: PipeWire v: 0.3.65 running: yes
```

Interestingly, pavucontrol thinks that it is outputting audio, but the receiver says that there is no digital data.
[image]
I'll also note that switching to "Analogue Surround 5.1 Output" works (but I get RF interference since it's not SPDIF) and switching to "Pro Audio" gives stereo output over SPDIF on "USB Audio Pro 2" but no option for passing through 5.1 audio.

Nightwing0815, 2023-08-30:
@somejavadev of course I can, here we go:

```
~$ sudo inxi --audio --machine --system
System:
  Host: Ultron Kernel: 6.2.0-31-generic x86_64 bits: 64
    Desktop: Cinnamon 5.8.4 Distro: Linux Mint 21.2 Victoria
Machine:
  Type: Desktop System: ASUS product: N/A v: N/A serial: N/A
  Mobo: ASUSTeK model: ROG STRIX Z790-F GAMING WIFI v: Rev 1.xx
    serial: xxxxxxxxxxxxxxx UEFI: American Megatrends v: 1202 date: 06/13/2023
Audio:
  Device-1: Intel driver: snd_hda_intel
  Device-2: NVIDIA driver: snd_hda_intel
  Device-3: ASUSTek USB Audio type: USB
    driver: hid-generic,snd-usb-audio,usbhid
  Sound Server-1: ALSA v: k6.2.0-31-generic running: yes
  Sound Server-2: PipeWire v: 0.3.48 running: yes
```
pavucontrol:
[image]

alsamixer:
[image]

How can I check, if I'm using all drivers for the Board?
I think, you have the same issues in dmesg, not only sound?

```
~$ sudo dmesg -x --color=always --level=err,warn
kern  :warn  : [    0.223080]  #17 #18 #19 #20 #21 #22 #23
kern  :warn  : [    0.256576] ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
kern  :warn  : [    0.495972] hpet_acpi_add: no address or irqs in _CRS
kern  :warn  : [    0.507364] device-mapper: core: CONFIG_IMA_DISABLE_HTABLE is disabled. Duplicate IMA measurements will not be recorded in the IMA log.
kern  :warn  : [    0.507383] platform eisa.0: EISA: Cannot allocate resource for mainboard
kern  :warn  : [    0.507384] platform eisa.0: Cannot allocate resource for EISA slot 1
kern  :warn  : [    0.507384] platform eisa.0: Cannot allocate resource for EISA slot 2
kern  :warn  : [    0.507384] platform eisa.0: Cannot allocate resource for EISA slot 3
kern  :warn  : [    0.507385] platform eisa.0: Cannot allocate resource for EISA slot 4
kern  :warn  : [    0.507385] platform eisa.0: Cannot allocate resource for EISA slot 5
kern  :warn  : [    0.507385] platform eisa.0: Cannot allocate resource for EISA slot 6
kern  :warn  : [    0.507386] platform eisa.0: Cannot allocate resource for EISA slot 7
kern  :warn  : [    0.507386] platform eisa.0: Cannot allocate resource for EISA slot 8
kern  :warn  : [    0.509378] efifb: Ignoring BGRT: unexpected or invalid BMP data
kern  :warn  : [    0.939514] acpi PNP0C14:06: duplicate WMI GUID 97845ED0-4E6D-11DE-8A39-0800200C9A66 (first instance was on PNP0C14:02)
kern  :warn  : [    0.945310] pci 10000:e0:06.0: Primary bus is hard wired to 0
kern  :warn  : [    0.945631] pci 10000:e0:1b.4: Primary bus is hard wired to 0
kern  :warn  : [    0.946143] pci 10000:e0:06.0: Primary bus is hard wired to 0
kern  :warn  : [    0.946146] pci 10000:e0:1b.4: Primary bus is hard wired to 0
kern  :err   : [    0.978182] i2c i2c-2: Failed to register i2c client ITE8853:00 at 0x40 (-16)
kern  :warn  : [    1.095516] nvidia: loading out-of-tree module taints kernel.
kern  :warn  : [    1.095520] nvidia: module license 'NVIDIA' taints kernel.
kern  :warn  : [    1.095521] Disabling lock debugging due to kernel taint
kern  :warn  : [    1.119413] pcieport 10000:e0:06.0: can't derive routing for PCI INT D
kern  :warn  : [    1.119413] pcieport 10000:e0:06.0: PCI INT D: no GSI
kern  :warn  : [    1.119540] pcieport 10000:e0:1b.4: can't derive routing for PCI INT A
kern  :warn  : [    1.119540] pcieport 10000:e0:1b.4: PCI INT A: no GSI
kern  :warn  : [    1.136850] ahci 10000:e0:17.0: can't derive routing for PCI INT A
kern  :warn  : [    1.136853] ahci 10000:e0:17.0: PCI INT A: no GSI
kern  :warn  : [    1.140166] pcieport 10000:e0:06.0: can't derive routing for PCI INT A
kern  :warn  : [    1.140167] nvme 10000:e1:00.0: PCI INT A: no GSI
kern  :warn  : [    1.140235] pcieport 10000:e0:1b.4: can't derive routing for PCI INT A
kern  :warn  : [    1.140236] nvme 10000:e2:00.0: PCI INT A: no GSI

kern  :warn  : [    1.224510] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  535.86.05  Fri Jul 14 20:46:33 UTC 2023
kern  :warn  : [    1.489043] ata8.00: supports DRM functions and may not be fully accessible
kern  :warn  : [    1.489211] ata6.00: supports DRM functions and may not be fully accessible
kern  :warn  : [    1.493984] ata8.00: supports DRM functions and may not be fully accessible
kern  :warn  : [    1.494279] ata6.00: supports DRM functions and may not be fully accessible
kern  :warn  : [    1.548573] ACPI Warning: \_SB.PC00.PEG1.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20221020/nsarguments-61)
[... 188 lines cut ...]
kern  :warn  : [    2.303487]   Arg2:   000000001666f675 <Obj>           Integer 0000000000000010
kern  :warn  : [    2.303489]   Arg3:   000000007db9d82a <Obj>           Buffer(4) 00 00 4B 56

kern  :err   : [    2.303492] ACPI Error: Aborting method \_SB.PC00.PEG1.PEGP._DSM due to previous error (AE_ALREADY_EXISTS) (20221020/psparse-529)
kern  :err   : [    2.303615] ACPI BIOS Error (bug): Failure creating named object [\_SB.PC00.PEG1.PEGP._DSM.USRG], AE_ALREADY_EXISTS (20221020/dsfield-184)
kern  :err   : [    2.303625] ACPI Error: AE_ALREADY_EXISTS, CreateBufferField failure (20221020/dswload2-477)

kern  :warn  : [    2.303633] No Local Variables are initialized for Method [_DSM]

kern  :warn  : [    2.303633] Initialized Arguments for Method [_DSM]:  (4 arguments defined for method invocation)
kern  :warn  : [    2.303634]   Arg0:   000000007db9d82a <Obj>           Buffer(16) 75 0B A5 D4 C7 65 F7 46
kern  :warn  : [    2.303637]   Arg1:   000000001666f675 <Obj>           Integer 0000000000000102
[... cut ...]
kern  :err   : [    2.303642] ACPI Error: Aborting method \_SB.PC00.PEG1.PEGP._DSM due to previous error (AE_ALREADY_EXISTS) (20221020/psparse-529)
kern  :warn  : [    4.049202] nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
kern  :warn  : [    4.071680] asus_wmi: Failed to get WMI notify code: AE_NOT_FOUND
kern  :warn  : [    4.071681] asus_wmi: Failed to get event during flush: -5
kern  :err   : [    4.256642] snd_hda_intel 0000:00:1f.3: no codecs found!
```

THX in advance,

ToM

somejavadev, 2023-08-30:
My dmesg logs are fairly clean:

`sudo dmesg -x --color=always --level=err,warn`

```
kern  :warn  : [    0.215427]  #17 #18 #19 #20 #21 #22 #23 #24 #25 #26 #27 #28 #29 #30 #31
kern  :warn  : [    0.263351] ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
kern  :warn  : [    0.522861] hpet_acpi_add: no address or irqs in _CRS
kern  :warn  : [    0.529139] tpm_crb: probe of MSFT0101:00 failed with error 378
kern  :warn  : [    0.533730] device-mapper: core: CONFIG_IMA_DISABLE_HTABLE is disabled. Duplicate IMA measurements will not be recorded in the IMA log.
kern  :warn  : [    0.533753] platform eisa.0: EISA: Cannot allocate resource for mainboard
kern  :warn  : [    0.533753] platform eisa.0: Cannot allocate resource for EISA slot 1
kern  :warn  : [    0.533754] platform eisa.0: Cannot allocate resource for EISA slot 2
kern  :warn  : [    0.533754] platform eisa.0: Cannot allocate resource for EISA slot 3
kern  :warn  : [    0.533755] platform eisa.0: Cannot allocate resource for EISA slot 4
kern  :warn  : [    0.533755] platform eisa.0: Cannot allocate resource for EISA slot 5
kern  :warn  : [    0.533756] platform eisa.0: Cannot allocate resource for EISA slot 6
kern  :warn  : [    0.533756] platform eisa.0: Cannot allocate resource for EISA slot 7
kern  :warn  : [    0.533756] platform eisa.0: Cannot allocate resource for EISA slot 8
kern  :warn  : [    0.774138] acpi PNP0C14:06: duplicate WMI GUID 97845ED0-4E6D-11DE-8A39-0800200C9A66 (first instance was on PNP0C14:02)
kern  :warn  : [    0.786894] usb: port power management may be unreliable
kern  :warn  : [    0.788325] nvme nvme1: missing or invalid SUBNQN field.
kern  :err   : [    0.808626] i2c i2c-2: Failed to register i2c client ITE8853:00 at 0x4e (-16)
kern  :warn  : [    7.749647] memfd_create() without MFD_EXEC nor MFD_NOEXEC_SEAL, pid=1 'systemd'
kern  :warn  : [    7.767034] block nvme0n1: the capability attribute has been deprecated.
daemon:warn  : [    7.805703] systemd[1]: Configuration file /lib/systemd/system/urbackupclientbackend.service is marked executable. Please remove executable permission bits. Proceeding anyway.
syslog:warn  : [    7.986212] systemd-journald[592]: File /var/log/journal/1a3c567e85da4f0b9d38eb143bdffa10/system.journal corrupted or uncleanly shut down, renaming and replacing.
kern  :err   : [    8.062349] Serial bus multi instantiate pseudo device driver INT3515:00: error -ENXIO: IRQ index 1 not found
kern  :err   : [    8.062381] Serial bus multi instantiate pseudo device driver INT3515:00: error -ENXIO: Error requesting irq at index 1
kern  :warn  : [    8.091553] asus_wmi: Failed to get WMI notify code: AE_NOT_FOUND
kern  :warn  : [    8.091553] asus_wmi: Failed to get event during flush: -5
kern  :warn  : [    8.105483] iwlwifi 0000:00:14.3: api flags index 2 larger than supported by driver
kern  :err   : [    8.284262] snd_hda_intel 0000:00:1f.3: no codecs found!
kern  :warn  : [    8.299278] thermal thermal_zone1: failed to read out thermal zone (-61)
kern  :err   : [    8.306945] iwlwifi 0000:00:14.3: WRT: Invalid buffer destination
kern  :warn  : [    9.332786] amdgpu 0000:03:00.0: Direct firmware load for amdgpu/gc_11_0_0_mes_2.bin failed with error -2
kern  :err   : [    9.661834] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
kern  :warn  : [   10.350337] amdgpu: SRAT table not found
kern  :warn  : [   16.583397] kauditd_printk_skb: 46 callbacks suppressed

```

Nightwing0815, 2023-09-03:
For the devs, here's an output of [alsa-info script](http://alsa-project.org/db/?f=e8e003c6d0b8116e277bc1d5f40d075cdaadbbf7)