alsa-project/alsa-lib #339: ALC3306 beeing registered as ALC287 on Lenovo Yoga

InfinityRed-Code, 2023-07-29:
I have a Lenovo Yoga 7 16IAP7 with Fedora Workstation 38 installed.

Kernel Verion: Linux 6.4.6-200.fc38.x86_64
alsa-lib version: alsa-lib-1.2.9-1.fc38.x86_64

I installed alsa-tools and went into hdajackretask to see what is going on. Then I noticed that my System is telling me that I have an ALC287 installed but actually it is a ALC3306. Sound is horrible and several fixes with ALC 287 for Legion Models did nit work. In the Settings I can see as well, that only 2 of the 4 integrated speakers are getting noticed.

Is there any known fix?

InfinityRed-Code, 2024-01-23:
Update to my Situation:

Upgraded to Fedora Workstation 39 with

Linux Kernel: 6.6.11-200.fc39.x86_64
alsa-lib version: alsa-lib-1.2.10-3.fc39.x86_64

any fixes now?

gdegruiter, 2025-04-07:
Got the exact same issue here on my Lenovo Yoga Pro 9i (16IMH9). The ACL3306 is also reported as an ALC287 in alsamixer. This might be the cause for my subwoofers shutting off after inactivity, resulting in horrible sound quality. I'm able to restart them by removing and re-adding modprobe rule "snd_hda_scodec_tas2871_i2c". 

On Linux kernel 6.14.0.

Torsteinws, 2025-04-30:
Same exact problem here. I am able to get sound from the subwoofers by running `sudo alsa force-reload`, but they will turn off again after being inactive for some time.

Laptop model: Lenovo Yoga Pro 9 16IRP8
Linux kernel: 6.14.0-15-generic
Alsa info: http://alsa-project.org/db/?f=47849853535c32841cd8dce39a781d8dba09e2ef

closed 2025-05-06 by perexg