alsa-project/alsa-lib #341: params_free functions don't work

imyxh, 2023-07-31:
The following code segfaults on the call to `snd_pcm_hw_params_free` on my machine:

```c
#include <alsa/asoundlib.h>
void main() {
	snd_pcm_hw_params_t *params;
	snd_pcm_hw_params_alloca(&params);
	snd_pcm_hw_params_free(params);
}
```

The same goes for `s/hw/sw/g`. Am I doing something wrong? In more complex examples it seems to fail with munmap_chunk complaining about an invalid pointer, or free complaining about an invalid size.

How are the `*_params_free` functions meant to be used?

perexg, 2023-08-01:
The free functions are for mallocated memory  - `snd_pcm_hw_params_malloc`. The alloca variants uses stack (see `man alloca` for the storage, thus this space is freed automatically when the caller function exits).

closed 2023-08-01 by perexg

imyxh, 2023-08-02:
Ah. That'll do it. I wasn't familiar with `alloca` and assumed that `_params_alloca` was just a funny way to say `_params_malloc`. Thanks for explaining.