alsa-project/alsa-lib #350: _dlsym_control_001 undeclared when compiling alsa-lib

cemkeylan, 2023-09-06:
Hello, as mentioned I'm getting the above error when compiling alsa-lib. Full error below.

```
In file included from ../../include/local.h:177,
                 from control_local.h:22,
                 from control.c:188:
control.c: In function 'snd_ctl_open_conf':
../../include/global.h:98:36: warning: implicit declaration of function '__STRING' [-Wimplicit-function-declaratio]
   98 | #define SND_DLSYM_VERSION(version) __STRING(version)
      |                                    ^~~~~~~~
control.c:1541:25: note: in expansion of macro 'SND_DLSYM_VERSION'
 1541 |                         SND_DLSYM_VERSION(SND_CONTROL_DLSYM_VERSION), 1);
      |                         ^~~~~~~~~~~~~~~~~
../../include/control.h:43:41: error: '_dlsym_control_001' undeclared (first use in this function)
   43 | #define SND_CONTROL_DLSYM_VERSION       _dlsym_control_001
      |                                         ^~~~~~~~~~~~~~~~~~
../../include/global.h:98:45: note: in definition of macro 'SND_DLSYM_VERSION'
   98 | #define SND_DLSYM_VERSION(version) __STRING(version)
      |                                             ^~~~~~~
control.c:1541:43: note: in expansion of macro 'SND_CONTROL_DLSYM_VERSION'
 1541 |                         SND_DLSYM_VERSION(SND_CONTROL_DLSYM_VERSION), 1);
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
../../include/control.h:43:41: note: each undeclared identifier is reported only once for each function it appears in
   43 | #define SND_CONTROL_DLSYM_VERSION       _dlsym_control_001
      |                                         ^~~~~~~~~~~~~~~~~~
../../include/global.h:98:45: note: in definition of macro 'SND_DLSYM_VERSION'
   98 | #define SND_DLSYM_VERSION(version) __STRING(version)
      |                                             ^~~~~~~
control.c:1541:43: note: in expansion of macro 'SND_CONTROL_DLSYM_VERSION'
 1541 |                         SND_DLSYM_VERSION(SND_CONTROL_DLSYM_VERSION), 1);
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:415: control.lo] Error 1
```

I have tried to understand the reason this occurs. This doesn't happen when compiling version `1.2.9`, yet I also can't understand how and where the declaration is for that version.

thesamesam, 2023-09-06:
We had a report of this downstream for musl systems too: https://bugs.gentoo.org/913573.

cemkeylan, 2023-09-06:
Oh that makes sense, I'm using musl libc as well.

perexg, 2023-09-06:
Does something like this work?

```
diff --git a/include/global.h b/include/global.h
index dfe9bc2b..3ecaeee8 100644
--- a/include/global.h
+++ b/include/global.h
@@ -51,6 +51,11 @@ const char *snd_asoundlib_version(void);
 #define ATTRIBUTE_UNUSED __attribute__ ((__unused__))
 #endif
 
+#ifndef __STRING
+/** \brief Return 'x' argument as string */
+#define __STRING(x)     #x
+#endif
+
 #ifdef PIC /* dynamic build */
 
 /** \hideinitializer \brief Helper macro for #SND_DLSYM_BUILD_VERSION. */
@@ -71,11 +76,6 @@ struct snd_dlsym_link {
 
 extern struct snd_dlsym_link *snd_dlsym_start;
 
-#ifndef __STRING
-/** \brief Return 'x' argument as string */
-#define __STRING(x)     #x
-#endif
-
 /** \hideinitializer \brief Helper macro for #SND_DLSYM_BUILD_VERSION. */
 #define __SND_DLSYM_VERSION(prefix, name, version) _ ## prefix ## name ## version
 /**
```

cemkeylan, 2023-09-06:
Yes, it compiles now :slightly_smiling_face: