alsa-project/alsa-lib #352: libasound 1.2.10 seg faults on RPi2 (32-bit) without --disable-largefile

heitbaum, 2023-09-09:
Since the default change in 1.2.10 Kodi 32 or RPi was crashing. Revert to previous with — disable-largefile

Ref:
- #333
- https://github.com/alsa-project/alsa-lib/pull/333#issuecomment-1712410543
- https://forum.libreelec.tv/thread/27500-libreelec-nightly-rpi2-latest-builds-fail-to-boot/
- https://github.com/LibreELEC/LibreELEC.tv/pull/8126
- https://github.com/LibreELEC/LibreELEC.tv/pull/8126#issuecomment-1712409175

HiassofT, 2023-09-09:
The seg fault can be triggered simply by outputting to the default or dmix pcm, eg on an USB audio card.

I used `speaker-test -c 2 -D default:CARD=DAC` here and gdb shows that it's crashing in pcm_direct.c:
```
Program received signal SIGSEGV, Segmentation fault.
snd1_pcm_direct_hw_refine (pcm=0x266968, params=0x7e92f594)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm_direct.c:828
828		return &params->masks[var - SND_PCM_HW_PARAM_FIRST_MASK];
(gdb) where
#0  snd1_pcm_direct_hw_refine (pcm=0x266968, params=0x7e92f594)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm_direct.c:828
#1  0x76ed6318 in snd_pcm_hw_refine (pcm=<optimized out>, params=params@entry=0x7e92f594)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm_params.c:2363
#2  0x76eed480 in snd_pcm_plug_hw_refine_sprepare (pcm=<optimized out>, sparams=0x7e92f594)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm_plug.c:730
#3  0x76ed616c in snd1_pcm_hw_refine_slave (pcm=0x266f08, params=0x7e92f8b0, cprepare=<optimized out>, 
    cchange=0x76eee9a4 <snd_pcm_plug_hw_refine_cchange>, sprepare=0x76eed438 <snd_pcm_plug_hw_refine_sprepare>, 
    schange=0x76eedafc <snd_pcm_plug_hw_refine_schange>, srefine=0x76eed4f0 <snd_pcm_plug_hw_refine_slave>)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm_params.c:2224
#4  0x76eee988 in snd_pcm_plug_hw_refine (pcm=<optimized out>, params=<optimized out>)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm_plug.c:997
#5  0x76ed6318 in snd_pcm_hw_refine (pcm=pcm@entry=0x266f08, params=params@entry=0x7e92f8b0)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm_params.c:2363
#6  0x76ecd310 in snd_pcm_hw_params_any (pcm=0x266f08, params=0x7e92f8b0)
    at /home/hias/libreelec/libreelec-master/build.LibreELEC-RPi2.arm-12.0-devel/build/alsa-lib-1.2.10/src/pcm/pcm.c:3906
#7  0x00009a68 in main ()
```

It seems pcm_direct.c was missed in the "include reshuffle commit" ad3a8b8. Adding a `#include "config.h"` to the top of the file so the system includes get the `_FILE_OFFSET_BITS` fixes the issue
```
--- a/src/pcm/pcm_direct.c
+++ b/src/pcm/pcm_direct.c
@@ -19,6 +19,7 @@
  *
  */
   
+#include "config.h"
 #include <stdio.h>
 #include <stdlib.h>
 #include <stddef.h>
```

I'll leave it to you how to best handle that and I guess it might be good idea to go over all the files in detail again and check if more files are missing the (indirect) config.h include

tiwai, 2023-09-09:
Thanks for the report and the quick analysis!

Also a few other pcm plugin code should be fixed.  And, I believe it should be better to put pcm_local.h inclusions instead of config.h. It's more consistent with other PCM code.

I'll push the fix.

perexg, 2023-09-13:
Fixed in 0e3dfb9f705ca78be34cd70fd59d67c431e29cc7 and maybe others in 81a7a93636d9472fcb0c2ff32d9bfdf6ed10763d .

closed 2023-09-13 by perexg