alsa-project/alsa-lib #366: Speaker volume control doesn't work by default on Lenovo Yoga 7 (AMD 7840hs)

smaug----, 2023-11-16:
(Not sure which alsa-repo is the correct one for this issue, or is this even a Fedora issue)
By default controlling headphone output volume works, but speaker output seems to be either muted or very loud. 

The following tweak fixes the issue.

+++ /usr/share/alsa-card-profile/mixer/paths/analog-output-speaker.conf	2023-11-14 02:45:24.182339962 +0200
@@ -79,8 +79,6 @@
 override-map.2 = all-left,all-right
 
 [Element Master]
-switch = mute
-volume = merge
 override-map.1 = all
 override-map.2 = all-left,all-right
 
@@ -243,4 +241,8 @@
 override-map.1 = all-center
 override-map.2 = all-center,lfe
 
+[Element Master]
+switch = mute
+volume = ignore
+
 .include analog-output.conf.common

Gpinchon, 2023-11-19:
Same here, adding these lines to the conf fixe the issue but they're erased on every updates of alsa, pretty annoying.

Gpinchon, 2023-11-29:
Omg, the latest update fucked up the volume again, this really needs to be addressed. The solution proposed by @smaug---- doesn't work for me, what works is
- edit /usr/share/alsa-card-profile/mixer/paths/analog-output.conf.common
- go down to [Element PCM]

it should looke like something like this, for me **EVERY UPDATE** removes Element Hardware Master and Element Master parts :
```
[Element Hardware Master]
switch = mute
volume = ignore

[Element Master]
switch = mute
volume = ignore

[Element PCM]
switch = mute
volume = merge
override-map.1 = all
override-map.2 = all-left,all-right

```

AboveYou, 2024-01-04:
Got the same problem and fixed it myself by automating the process, check it out [here](https://github.com/AboveYou/fix_alsa_mixer). @Gpinchon 

I might submit a pull-request later.

then4p, 2024-01-11:
I'm not quite sure if this is due to the kernel, but on 6.6 and 6.7 I don't need the fix anymore, also the tweeters work as well! Sadly, this breaks headphones. They're completely silent and have to be turned up in alsamixer. 

Also mic volume with a headset connected controls mic boost instead of mic volume...

Is there an easy way to fix these paths, is it just an incorrect pinout? I found the HDA Analyzer but it's built for python 2 and I haven't set that up atm.

smaug----, 2024-01-12:
Yeah, the headphones issue is why I need to stay on kernel 6.6.6. I think 6.6.7 broke it, but I haven't had time to investigate which change exactly.

smaug----, 2024-01-12:
Unfortunately I don't see anything for "rear speaker" in alsamixer.

soyersoyer, 2024-01-16:
6.6.7 has this patch:
https://patchwork.kernel.org/project/alsa-devel/patch/20231207182035.30248-1-tiwai@suse.de/

The ALC287_FIXUP_YOGA9_14IAP7_BASS_SPK_PIN added with this patch:
https://www.spinics.net/lists/alsa-devel/msg145801.html

perexg, 2024-01-17:
This issue is not related to alsa-lib. Use the kernel discussion mailing list - linux-sound@vger.kernel.org - for the further discussion.

closed 2024-01-17 by perexg

infoat, 2024-04-29:
The workaround from Gpinchon works for my Lenovo Slim Pro 7 (14APH8) https://github.com/alsa-project/alsa-lib/issues/366#issuecomment-1832668734 - for me headphones in alsamixer show up as 'bass speaker.'