alsa-project/alsa-lib #373: Linux native games don't have sound

hyunlee1o, 2023-12-30:
### Explanation
Sometimes when i run Scourgebringer game i get this log:
```
$ ./start.sh
ALSA lib pcm_direct.c:2180:(_snd_pcm_direct_new) unable to create IPC semaphore
ALSA lib pcm_direct.c:2180:(_snd_pcm_direct_new) unable to create IPC semaphore
ALSA lib pcm_direct.c:2180:(_snd_pcm_direct_new) unable to create IPC semaphore
```

I think something it may be related to this.
Multiple linux native games don't have sound at all, i haven't got this problem when running wine games.

Seems alsa has problems when i don't run processes with root permissions and the same error can be seen.
```
 $ alsabat
alsa-utils version 1.2.10

Entering playback thread (ALSA).
ALSA lib pcm_direct.c:2180:(_snd_pcm_direct_new) unable to create IPC semaphore
Cannot open PCM playback device: Permiso denegado(-13)
Exit playback thread fail: -13
Entering capture thread (ALSA).
$ sudo alsabat
[sudo] contraseña para user: 
alsa-utils version 1.2.10

Entering playback thread (ALSA).
Get period size: 940  buffer size: 15052
Playing generated audio sine wave
Entering capture thread (ALSA).
Get period size: 940  buffer size: 15052
Recording ...
Playback completed.
Capture completed.

BAT analysis: signal has 65536 frames at 44100 Hz, 1 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 997,00 Hz
Amplitude: 564,3; Percentage: [1]
Detected peak at 8,75 Hz of 18,40 dB
 Total 18,4 dB from 8,75 to 8,75 Hz
 FAIL: Peak freq too low 8,75 Hz
Detected peak at 14,13 Hz of 18,49 dB
 Total 21,5 dB from 14,13 to 14,13 Hz
 FAIL: Peak freq too low 14,13 Hz
Detected peak at 18,84 Hz of 18,11 dB
 Total 25,1 dB from 17,50 to 18,84 Hz
 FAIL: Peak freq too low 18,84 Hz
Detected peak at 24,22 Hz of 23,71 dB
 Total 32,5 dB from 20,19 to 28,26 Hz
 FAIL: Peak freq too low 24,22 Hz
Detected peak at 30,95 Hz of 23,09 dB
 Total 33,7 dB from 30,28 to 32,30 Hz
 FAIL: Peak freq too low 30,95 Hz
Detected peak at 33,65 Hz of 18,00 dB
 Total 33,8 dB from 33,65 to 33,65 Hz
 FAIL: Peak freq too low 33,65 Hz
Detected peak at 35,66 Hz of 18,12 dB
 Total 33,9 dB from 35,66 to 35,66 Hz
 FAIL: Peak freq too low 35,66 Hz
Detected peak at 37,01 Hz of 17,76 dB
 Total 34,0 dB from 37,01 to 37,01 Hz
 FAIL: Peak freq too low 37,01 Hz
Detected peak at 38,36 Hz of 17,66 dB
 Total 34,1 dB from 38,36 to 38,36 Hz
 FAIL: Peak freq too low 38,36 Hz
Detected peak at 41,05 Hz of 23,87 dB
 Total 35,4 dB from 39,70 to 42,39 Hz
 FAIL: Peak freq too low 41,05 Hz
Detected at least 10 signal(s) in total

Return value is -1003

```
What i find strange that after running alsabat using sudo/root i can run it without sudo without issue
```
$ alsabat
alsa-utils version 1.2.10

Entering playback thread (ALSA).
Get period size: 940  buffer size: 15052
Playing generated audio sine wave
Entering capture thread (ALSA).
Get period size: 940  buffer size: 15052
Recording ...
Playback completed.
Capture completed.

BAT analysis: signal has 65536 frames at 44100 Hz, 1 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 997,00 Hz
Amplitude: 562,4; Percentage: [1]
Detected peak at 8,75 Hz of 18,91 dB
 Total 18,9 dB from 8,75 to 8,75 Hz
 FAIL: Peak freq too low 8,75 Hz
Detected peak at 13,46 Hz of 19,70 dB
 Total 22,3 dB from 13,46 to 13,46 Hz
 FAIL: Peak freq too low 13,46 Hz
Detected peak at 16,15 Hz of 18,30 dB
 Total 23,8 dB from 16,15 to 16,15 Hz
 FAIL: Peak freq too low 16,15 Hz
Detected peak at 18,84 Hz of 18,26 dB
 Total 24,9 dB from 18,84 to 18,84 Hz
 FAIL: Peak freq too low 18,84 Hz
Detected peak at 24,90 Hz of 25,33 dB
 Total 33,5 dB from 22,88 to 28,94 Hz
 FAIL: Peak freq too low 24,90 Hz
Detected peak at 38,36 Hz of 19,80 dB
 Total 33,8 dB from 38,36 to 39,03 Hz
 FAIL: Peak freq too low 38,36 Hz
Detected peak at 42,39 Hz of 22,61 dB
 Total 34,3 dB from 41,72 to 42,39 Hz
 FAIL: Peak freq too low 42,39 Hz
Detected peak at 44,41 Hz of 19,27 dB
 Total 34,4 dB from 44,41 to 44,41 Hz
 FAIL: Peak freq too low 44,41 Hz
Detected peak at 49,80 Hz of 18,78 dB
 Total 34,5 dB from 49,80 to 49,80 Hz
 FAIL: Peak freq too low 49,80 Hz
Detected peak at 119,78 Hz of 21,63 dB
 Total 34,9 dB from 119,78 to 120,45 Hz
 FAIL: Peak freq too low 119,78 Hz
Detected at least 10 signal(s) in total

Return value is -1003
```

### My system
```
       _,met$$$$$gg.          user@debian 
    ,g$$$$$$$$$$$$$$$P.       ------------ 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux trixie/sid x86_64 
 ,$$P'              `$$$.     Kernel: 6.6.8-amd64 
',$$P       ,ggs.     `$$b:   Uptime: 6 mins 
`d$$'     ,$P"'   .    $$$    Packages: 4328 (dpkg), 8 (flatpak) 
 $$P      d$'     ,    $$P    Shell: fish 3.6.4 
 $$:      $$.   -    ,d$$'    Resolution: XXXXXXXXXXXXXXX
 $$;      Y$b._   _,d$P'      DE: i3-with-shmlog 
 Y$$.    `.`"Y$$$$P"'         WM: i3 
 `$$b      "-.__              Theme: Breeze [GTK2/3] 
  `Y$$                        Terminal: x-terminal-emul 
   `Y$$.                      CPU: XXXXXXXXXXXXXXXXXXXXXXXXX
     `$$b.                    GPU: XXXXXXXXXXXXXXXXXXXXXXXXx
       `Y$$b.                 Memory: 3087MiB / 15903MiB 
          `"Y$b._
              `"""                                 
```

perexg, 2024-01-02:
The IPC permissions are set in `/usr/share/alsa/alsa.conf`. See `defaults.pcm.ipc_*` keys. You may list currently allocated IPC resources using command `ipcs` like `ipcs | grep 567` (see `defaults.pcm.ipc_key 5678293` in config).

hyunlee1o, 2024-01-02:
In `alsa.conf` 
```
defaults.pcm.ipc_key 5678293
defaults.pcm.ipc_gid audio
defaults.pcm.ipc_perm 0660
```

perexg, 2024-01-02:
Yes, then look for the allocated IPC key (and used permissions) using the`ipcs` command when things does not work for the given user.

If you know the key, you can use command like `ipcs -i KEY -m -p` to show all details about resource like group etc.

hyunlee1o, 2024-01-02:
```
$ ipcs -p -c --human -t

------ Message Queues Send/Recv/Change Times --------
msqid    owner      send                 recv                 change

------ Shared Memory Attach/Detach/Change Times --------
shmid      owner      attached             detached             changed
1          sddm       Jan  2 12:32:49      Jan  2 12:33:04      Jan  2 12:32:49
2          sddm       Jan  2 12:32:49      Jan  2 12:33:04      Jan  2 12:32:49
6          user      Jan  2 12:32:54      Not set              Jan  2 12:32:54
11         user      Jan  2 12:32:55      Not set              Jan  2 12:32:55
14         user      Jan  2 12:32:55      Not set              Jan  2 12:32:55
17         user      Jan  2 12:32:55      Not set              Jan  2 12:32:55
20         user      Jan  2 12:32:55      Not set              Jan  2 12:32:55
24         user      Jan  2 12:32:58      Not set              Jan  2 12:32:58
27         user      Jan  2 12:32:58      Not set              Jan  2 12:32:58
29         user      Jan  2 12:33:00      Not set              Jan  2 12:33:00
30         user      Jan  2 12:33:00      Not set              Jan  2 12:33:00
52         user      Jan  2 12:40:56      Jan  2 12:40:58      Jan  2 12:35:19

------ Semaphore Operation/Change Times --------
semid    owner      last-op                    last-changed
0        sddm        Tue Jan  2 12:33:04 2024   Tue Jan  2 12:32:49 2024
2        user       Tue Jan  2 12:32:54 2024   Tue Jan  2 12:32:54 2024
3        user       Tue Jan  2 12:40:58 2024   Tue Jan  2 12:40:56 2024
```

```
$ ipcs -m -i 52

Shared memory Segment shmid=52
uid=1000        gid=29  cuid=1000       cgid=1000
mode=0660       access_perms=0660
bytes=488       lpid=71488      cpid=25954      nattch=0
att_time=Tue Jan  2 12:40:56 2024  
det_time=Tue Jan  2 12:40:58 2024  
change_time=Tue Jan  2 12:35:19 2024
```

hyunlee1o, 2024-01-03:
Okay, the default semaphore from alsa i think it's id 0.
When i run `alsabat-test` it created another one  

```
---- Creadores/propietarios matrices semáf. ----
semid      perms      cuid       cgid       uid        gid 
0          660        sddm       sddm       sddm       sddm 
2          600        jorge      jorge      jorge      jorge
3          660        jorge      jorge      jorge      audio
```

```
$ sudo ipcs -s -i 0
[sudo] password for jorge: 

Semaphore Array semid=0
uid=113  gid=121         cuid=113        cgid=121
mode=0660, access_perms=0660
nsems = 1
otime = Wed Jan  3 14:55:58 2024  
ctime = Wed Jan  3 14:55:46 2024  
semnum     value      ncount     zcount     pid       
0          0          0          0          3323      

 $ ipcs -s -i 3

Semaphore Array semid=3
uid=1000         gid=29  cuid=1000       cgid=1000
mode=0660, access_perms=0660
nsems = 1
otime = Wed Jan  3 15:00:47 2024  
ctime = Wed Jan  3 15:00:44 2024  
semnum     value      ncount     zcount     pid       
0          0          0          0          46625   
```

hyunlee1o, 2024-01-06:
Now my problem mutated for some reason
```
Started initial process 22247 from ./start.sh
Start monitoring process.
Running ScourgeBringer
ALSA lib pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
Cannot get required symbol EVP_MD_size from libssl
./Launch.sh: línea 24: 22266 Abortado                ./ScourgeBringer $@
Monitored process exited.
Initial process has exited (return code: 34304)
All processes have quit
Exit with return code 34304
```

perexg, 2024-01-17:
What is `sddm` user ? It's probably a login / session manager. Thus your distro configuration is wrong. This user should be in audio group, too.

hyunlee1o, 2024-01-17:
Simple Desktop Display Manager (SDDM)
The thing is for some random reason, it's not giving me problems anymore :clown_face:

closed 2024-01-30 by perexg