alsa-project/alsa-lib #387: Strange trigger_htstamp after snd_pcm_drop

z-s-e, 2024-03-07:
I have conceptually the following situation (actual code can be found [here](https://gitlab.com/z-s-e/alpe/-/blob/main/src/alpe_alsa.cpp?ref_type=heads#L590)):
```
// sound is currently playing
snd_pcm_status(pcm, pcm_status);
snd_pcm_status_get_htstamp(pcm_status, &ts1);
snd_pcm_drop(pcm);
snd_pcm_status(pcm, pcm_status);
snd_pcm_status_get_trigger_htstamp(status, &ts2);
```
On my hardware I get a `ts2` timestamp that is earlier than `ts1`.  From my understanding that should be impossible, as `ts2` should contain the timestamp of when the drop happened, which should obviously be after `ts1`. Is that a bug or do I misunderstand the meaning of those timestamps?

perexg, 2024-03-07:
It looks like a bug. Could you show the plugin chain (using `snd_pcm_dump` or so) ? Or are you using direct `hw:X` device?

z-s-e, 2024-03-07:
I'm using `front:CARD=Generic_1,DEV=0`

snd_pcm_dump gives:

perexg, 2024-03-07:
There should be more information (lines after `Slave:`).

z-s-e, 2024-03-07:
ah yeah, sorry.

```
Soft volume PCM
Control: PCM Playback Volume
min_dB: -51
max_dB: 0
resolution: 256
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 96000
  period_size  : 48000
  period_time  : 1000000
  tstamp_mode  : ENABLE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 48000
  period_event : 0
  start_threshold  : 9223372036854775807
  stop_threshold   : 96000
  silence_threshold: 0
  silence_size : 0
  boundary     : 6755399441055744000
Slave: Hardware PCM card 1 'HD-Audio Generic' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 96000
  period_size  : 48000
  period_time  : 1000000
  tstamp_mode  : ENABLE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 48000
  period_event : 0
  start_threshold  : 9223372036854775807
  stop_threshold   : 96000
  silence_threshold: 0
  silence_size : 0
  boundary     : 6755399441055744000
  appl_ptr     : 96000
  hw_ptr       : 47178
```

z-s-e, 2024-03-07:
The code is called directly from a switch statement that checks for the state [here](https://gitlab.com/z-s-e/alpe/-/blob/main/src/alpe_alsa.cpp?ref_type=heads#L643), so this code gets only called when the last update is in `SND_PCM_STATE_RUNNING` but I also just re-checked it by verifying the value of `snd_pcm_status_get_state` right before the drop.

perexg, 2024-03-07:
Is trigger timestamp identical before and after drop?

z-s-e, 2024-03-07:
Good call, indeed they are the same.

perexg, 2024-03-11:
I'd like to keep this open until resolved.

z-s-e, 2024-03-19:
Some more info: This happens on my integrated audio (ALC287 Analog) as well as when outputting to hdmi. I've also tested a usb interface, there it does not happen.

z-s-e, 2024-07-23:
@perexg I do not have a kernel dev setup yet to test/verify, but looking at the code I might have an idea of what the problem is: It seems the hda_controller sets the trigger_tstamp_latched boolean at stream start, but it never gets reset. From my understanding the following may fix this:

```
diff --git a/sound/core/pcm_native.c b/sound/core/pcm_native.c
index 4057f9f10aee..7ddc4dc9829d 100644
--- a/sound/core/pcm_native.c
+++ b/sound/core/pcm_native.c
@@ -1196,6 +1196,7 @@ static void snd_pcm_trigger_tstamp(struct snd_pcm_substream *substream)
        if (runtime->trigger_master == substream) {
                if (!runtime->trigger_tstamp_latched)
                        snd_pcm_gettime(runtime, &runtime->trigger_tstamp);
+               runtime->trigger_tstamp_latched = false;
        } else {
                snd_pcm_trigger_tstamp(runtime->trigger_master);
                runtime->trigger_tstamp = runtime->trigger_master->runtime->trigger_tstamp;
```

z-s-e, 2024-08-09:
I can now confirm that my idea was correct. Attached is a [reproducer](https://github.com/user-attachments/files/16566958/alsa_timestamp_bug.tar.gz), it can be verified under qemu with an ich9-intel-hda device.

Actually at first my fix didn't seem to work, but it turned out that the default alsa device contained dmix here, which does not call the SNDRV_PCM_IOCTL_DROP of the underlying device when dropping, so with dmix the timestamps are still broken even with my patch, just as a heads-up.

I had one more idea for an alternate fix; just remove this trigger_tstamp_latched stuff. The hda_controller is the only driver using that, and while I understand the timestamp under the spinlock there can potentially be more accurate, I am not so sure it is worth it. In qemu I measured around 300 nanoseconds difference between the timestamp taken in the driver vs the timestamp in snd_pcm_trigger_tstamp, so unless one gets unlucky and there is some preemption in-between it should be negligible. Or it may be even possible to generally disable preemption until the timestamp in snd_pcm_trigger_tstamp is taken, to potentially make the timestamps better for all drivers...

Anyway, I'd like to get started contributing to the kernel, so this seems like a good opportunity. Could you tell me which fix you prefer, so I can prepare a patch for the ML?

z-s-e, 2024-08-12:
I'll just go ahead with proposing the removal. That appears to me to be the most robust solution, and personally I am always in favor of removing unneeded complexity.

perexg, 2024-08-12:
Latched timestamps were added in https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=2b79d7a6bf341ebeffb85fc2b3577d0cfa810c49 . Just fix the trigger callback in hda_controller.c.

z-s-e, 2024-08-12:
Yes, I've seen that commit already. Just adding that to the hda_controller might not fix it for other uses of snd_hdac_stream_timecounter_init (which is the place that raises that flag, and grepping shows it is also called by sound/soc/intel/skylake/skl-pcm.c), so that seems like a more risky fix that potentially needs to be kept in mind for all hda drivers, that is why I rather would remove it, especially since it does not seem to provide any real value.

perexg, 2024-08-12:
Please, sent a proposed commit with a good description to linux-sound@vger.linux.org mailing list with Cc to all relevant maintainers of affected drivers (including sound/soc/intel tree). Here is not a right place to discuss this removal.

z-s-e, 2024-08-12:
Yep, thanks.

perexg, 2024-10-02:
Resolved in https://github.com/tiwai/sound/commit/df5215618fbe425875336d3a2d31bd599ae8c401 .

closed 2024-10-02 by perexg